emsesp/EMS-ESP32 #1342: Buderus heatpump Logatherm WLW176i-12 T180 not recognized (unknown device)

fra-ol, 2023-10-17:
**PROBLEM DESCRIPTION**

My system setup is the following:

Buderus Logatherm WLW176i-12 T180 heatpump
Buderus GUI Logamatic BC400 (integrated in heatpump unit for  control unit XCU-THH-11.)
Buderus WLAN module MX300

When using the latest developer SW3.6.2-dev.2, all of the devices listed above are not recognized correctly or not recognized at all. The heat pump is displayed as unknown device 0x08.
Therefore, could you please add these devices to the next EMS-ESP development SW?

**REQUESTED INFORMATION**

{
  "System Info": {
    "version": "3.6.2-dev.2",
    "platform": "ESP32-S3",
    "uptime": "000+09:06:55.555",
    "uptime (seconds)": 32815,
    "free mem": 181,
    "max alloc": 167,
    "free app": 6096,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -44,
    "IPv4 address": "192.168.178.92/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 135269,
    "bus reads (tx)": 10972,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 2,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 75,
    "web log buffer": 75
  },
  "Devices": [
    {
      "type": "thermostat",
      "name": "Rego 3000/UI800",
      "device id": "0x10",
      "product id": 253,
      "version": "47.07",
      "entities": 56,
      "handlers received": "0x06 0x02BA 0x02B0 0x029C 0x0472 0x02BB 0x02B1 0x029D 0x0473 0x02BC 0x02B2 0x029E 0x0474 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02A7 0x02A8 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E",
      "handlers ignored": "0xE7 0x02E0 0x02EA 0x0507 0x0508 0x0509 0x050A 0x061E 0x059D 0xF9 0xBF 0xF7 0x0165 0x0166 0x0167 0x0168 0x0169 0x016A 0x016B 0x016C 0xF6 0x01 0x016E 0x02FF 0x0AF7 0x0AF8 0x0AF9 0x0AFA 0x0B01 0x0B15 0x0B16 0x0B17 0x0B18 0x0B0B 0x0B0C 0x0B0D 0x0B0E 0x0B0F 0x0B10 0x0B11 0x0B12 0x02C3 0x02C4 0x02C5 0x0"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "07.02",
      "entities": 0,
      "handlers ignored": "0xF7 0x0CC2 0x0CBC 0x01 0x0CC3 0x0CBD 0x1259 0xF6 0x125A 0x0CBF 0x1258 0x2040"
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x08",
      "product id": 8,
      "version": "00.00",
      "entities": 0,
      "handlers ignored": "0x02D6 0xE3 0xE4 0xE9 0xD1 0xE5 0x048F 0x0494 0x0495 0x04A2 0x04A8 0x04AC 0x04AE 0x04AF 0x04CA 0x04CB 0x048D 0xF9 0xBF 0x14 0x049F 0x0488 0x04A0 0xF7 0xF6 0x01 0x03 0x04 0x17 0xE6 0xEA 0xEF 0x02E0 0x0484 0x0499 0xFF06 0xFF07 0xFE06 0xFE07"
    }
  ]
}

**ems-esp:$ read 8 2**
000+09:20:22.343 N 2: [emsesp] unknown(0x08) -W-> Me(0x0B), Version(0x02), data: 08 00 00 00 00 00 00 00 00 03 03 18

**EXPECTED BEHAVIOUR**

Buderus Logatherm WLW176i-12 T180  heatpump (device ID 0x08) recognized as known device incl. the related entities of the message IDs 0x02D6 0xE3 0xE4 0xE9 0xD1 0xE5 0x048F 0x0494 0x0495 0x04A2 0x04A8 0x04AC 0x04AE 0x04AF 0x04CA 0x04CB 0x048D 0xF9 0xBF 0x14 0x049F 0x0488 0x04A0 0xF7 0xF6 0x01 0x03 0x04 0x17 0xE6 0xEA 0xEF 0x02E0 0x0484 0x0499 0xFF06 0xFF07 0xFE06 0xFE07

**SCREENSHOTS**

[image]
[image]
[image]
[image]

MichaelDvP, 2023-10-18:
It's the same type as the WS170 in #1334, Please try https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

fra-ol, 2023-10-18:
I tried build v3.6.3-dev.3a and it seems fine. The heat pump has now 143 entities. Thanks!

[image]
[image]
[image]
[image]

proddy, 2023-10-19:
wow, 143 entities. I didn't know the devices page could handle that load.

bbqkees, 2023-10-20:
Could you perhaps make screenshots of the list of entities?
I have a few examples for boilers but not yet of a heatpump with this many entities.
I can assemble them then.

proddy, 2023-10-20:
> Could you perhaps make screenshots of the entire list of entities? I have a few examples for boilers but not yet of a heatpump with this many entities.
> 
> You may need a plugin for a rolling screenshot, otherwise I can stitch them together. https://www.makeuseof.com/how-to-full-page-screenshot-chrome-firefox/

there's an export button, use that!

fra-ol, 2023-10-20:
I emailed the export to kees.

bluegaspode, 2023-10-22:
do you need another export? 

Buderus WSW196i gives me 151 entities in the ui, also on export  (and the devices pages seems to handle the load without any problems)

proddy, 2023-10-22:
would be good, so I can add to the demo on https://ems-esp.derbyshire.nl/dashboard/devices

bluegaspode, 2023-10-22:
Here is an export (from a Buderus WSW196i)

[Boiler_Enviline_Compress 6000AW_Hybrid 3000-7000iAW_SupraEco_Geo 5xx_WLW196i_WSW196i.csv](https://github.com/emsesp/EMS-ESP32/files/13064581/Boiler_Enviline_Compress.6000AW_Hybrid.3000-7000iAW_SupraEco_Geo.5xx_WLW196i_WSW196i.csv)

bbqkees, 2023-10-23:
Thanks for the exports.

@proddy What I am missing in these exports is whether the entity is only readable or also writeable.

proddy, 2023-10-23:
> @proddy What I am missing in these exports is whether the entity is only readable or also writeable.

I'll add

bbqkees, 2023-10-23:
Ideally an export would also include the range of data that can be set, or the list of option.
If you have such an export it is much more easy to integrate it into systems that need manual integration.

proddy, 2023-10-23:
> Ideally an export would also include the range of data that can be set, or the list of option. If you have such an export it is much more easy to integrate it into systems that need manual integration.

I've extended the export to show the options and range. Note, it is an export of what the user has filtered. If we want a dump of all available entities then we could add a new feature to the Download page

closed 2023-10-28 by fra-ol