emsesp/EMS-ESP32 #1347: Only 3 entities appear on Bosch FR100 thermostat from 2011

azertyfun, 2023-10-20:
### PROBLEM DESCRIPTION

I bought a second-hand Bosch FR100, which does controls the boiler and I see on the EMS bus. However I only see 3 entities: `date/time` (ro), `dhw charge` (r/w), `hc5 target flow temperature` (ro)

In #105 it is said that devices built after 2008 should be EMS-writable, but mine is from Nov 2011 (FD tag `111`) and is barely even readable.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [X] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

```json
{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32-S3",
    "uptime": "000+15:32:53.531",
    "uptime (seconds)": 55973,
    "free mem": 184,
    "max alloc": 167,
    "free app": 6096,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -68,
    "IPv4 address": "192.168.178.8/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "1.1.1.2",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "pool.ntp.org",
    "tz label": "Europe/Brussels"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 18156,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 3,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
[... 157 lines cut ...]
2023-10-19 19:55:42.655 T 95: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 E2 49 00 00 02 00 C0 00 BC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
2023-10-19 19:55:42.870 T 96: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 32 00 BC 80 00 81 00 00 01 00 00 01 68 00 01 3C 00
2023-10-19 19:55:52.636 T 97: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 E4 49 00 00 02 00 C0 00 BC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
2023-10-19 19:55:52.878 T 98: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 32 00 BC 80 00 81 00 00 01 00 00 01 68 00 01 3C 00
2023-10-19 19:56:02.596 T 99: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 10 00 00 00 00 00 00 00 00 00 00 00 00
2023-10-19 19:56:02.826 T 100: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 E2 49 00 00 02 00 C0 00 BC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
2023-10-19 19:56:03.036 T 101: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 32 00 BC 80 00 81 00 00 01 00 00 01 68 00 01 3C 00
2023-10-19 19:56:03.267 T 102: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 00 00 27 9F 00 53 AE 00 00 00 00 52 46 00 26 63 80 00
2023-10-19 19:56:09.142 T 103: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
2023-10-19 19:56:09.181 T 104: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 06 D1 4A
2023-10-19 19:56:09.384 T 106: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 3B 49 00 00 FB 03 01 03 64 0A 02
2023-10-19 19:56:09.501 T 107: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
2023-10-19 19:56:09.534 T 108: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 00 00 02 4B 00 FF FF
2023-10-19 19:56:09.661 T 109: [emsesp] Me(0x0B) -R-> thermostat(0x1C), JunkersDhw(0x01D3), length: 0x19
2023-10-19 19:56:09.697 T 110: [emsesp] thermostat(0x1C) -W-> Me(0x0B), JunkersDhw(0x01D3), data: 00 00 00
2023-10-19 19:56:12.615 T 111: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 E2 49 00 00 02 00 C0 00 BC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
2023-10-19 19:56:12.825 T 112: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 32 00 BC 80 00 81 00 00 01 00 00 01 68 00 01 3C 00
2023-10-19 19:56:22.659 T 113: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 E2 49 00 00 02 00 C0 00 BC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
2023-10-19 19:56:22.900 T 114: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 32 00 BC 80 00 81 00 00 01 00 00 01 68 00 01 3C 00
2023-10-19 19:56:31.025 T 115: [emsesp] thermostat(0x1C) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00
2023-10-19 19:56:31.055 T 116: [emsesp] thermostat(0x1C) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 03 (offset 2)
2023-10-19 19:56:31.085 T 117: [emsesp] thermostat(0x1C) -W-> boiler(0x08), JunkersSetMixer(0x23), data: 00 00 00
2023-10-19 19:56:31.115 T 118: [emsesp] thermostat(0x1C) -W-> boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
2023-10-19 19:56:32.630 T 119: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 E2 49 00 00 02 00 C0 00 BC 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
```

bbqkees, 2023-10-20:
Can you put the coding CV circuit to 1?
5 is a remote heating zone, not the primary.
[image]

You don't have a IPM so thats good.

MichaelDvP, 2023-10-20:
Actually ems-esp supports only hc1-4, is there a special reason for you to configure to heatingcircuit 5? 
For hc5 we do not know the telegrams for settings, in the log only the monitor is published.
`thermostat(0x1C) -B-> All(0x00), ?(0x0173), data: 03 02 00 C8 00 CD`
Can you check in terminal `read 1C 17D` if you get a reply, this could be settings for hc5 (just a guess).

azertyfun, 2023-10-20:
It works with code 1, thanks! The language barrier wasn't helpful, I was looking at an English manual but must have missed that section.

Thanks a lot!

bbqkees, 2023-10-20:
He bought it second hand, and only has a boiler and no MM modules so he needs to set it to HK1 first.

closed 2023-10-20 by azertyfun