emsesp/EMS-ESP32 #1348: Add two new temperature entities to message 0x48F (HpTemperatures)

KarstenBr, 2023-10-21:
### PROBLEM DESCRIPTION
Up to SW version 3.6.3dev3, the message 0x48F (HpTemperatures) does not show the entities for the temperatures TA4 (temperature of the basin for condensated water) and TW1 (actual temperature of the warm water reservoir).
My heatpump (Bosch CS6800i) does provide these temperatures so that I was able to identify the respective bytes in the raw data of the message:
TA4 is a 2 byte integer value with offset 22 and a divider of 10, unit is degree Celsius
TW1 is a 2 byte integer value with offset 24 and a divider of 10, unit is degree Celsius

### REQUESTED INFORMATION
I did several read 8 0x48f and copied an example here:
003+01:31:55.592 N 7: [emsesp] boiler(0x08) -W-> Me(0x0B), HpTemperatures(0x048F), data: 01 49 01 EE 02 12 01 C9 00 00 00 00 00 6C 01 8F 01 3C 01 2E 01 3C 00 67
003+01:31:55.647 N 8: [emsesp] boiler(0x08) -W-> Me(0x0B), HpTemperatures(0x048F), data: 02 36 00 70 00 71 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
For these reads, I checked at the same time TA4 and TW1 in the heatpumpump's diagnostic screen and got these values.
For the time of above shown read example, the display showed TA4 with 10.3°C and TW1 with 56.6°C
This matches the following data of the message:
At offset 22 the data shows 00 67 which compiles to 103 (decimal)
At offset 24 the data shows 02 36 which compiles to 566 (decimal)
Both offsets were so far not processed in message 0x48F.

### EXPECTED BEHAVIOUR
SW provides also entities for TA4 and TW1 in message 0x48F

MichaelDvP, 2023-10-25:
I've added this, #1349 and #1350 to my [testbuild](https://github.com/MichaelDvP/EMS-ESP32/releases), please check is displayed correctly. I dont know what meter values in kWh are, please check/add names/translations.

KarstenBr, 2023-10-25:
Hi @MichaelDvP I checked out your testbuild. While it did show the correct temperatures TA4 and TW1, I did not recognize the data from message 0x48F at all

Furthermore I have now these issues:
- the gateway **does not recognize my WiFi network any longer** (FritzBox based Mesh) while it was properly connected all the time using 3.6.3dev2i before, so that I can only connect to it via its own acces point
- I'm **not able to downgrade to any other 3.6.3dev firmware** as it is refusing all of the now saying the file would be incompatible

So, would should I do now.

MichaelDvP, 2023-10-26:
> So seemingly there is something severly broken in 3.6.3-dev4e

Yes, using wifi-scan-all-channels blocks connections, i have disabled and go back to old connection routine. For using the strongest AP in mesh the BSSID option should do.

MichaelDvP, 2023-10-27:
>  While it did show the correct temperatures TA4 and TW1, I did not recognize the data from message 0x48F at all

Sorry, i don't understand, TA4 and TW1 are in 48F, if they show correct, what is missing? What about the data from 0x4AE and 0x4AF?

KarstenBr, 2023-10-28:
Sorry - my mistake, 0x48f was a typo. What I wanted to say was, that I do not recognize the data of 0x4AE and 0x4AF in the boiler's list of entities. Maybe I missed something while reviewing the entity list - therefore it would help me checking the list again with dev4f (or newer), if you could provide me the names you have assigned to the new entities originating from these 2 messages. (side info: both messages are being present on the EMS bus as I can see their raw data in my respectively configured custom entities window)

KarstenBr, 2023-10-28:
Ok, after checking now with dev4g, I have identified the new entities of message 0x4AE in the boiler's list: metertotal, metercomp, metereheat. While those are correctly send via MQTT incl their decimals, they show in the entity list only rounded w/o decimals - is this intended?

KarstenBr, 2023-10-28:
Also with dev4g, I have found the new entities of message 0x4AE in the boiler's list: nrgtotal, nrgww, nrgcomp. Same as in 0x4AF, those are correctly send via MQTT incl their decimals, but are also shown in the entity list only rounded w/o decimals.

KarstenBr, 2023-10-28:
Coming back to your comment of Oct 25th, where you had been asking for names/content of the new entities in 0x4AF:
metertotal
(EN) = electrical meter for overall power consumption of the heatpump
(DE) = Stromzähler (gesamt)

metercomp
(EN) = electrical meter for power consumption of the heatpump's compressor
(DE) = Stromzähler (Kompressor)

metereheat
(EN) = electrical meter for power consumption of the heatpump's add. electrical heater
(DE) = Stromzähler (elektrischer Zuheizer)

MichaelDvP, 2023-10-28:
> show in the entity list only rounded w/o decimals - is this intended?

Yes, see https://github.com/emsesp/EMS-ESP32/discussions/1268#discussioncomment-6895590

Are the energy-values the produced heating energy and the meters the electrical? e.g. `nrgTotal = copFactor * meterTotal`

KarstenBr, 2023-10-28:
The meters are electrical but the "total" meter also seem to contain the power consumption of the heatpump's electronics etc. as it increases also slightly while compressor and electrical heaters are both off. Therefore the formula in my opinion is not exactly correct for the total meter.

KarstenBr, 2023-10-28:
As for the energy values:
ngrheat is reflecting the energy used for heating = electrical energy consumed by compressor for heating + electrical energy consumed by add. heater for heating + energy retrieved from outside environment for heating
ngrww is reflecting the energy used for warmwater = electrical energy consumed by compressor for warmwater + electrical energy consumed by add. heater for warmwater + energy retrieved from outside environment for warmwater

But I noticed, that the values of nrgheat and nrgww are mixed-up in dev4g, i.e. nrgheat shows the value for warmwater and nrgww shows the value for heating

MichaelDvP, 2023-10-29:
> But I noticed that the values of nrgheat and nrgww are mixed-up in dev4g, i.e. nrgheat shows the value for warmwater and nrgww shows the value for heating

So we have to change to: nrgHeat at offset 4 and nrgWw at offset 12?

KarstenBr, 2023-10-30:
Sorry for not answering earlier. I just have tested SW 3.6.3-dev5a where you fixed nrgheat and nrgww: now both values correctly correspond to the ones shown on the heatpump’s display

proddy, 2023-11-01:
Ok closing. We'll merge into dev soon.

KarstenBr, 2023-11-01:
> Ok closing. We'll merge into dev soon.

Thx. I assume this will then include also the changes you did for #1349 and #1350 as you addressed them both also here.

proddy, 2023-11-04:
yes in https://github.com/emsesp/EMS-ESP32/releases/tag/test, if you would mind testing, and re-open if not working as expected

closed 2023-11-04 by proddy

KarstenBr, 2023-11-05:
I tested 3.6.3-dev5b and all changes done for #1348, #1349 and #1350 were functioning as expected.