emsesp/EMS-ESP32 #1351: AsyncTCP memory optimization

proddy, 2023-10-21:
@proddy Some of the tests looks to me as we have still a memory issue. The test with own mqtt task gives on one hand:
`Publish failed: low memory`, pointing to low heap < 60k, on the other hand `Stack canary watchpoint triggered (mqttclient)` the stack of 5k-Words is not enough.
I've checked the stack and found ~4.3 kWords free for main stack, but >13kWords free for AsyncTCP stack. This stack is 16kWords! Reducing to 8kWords gives 16kB more heap and is running now a day on my ems-esp (esp32 and S3). I've also added somme or the PRs found on the AsyncTCP repo. 
It's now in my dev2-testbuild. Please check. Do you see any risk in reducing the stack, or know why it is so big? 
On the S3 i've tried with mqtt publish on change, API 15sec with Thomas' ioBroker adapter, syslog enabled, 3 web connections parallel (mobile/tablet/PC). With 8k asyncTCP stack still 4,6k free:
[image]

Check on https://github.com/MichaelDvP/EMS-ESP32/releases

_Originally posted by @MichaelDvP in https://github.com/emsesp/EMS-ESP32/issues/1321#issuecomment-1773104894_

proddy, 2023-10-21:
I don't see any risk, I think it was one of the changes I made a few years back when we had so many issues with the Web stability. I started using snipets of other's peoples PRs for both the AsyncWebServer and AsyncTCP. The 8192 * 2 came from this port : https://github.com/yubox-node-org/AsyncTCPSock/tree/master 

If it works, let's bring it back to the original 8192 and stress test it on a ESP32 4MB (not S3). I'll test it myself now and if it looks good I see no reason to not put it into dev.

Those low-memory errors are worrying. I think it should be easy to reproduce, though if we mimic that user's environment. I can create a new test that simulates many devices and entities that we can run on an isolated ESP32. You need to compile with EMSESP_TEST and then invoke the test via a URL like `http://<IP>/api?device=system&cmd=test&data=test_name`. There's already a script `/scripts/run_memory_test.py` that does something similar. If I can get the ESP to crash I can use my JTAG debugger to step through the code.

MichaelDvP, 2023-10-21:
The large stack is also in the official build: https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L221
My esp32/4M is also running with the 8k stack, no issue last 3 days, but also no stress test.

proddy, 2023-10-22:
I see others have reduced the stack size. Like this project [yoradio](https://github.com/e2002/yoradio) which uses 6174 bytes. It was initially half that (https://github.com/e2002/yoradio#v06494) which I assume was not enough. The guy also has his own local tweaked copies of AsyncTCP and AsyncWebServer.

My S3 has been running fine with 8192 so far.

closed 2023-10-31 by proddy