emsesp/EMS-ESP32 #1354: MQTT Homeassistant

zibous, 2023-10-22:
### PROBLEM DESCRIPTION

Whenever the mqtt broker (docker container) is updated and restarted, the entities are not
displayed in the Homeassistant.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ x] Provide the output of http://ems-esp.local/api/system :

```lua
  {
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "000+19:09:17.014",
    "uptime (seconds)": 68957,
    "free mem": 140,
    "max alloc": 75,
    "free app": 437,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-heizung.siebler.home",
    "RSSI": -50,
    "IPv4 address": "*****",
    "IPv4 gateway": "*****",
    "IPv4 nameserver": "*****",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "europe.pool.ntp.org",
    "tz label": "Europe/Vaduz"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 22307,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 3,
    "enabled": true,
    "client id": "ems-esp32",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "ems_heizung",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": true,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 50,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": true,
    "syslog started": true,
    "syslog level": "all",
    "syslog ip": "10.1.1.96",
    "syslog queue": 1
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 5,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 78525,
    "bus reads (tx)": 20729,
    "bus writes (tx)": 4,
    "bus incomplete telegrams": 1,
    "bus reads failed": 4,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": true,
    "shower alert coldshot": 10,
    "shower alert trigger": 7,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 42
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 75,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.02",
      "entities": 50,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlers ignored": "0x23 0x1A 0x35 0xBF"
    },
    {
      "type": "controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

Stop and start exlipse MQTT Docker container

### EXPECTED BEHAVIOUR

I use the MQTT Brocker in a Docker container. 

Whenever this is updated and restarted, the entities are not 
displayed in the Homeassistant.

In this case I always have to restart the MQTT on the EMS-ESP.

Is this due to the home assistant or the EMS-ESP?
How can I fix this?

proddy, 2023-10-22:
See https://emsesp.github.io/docs/Troubleshooting/#integration-with-home-assistant-breaks-when-mqtt-broker-restarts

zibous, 2023-10-22:
@proddy 

> Integration with Home Assistant breaks when MQTT broker restarts

negativ, currenty used.

```
persistence true
persistence_location /mosquitto/data
password_file /mosquitto/data/pwfile
allow_anonymous false
listener 1883
protocol mqtt
connection_messages true
log_type error
log_dest topic
log_timestamp true
log_timestamp_format %Y-%m-%dT%H:%M:%S
log_dest file /mosquitto/log/mosquitto.log

```

All other MQTT Clients (Tasmota, Zibgee2MQTT, Services) works w/o errors.

proddy, 2023-10-22:
I read this, so the MQTT Broker restarts and the HA dashboards are empty? I will try and reproduce. Which version of HA are you using?

MichaelDvP, 2023-10-22:
@zibous You have to disable "clean session".

zibous, 2023-10-22:
@MichaelDvP 

I am not shure, but `cleanSession`is only if a mqtt bridge is used ?
```
cleanSession
	
Starts a clean session with the MQTT broker. 
Set this to false if the previous session is to be resumed. 
You should only do this in conjunction with setting the mqttClientId.
Type: bool.
Default: true.```

zibous, 2023-10-22:
> I read this, so the MQTT Broker restarts and the HA dashboards are empty? I will try and reproduce. Which version of HA are you using?

I think it is based on MQTT, not Homeassistant.

```
Dockerhub: eclipse-mosquitto:latest
Docker 20.10.23 on linux, amd64
```

proddy, 2023-10-22:
I can't reproduce. Read about clean session on https://www.emqx.com/en/blog/mqtt-session  

Disable Clean Session and try again. I'll update the wiki.

zibous, 2023-10-22:
@proddy 

Thanks, no runnig emqx instead of eclipse-mosquitto.

All works now, so the problem was eclipse-mosquitto based.
Sorry.

closed 2023-10-22 by zibous