emsesp/EMS-ESP32 #1360: MQTT autodiscovery in Domoticz not working

Hiu80, 2023-10-25:
### PROBLEM DESCRIPTION

Last week the sensors configured in my Domoticz system stopped working. After deleting and re-adding the MQTT autodiscovery plugin, the sensors are registered as switches, wich do not respond to any command en show now logging.

### REQUESTED INFORMATION

```lua

```

### TO REPRODUCE

Add plugin MQTT autodiscovery in Domoticz, and check discovered devices.

### EXPECTED BEHAVIOUR

Add the discovered devices as sensors and be able to use them.

### SCREENSHOTS

[image]
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT

Reinstalled my raspberry PI with the latest distro Bookworm. Whas running Buster. Reinstalled Domoticz and the laters version of Mosquitto.
Bought the latets version ESP module from Kees and hooked it up, replaced my older ESP module.

Hiu80, 2023-10-25:
No, I am not sure. After loads of testing and checking, I do not think there is a problem with the ESP or its firmware. I think the problem resides with Domoticz, or something in Linux at this point. I hoped you or someone else recognized this problem and could help me in the right direction.

Hiu80, 2023-10-25:
I tested with multiple firmware versions too. I opened an topic on the domiticz forums and recieved this reply after uploading some logs:

“Looks like you are missing sensor data in the ems-esp topics. So Ems is not sending the data as configured in the homeassistant config topics.”

Does this help in any way?

bbqkees, 2023-10-25:
[image]
here I do have the discovery topics.
I'll have a better look at in the coming days. There were some MQTT changes in EMS-ESP due to new specs for HA Discovery.
Maybe Domoticz cannot currently handle that.

proddy, 2023-10-26:
I can't imagine any MQTT changes affecting this. Without logs and the EMS-ESP System Info it's hard to locate what is not working.

Hiu80, 2023-10-26:
Maybe i can provide the logs and system info? I do not know wich logging you need so if tou tell me wich logging is needed i can check it out?

Bingo2023, 2023-10-27:
not sure if I understand our documentation right, but I would expect on this page a setting to enable HA or Domoticz support somewhere?
Can't find any setting for that on my EMS-ESP...

[image]

proddy, 2023-10-27:
that's a very old version. 

[image]

Bingo2023, 2023-10-27:
I'm using 3.6.1.
3.6.2 is not runing stable, after ESP resetting itself it boots up "old" Version 3.6.1 .
As we diskussed in other issues 3.6.2 and newer dev version are again very unstable.

[image]

Bingo2023, 2023-10-27:
PS see issue 1363:
https://github.com/emsesp/EMS-ESP32/issues/1363
I'm facing the same update problems. If I flash 3.6.2 twice I can boot it up but it is only runing some time until crash and then rebooting 3.6.1

proddy, 2023-10-28:
I think something is wrong with your partitions on the ESP32. I'm not sure how but it's probably booting to an older partition and picking up a previous version. I would recommend formatting/erasing the whole ESP32 (using the CLI) and flashing again using the EMS-ESP flashing tool via USB. Make sure you make a backup of your configuration files. If it's connected via USB to a computer you'll be up and running in minutes.

I also noticed from the screenshot above your max alloc memory is 49KB which is low and probably accounts for the crashes.

And that Domoticz menu option was added to version 3.6.0 back in February so really strange it still using old cached web pages.

Bingo2023, 2023-11-01:
Just a quick feedback, @proddy you were absolutely right !
After a fresh flashing&formating with "ESP-Flasher.exe" it bootet up correctly.
After importing my setting back from file, the Home Assistant setting disappeared.
So I reflashed with "ESP-Flasher.exe" and hand copied the settings. Since than is running stable.

see free memory comparison (red marked):
[image]

I wonder how much:
1.  fresh formatting with "ESP-Flasher.exe"
2. and **NOT** use old setting file (emsesp_settings.json)

could solve some of our "unstable" behavior we are facing since some month...

proddy, 2023-11-04:
thanks for reporting this back. 

The stability issues we've seen these last months were mainly related to WiFi AP reconnects (and changes in the Core espressif libraries) that I think we have all solved now.

I'm still interesting however how you managed to get an old version 3.6.1 onto the fall-back boot partition? Did it come preinstalled like this? How were you flashing the firmware before? Understanding this we can make some changes and update the Wiki articles.

RonaldRegister, 2023-11-05:
I have the same problem as topicstarter. Domoticz registers devices as switches and their status or (lack of) value is only updated once when adding the hardware to Domoticz.

I’ve done a lot of troubleshooting and couldn’t figure it out. I tried every MQTT-setting in EMS-ESP (by following the great usermanual and wiki from @bbqkees) After hours of testing, reconfiguring mqtt, settings within EMS-ESP and Domoticz I decided to downgrade EMS-ESP firmware -one version at a time- because at some point I knew it had been working. 

After downgrading to every individual version downwards from 3.6.2 I ended up with 3.5.0 and got working devices instantly. Didn’t even have to change any config settings. I don’t see a usual suspect in the changelog for 3.5.1 but from 3.5.1 and onwards I can’t seem to get it working.

So for now a working workaround for me is to stay at a downgraded 3.5.0. Maybe @Hiu80 can try and confirm this? This narrows down significantly where/when our issue has started. 

I should also mention I did a complete reset of EMS-ESP and started from scratch. As a precaution for stuck old settings. As wel as upgrading to the latest version, reset to factory defaults, and manually configure it without the use of a settings backup. Did not make any difference.

All the while I kept looking at MQTT-explorer and at first and second glance I see the same topics and messages being created. So I have no clue what is happening. Yes, I also cleared them all before re-adding a device to Domoticz. 

Sidenote, it has been a breeze being able to up/downgrade so easily (and fast!) with the benefit of retaining all settings when needed.

proddy, 2023-11-05:
@RonaldRegister, that's some good detective work. If you could grab a dump of all MQTT Discovery topics (the /config ones) I can see what the difference is and code it back into 3.6.3.

RonaldRegister, 2023-11-05:
@proddy thanks! I'll be happy to, but how should I do that? I haven't found an export function in MQTT-explorer.

Hiu80, 2023-11-05:
Hi Ronald, i will try tomorrow and let you know!

proddy, 2023-11-05:
> @proddy thanks! I'll be happy to, but how should I do that? I haven't found an export function in MQTT-explorer.

good point! You could either just take some examples that don't work and copy & paste them over

MQTT Explorer is great but development has stopped long ago. An alternative is https://mqttx.app/downloads - here you'll need to create a new connection, create a new subscription to `homeassistant/#` and then `Export Data` as JSON

RonaldRegister, 2023-11-06:
Great tip, I was looking for a replacement for MQTT Explorer, will try mqttx later today. For now I hope this will suffice?

I used the commandline mosquitto_sub to subscribe and grep all the /config ones and the line after that 9with uniq_id and name) I hope the formatting will stay intact after I press the commentbutton... 

`pi@Domoticz:~ $ mosquitto_sub -h 127.0.0.1 -p 1883 -d -t homeassistant/# |grep config -A1
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/system_status/config', ... (325 bytes))
{"uniq_id":"system_status","obj_id":"system_status","stat_t":"ems-esp/status","name":"EMS-ESP status","pl_on":"online","pl_off":"offline","stat_cla":"measurement","dev_cla":"connectivity","ent_cat":"diagnostic","dev":{"name":"EMS-ESP","sw":"v3.5.0","mf":"proddy","mdl":"EMS-ESP","cu":"http://10.10.10.112","ids":["ems-esp"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/system_ntp_status/config', ... (274 bytes))
{"uniq_id":"system_ntp_status","obj_id":"system_ntp_status","stat_t":"ems-esp/heartbeat","name":"NTP status","val_tpl":"{{value_json.ntp_status}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev_cla":"connectivity","ent_cat":"diagnostic","dev":{"ids":["ems-esp"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_heatingactive/config', ... (326 bytes))
{"uniq_id":"boiler_heating_active","obj_id":"boiler_heating_active","stat_t":"ems-esp/boiler_data","name":"Heating active","val_tpl":"{{value_json.heatingactive}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"],"name":"EMS-ESP Boiler","mf":"Bosch","mdl":"Proline","via_device":"ems-esp"}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_tapwateractive/config', ... (254 bytes))
{"uniq_id":"boiler_tapwater_active","obj_id":"boiler_tapwater_active","stat_t":"ems-esp/boiler_data","name":"Tapwater active","val_tpl":"{{value_json.tapwateractive}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_burngas/config', ... (211 bytes))
{"uniq_id":"boiler_gas","obj_id":"boiler_gas","stat_t":"ems-esp/boiler_data","name":"Gas","val_tpl":"{{value_json.burngas}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_burngas2/config', ... (236 bytes))
{"uniq_id":"boiler_gas_stage_2","obj_id":"boiler_gas_stage_2","stat_t":"ems-esp/boiler_data","name":"Gas stage 2","val_tpl":"{{value_json.burngas2}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_heatingpump/config', ... (242 bytes))
{"uniq_id":"boiler_heating_pump","obj_id":"boiler_heating_pump","stat_t":"ems-esp/boiler_data","name":"Heating pump","val_tpl":"{{value_json.heatingpump}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_fanwork/config', ... (211 bytes))
{"uniq_id":"boiler_fan","obj_id":"boiler_fan","stat_t":"ems-esp/boiler_data","name":"Fan","val_tpl":"{{value_json.fanwork}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_ignwork/config', ... (226 bytes))
{"uniq_id":"boiler_ignition","obj_id":"boiler_ignition","stat_t":"ems-esp/boiler_data","name":"Ignition","val_tpl":"{{value_json.ignwork}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_oilpreheat/config', ... (247 bytes))
{"uniq_id":"boiler_oil_preheating","obj_id":"boiler_oil_preheating","stat_t":"ems-esp/boiler_data","name":"Oil preheating","val_tpl":"{{value_json.oilpreheat}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_wwcharging/config', ... (244 bytes))
{"uniq_id":"boiler_dhw_charging","obj_id":"boiler_dhw_charging","stat_t":"ems-esp/boiler_data_ww","name":"dhw Charging","val_tpl":"{{value_json.wwcharging}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_wwrecharging/config', ... (252 bytes))
{"uniq_id":"boiler_dhw_recharging","obj_id":"boiler_dhw_recharging","stat_t":"ems-esp/boiler_data_ww","name":"dhw Recharging","val_tpl":"{{value_json.wwrecharging}}","pl_on":"on","pl_off":"off","stat_cla":"measurement","dev":{"ids":["ems-esp-boiler"]}}
Client mosqsub|26116-Domoticz received PUBLISH (d0, q0, r1, m0, 'homeassistant/binary_sensor/ems-esp/boiler_wwtempok/config', ... (260 bytes))
[... cut ...]
{"~":"ems-esp","uniq_id":"thermostat_hc1","obj_id":"thermostat_hc1","name":"Hc1","mode_stat_t":"~/thermostat_data","mode_stat_tpl":"{%if value_json.hc1.mode=='manual'%}heat{%elif value_json.hc1.mode=='day'%}heat{%elif value_json.hc1.mode=='night'%}off{%elif value_json.hc1.mode=='off'%}off{%else%}auto{%endif%}","temp_cmd_t":"~/thermostat/hc1/seltemp","temp_stat_t":"~/thermostat_data","temp_stat_tpl":"{{value_json.hc1.seltemp}}","mode_cmd_t":"~/thermostat/hc1/mode","curr_temp_t":"~/thermostat_data","curr_temp_tpl":"{{value_json.hc1.currtemp}}","min_temp":"5","max_temp":"30","temp_step":"0.5","modes":["auto","heat","off"],"dev":{"ids":["ems-esp-thermostat"]}}`

RonaldRegister, 2023-11-06:
[dump.txt](https://github.com/emsesp/EMS-ESP32/files/13264767/dump.txt)
dump.txt attached

RonaldRegister, 2023-11-06:
[dump.json](https://github.com/emsesp/EMS-ESP32/files/13265000/dump.json)
and here a proper JSON dump made with MQTTX

Hiu80, 2023-11-06:
Tried to downgrade my ESP to 3.5.0, but i have a new ESP32-S3 board. I do not know if i can load the older 3.5.0 firmware to this newer device. My older ESP is bricked since i have blown it by modifying the voltage regulator in the wrong way. So at this moment i can not confirm if MQTT starts working again if i run 3.5.0

proddy, 2023-11-06:
> Tried to downgrade my ESP to 3.5.0, but i have a new ESP32-S3 board. I do not know if i can load the older 3.5.0 firmware to this newer device. My older ESP is bricked since i have blown it by [...]

i would need to create a build of 3.5.0 for an S3.

bbqkees, 2023-11-07:
I tried putting up a new test system for Gizmocuz so he can check the actual MQTT topics and contents, however because I use a Unifi router behind my ISP router (ISP router cannot be set in bridge mode) I have to log in to the ISP router to do some additional port forwarding first. 
If I have it working I'll contact him to see what we can do about the MQTT Discovery.

proddy, 2023-12-02:
> Does 3.6.4 solve this issue? Not sure what to look for in the changelog

we haven't changed anything yet. Can you remind me what is not working? The JSON dump you sent looks fine.

RonaldRegister, 2023-12-03:
> > 
> 
> we haven't changed anything yet. Can you remind me what is not working? The JSON dump you sent looks fine.

The JSON dump is from my 3.5.0 where it is working as expected. From 3.5.1 and higher autodiscovery is not working anymore in domoticz. "the sensors are registered as switches, which do not respond to any command en show no logging"

proddy, 2023-12-03:
@RonaldRegister so you're saying only the few sensors which are created as Discovery switches are not working because Domoticz does not support this?

RonaldRegister, 2023-12-04:
@proddy no that's not at all what I'm saying, none are working from 3.5.1 and later. Domoticz works just fine when I'm on 3.5.0. Only after updating to EMS-ESP 3.5.1 and higher they stop working all together. In the openingpost of this issue by @Hiu80 you can see the screenshot. 

@bbqkees did you find the time after your holiday to do some testing?

bbqkees, 2023-12-04:
Well yes I tried it with 3.6.4 and a test install of Domoticz 2023.2 beta 15676 with my test boiler and i got like 82 entities discovered. So not sure whether this is due to EMS-ESP or the latest Domoticz version.
[image]
Not all are correct but at least it's usable again.

RonaldRegister, 2023-12-04:
well @bbqkees, thank you. I'm on Domoticz 2023.2 15453 (stable) I'm hesitant to run domoticz beta for this, but I can easily try newest EMS-ESP 3.6.4 or .5dev3 build later this week. Still would not explain why 3.5.0 is working for me right now.

bbqkees, 2023-12-04:
It was indeed broken because with my previous test a few weeks ago with 3.6
or something it was completely messed up.
But now with 3.6.4 and the latest beta at least the discovered entities are
created as Domoticz devices.

I think this needs further investigation but first we need to check what is
now working again and what is still broken before we bother Gizmocuz with
this.

Hiu80, 2023-12-06:
I have upgraded my ESP to 3.6.4 and my domoticz to version 2023.2 15697. The autodiscovery is adding sensors now, and it looks like they are updating again. I still see a lot of switches, but i can not remember if there were switches discovered before in 3.5.0. For now it looks like it is doing someting again. Will check a bit further and see if everyting works again.

RonaldRegister, 2023-12-16:
> I think this needs further investigation but first we need to check what is now working again and what is still broken before we bother Gizmocuz with this. Op ma 4 dec 2023 om 19:45 schreef RonaldRegister ***@***.***>

RonaldRegister, 2023-12-16:
just to be sure I tried 3.6.4 and 3.6.5-dev.4 on my Domoticz stable 2023.2 15453. No dice. Downgrading again as we speak.

[image]
[image]

proddy, 2023-12-16:
find out what the difference is, use MQTT Explorer to tweak the config topics and then we can code it into EMS-ESP.

bbqkees, 2023-12-19:
If someone wants to test this over several firmware versions you don't need to delete the existing discovered devices.
With the following method you can keep your existing devices and data logs in case you want to revert to an older firmware version if you are done testing other versions.
Also during testing you wont see the 'older' devices from the older versions to avoid confusion.

First use www.mqtt-explorer.com and delete all existing MQTT Discovery topics from the EMS-ESP device.
In Domoticz then temporarily disable the existing MQTT Autodiscovery hardware. Next add a new hardware of the same type (so a new MQTT Autodiscovery instance).
Set it up as required and enable.
Then do the firmware update on the Gateway and reboot. After reboot EMS-ESP will recreate the MQTT Discovery topics.
This will trigger the newly created MQTT Discovery hardware instance in Domoticz to create all the devices in Domoticz.
Now you can see the real effect of a certain EMS-ESP firmware version in Domoticz.

If you want to revert, delete the MQTT topic tree with MQTT explorer, upload the older (working) firmware on the Gateway and reboot. You can then delete or disable the last MQTT Discovery hardware and enable the older working one again.
In theory everything should work as before and you did not lose any devices or data.

LubomirG, 2023-12-20:
Ahoj.  Teplotné senzory Dalas DS18x.. nefungujú od 3.5.1 a neskôr (pre Domoticz). Domoticz funguje dobre, keď som na 3.5.0. Až po aktualizácii na EMS-ESP 3.5.1 a vyššie. Má to súvis s oddelením senzorov od zariadení? Zmenil sa výstup pre MQTT, ktorý Domoticz nesprávne interpretuje.

LubomirG, 2023-12-20:
Hi. Dallas DS18x.. temperature sensors do not work from 3.5.1 and later (for Domoticz). Domoticz works fine when I'm on 3.5.0. Only after updating to EMS-ESP 3.5.1 and above. Does it have to do with separating the sensors from the devices? Changed the output for MQTT that Domoticz interprets incorrectly.

Tigerstrength, 2023-12-24:
I too lost hours of fixing my Domoticz integrations due to the update from v3.5.0 to 3.6.4. Autodiscovery doesn't work anymore after the update. I tried every setting, but no new sensors where created. 

To compare the AD MQTT messages I looked into one sensor as an example:

AD message from 3.5.0, working perfectly. 
`{
  "uniq_id": "thermostat_hc1_dew_point_temperature",
  "obj_id": "thermostat_hc1_dew_point_temperature",
  "stat_t": "Nefit/thermostat_data",
  "name": "hc1 Dew point temperature",
  "val_tpl": "{{value_json.hc1.dewtemperature}}",
  "unit_of_meas": "°C",
  "stat_cla": "measurement",
  "dev_cla": "temperature",
  "dev": {
    "ids": [
      "ems-esp-thermostat"
    ]
  }
}`

And the failing AD message from 3.6.4. (please ignore the ids and topic prefix, as this is a test setup and I did change the names to preserve the working config in MQTT)

`{
  "uniq_id": "thermostat_hc1_dew_point_temperature",
  "obj_id": "thermostat_hc1_dew_point_temperature",
  "stat_t": "Nefit_test/thermostat_data",
  "name": "hc1 Dew point temperature",
  "unit_of_meas": "°C",
  "val_tpl": "{{value_json.hc1.dewtemperature if value_json.hc1 is defined and value_json.hc1.dewtemperature is defined else 0}}",
  "stat_cla": "measurement",
  "dev_cla": "temperature",
  "dev": {
    "ids": [
      "Nefit_test-thermostat"
    ],
    "name": "Nefit_test Thermostat"
  },
  "avty": [
    {
      "t": "Nefit_test/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "Nefit_test/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.dewtemperature is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}`

My best guess is that the extra data is mishandled by Domoticz? I also tried the Domoticz AD style, but no luck either. Something between 3.5.0. and 3.6.4. must have messed thing up. 

Merry chrismays days at all!

proddy, 2023-12-24:
I don't have Domoticz so can't test, but what I would do is fire up MQTTExplorer and edit the payload directly (on the latest 3.6.5 dev) until it works, by removing/changing parts of the config topic. Old fashioned trial and error. I would start with for example the `val_tpl` which has template code in it.

Tigerstrength, 2023-12-24:
Ok, back with some more info :) 

Upgraded Domoticz to build15760 and EMS-ESP to 3.6.5. testbuild 4. All seems to work with this combi, beside the Dallas sensors.

I found out that this discovery fails (sensor is detected by Domoticz, but seems never to get an update and because of that Domoticz doesn't create the sensor)
`"val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'] is defined and value_json['28-4010-631F-1301'].temp is defined else -55}}"`

If I manually edit this to:
` "val_tpl": "{{value_json['28-4010-631F-1301'].temp}}"`

The sensor is discovered and gets updated correctly.

Tigerstrength, 2023-12-24:
Yes, so far only the Dallas sensors are breaking. I think the work Gizomocus did on Domoticz fixed the other major issues.

Thanks for you effort @proddy!

RonaldRegister, 2023-12-27:
Thank you all for chipping in again. @bbqkees you probably saw my hardware numbering going up to 45 and 46 in my screenshots? Thank you for this better way of testing without hardware and device-id's growing exponentially within Domoticz.

I still think updating Domoticz to a beta version is a risk, and it circumvents the real issue. 

@proddy does " adjust the 'val_tpl' " also apply to my problem? Will try to have another look at tweaking the config topics soon.

proddy, 2023-12-27:
> @proddy does " adjust the 'val_tpl' " also apply to my problem? Will try to have another look at tweaking the config topics soon.

it could be Domoticz doesn't support MQTT Discovery templates, so we add that to EMS-ESP if more people confirm (and also which versions of Domoticz so we can't document this).

Hiu80, 2023-12-27:
Running dev.6 atm, with domoticz 2023.2 build 15717. Some switches work, and others do not. Example of one switch working in domoticz is "tapwater Active". MQTT explorer shows this configuration:

{
  "uniq_id": "boiler_wwtapactivated",
  "obj_id": "boiler_wwtapactivated",
  "cmd_t": "ems-esp/boiler/wwtapactivated",
  "stat_t": "ems-esp/boiler_data_ww",
  "name": "dhw Turn on/off",
  "pl_on": "on",
  "pl_off": "off",
  "val_tpl": "{{value_json.wwtapactivated if value_json.wwtapactivated is defined else 'off'}}",
  "dev": {
    "ids": [
      "ems-esp-boiler"
    ],
    "name": "ems-esp Boiler"
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "ems-esp/boiler_data_ww",
      "val_tpl": "{{'online' if value_json.wwtapactivated is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}

The switch "Heating Activated" does not work. MQTT explorer shows this:

{
  "uniq_id": "boiler_heatingactivated",
  "obj_id": "boiler_heatingactivated",
  "cmd_t": "ems-esp/boiler/heatingactivated",
  "stat_t": "ems-esp/boiler_data",
  "name": "Heating activated",
  "pl_on": "on",
  "pl_off": "off",
  "val_tpl": "{{value_json.heatingactivated if value_json.heatingactivated is defined else 'off'}}",
  "dev": {
    "ids": [
      "ems-esp-boiler"
    ],
    "name": "ems-esp Boiler"
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "ems-esp/boiler_data",
      "val_tpl": "{{'online' if value_json.heatingactivated is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all"
}

I see some differences, but my knowledge of this is not enough to figure out if something is not correct.

Hiu80, 2023-12-27:
Nevermind, got confused between "activated" and "active". Active works, activated only changes status if one de-activates or activates the function in the boiler. 

For me, it looks like everything is working again.

Tigerstrength, 2023-12-28:
> Ok, back with some more info :)
> 
> Upgraded Domoticz to build15760 and EMS-ESP to 3.6.5. testbuild 4. All seems to work with this combi, beside the Dallas sensors.
> 
> I found out that this discovery fails (sensor is detected by Domoticz, but seems never to get an update and because of that Domoticz doesn't create the sensor) `"val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'] is defined and value_json['28-4010-631F-1301'].temp is defined else -55}}"`
> 
> If I manually edit this to: `"val_tpl": "{{value_json['28-4010-631F-1301'].temp}}"`
> 
> The sensor is discovered and gets updated correctly. Maybe the '-' is causing issues with the json?

Today somehow the AD was renewed and the topic was back on `"val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'] is defined and value_json['28-4010-631F-1301'].temp is defined else -55}}"` again. The dallas sensors instantly stopped working. So it is confirmed the issue is in that topic. Changing it back to  `"val_tpl": "{{value_json['28-4010-631F-1301'].temp}}"` and the start working straight away.

proddy, 2024-01-02:
Domoticz 2024.1 was just released and the change logs shows these MQTT Discovery updates:

```
MQTT-AD, added support for climate min_temp/max_temp and temp_step
MQTT-AD, added support for number voc type
MQTT-AD, added support for Wh and Wm sensors
MQTT-AD, don't add new devices when this is disabled in the system
MQTT-AD: Fan percentage_command_template merged with preset_modes
MQTT, added option to specify devices to be published to MQTT
MQTT-AD: better handling of Push-On/Push-Off devices
MQTT-AD: better handling of Switch commands that needs a 'state' object
```

worth trying before coding around the anomalies in EMS-ESP

Tigerstrength, 2024-01-03:
Thanks for the headsup @proddy. I did the testing for you :) 

Domoticz running on 2024.1 stable and ESP-EMS on 3.6.5-test.4. 

When changing the "val_tpl" to the original long format domoticz does notify me of a new sensor (the exact samen sensor as it was before with the same ID etc.), but the sensor will never update it's value. When changing back to the short "val_tpl" value the sensor works straight away again. 

In conclusion: the behaviour is exactly the same as with the latest Domoticz Beta's I used when mentioning it in this topic. Nothing is fixed unfortunately.

proddy, 2024-01-03:
no worries, thanks a lot for testing. Shall I just set "val_tpl" to use the simple value from the MQTT payload and remove the conditional syntax (which is based on [Jinja2](https://jinja.palletsprojects.com/en/3.1.x/)) ?

Tigerstrength, 2024-01-03:
I agree that would be a good fix, but isn't it just something syntax related that causes this?

I mean, this works:
`  "val_tpl": "{{value_json.hptc1 if value_json.hptc1 is defined else 0}}"`

And this doesn't:
`  "val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'] is defined and value_json['28-4010-631F-1301'].temp is defined else -55}}"`

If it is not syntax related then I would go for your solution of the simple payload.

Tigerstrength, 2024-01-03:
I did some more testing and ruled out the '-' sign in the name as the cause. Changing it to a new sensor with '_' in the name (28_4010_631F_1301) did **not** fix the problem. Also trying different versions of the same payload did **not** work:

`"val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'] is defined and value_json['28-4010-631F-1301'].temp is defined else 0}}"`
`"val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'].temp is defined else -55}}"`
`"val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'] is defined else -55}}"`

Then I tried the conditions with other sensors in the same MQTT AD, and this works:
 "val_tpl": "{{value_json['28-4010-631F-1301'].temp if value_json.hpph1 is defined and value_json.hppl1 is defined else -55}}",

So it really seems to boil down to just the `value_json['28-4010-631F-1301'] ` part. That's is how far I can oversee it, rest is up to you knowlegeable people.

proddy, 2024-01-03:
Does

```
"{{value_json['28-4010-631F-1301'].temp if value_json['28-4010-631F-1301'].temp is defined else -55}}"
```

work? if so maybe it can't check for json objects, only values.

Tigerstrength, 2024-01-03:
Nope, that was one of the formats I had tried. I thought the same, but that is not the case.

proddy, 2024-01-03:
okay, for now I'll push a change in to ignore the condition of the val_tpl for Domoticz, and we'll see how that goes.

Tigerstrength, 2024-01-03:
Nice! Please don’t attach the solution to the Domoticz AD formatting option only. Domoticz has basic support for number types AD now. They show in as settings in the hardware menu. Therefore I and others probably too use the Homeassisant AD as the workaround seems legacy now.

proddy, 2024-01-03:
the only "domoticz" logic in the code is for handling numbers. This wasn't supported before so they were set to read-only sensors. If you're saying `number` is now working (for example `homeassistant/number/ems-esp/thermostat_hc1_seltemp/config`) I can also change that.

Tigerstrength, 2024-01-04:
Yes numbers are working in Domoticz 2024.1. Was a feature request of me back in the time, also discussed here. There is a small bug in the device names but it works perfectly. 

[image]

However, I would leave the Domoticz choice in there and mark it as specific for pre 2024.1 Domoticz versions. Reason is that the 2024.1 Domoticz lost support for ArmV6 Raspberry's, so a lot off people will be on the older Domoticz for some time.

proddy, 2024-01-04:
ok, I'll make another option for the discovery type, Homeassistant, Domoticz, Domoticz 2024

proddy, 2024-01-04:
I'll drop in a firmware here so you can test. which bin do you need (S, S3, 4MB, 8MB etc?)

Tigerstrength, 2024-01-04:
[image]

Please don’t change any mqtt id’s, I do want to lose my Domoticz devices 😇

[5 more replies]

closed 2024-01-04 by proddy