emsesp/EMS-ESP32 #1361: Setting the values of the fresh water station (with SM100) is not working correct

Cali1205, 2023-10-25:
### PROBLEM DESCRIPTION
Setting the values of the fresh water station (with SM100) is not working correct.
I observe the following behavior:

* solar/wwcirc is enable/disable the Impulse Circulation
* solar/wwcircmode is chaneging the setting of the duty cycle of the circulation

It seems that the interpretation of the Register 0x7A5 is not correct.

### EXPECTED BEHAVIOUR

I would propose the following interpretation of Register 0x7A5:

| Offset  | Description|
|----|------------------------------|
| 0  | Circulation Impulse          |
| 3  | Circulation Duty Cycle       |
| 4  | Circulation Time Program     |

#### Circulation Duty Cycle
| Value | Times 3 min/h |
|-------|---------------|
| 1     | 1             |
| 2     | 2             |
| 3     | 3             |
| 4     | 4             |
| 5     | 5             |
| 6     | 6             |
| 7     | constant      |

MichaelDvP, 2023-11-06:
I'll change the mode to duty cycle settings, Is the time program only a number? What settings are possible?

proddy, 2024-03-24:
@Cali1205  ?

proddy, 2024-05-08:
@Cali1205  ?

proddy, 2024-06-21:
closing this - no activity

closed 2024-06-21 by proddy