emsesp/EMS-ESP32 #1363: Unable to update from 3.6.1 to higher version via web-interface

kjw1985, 2023-10-27:
### PROBLEM DESCRIPTION

A while back tried to upgrade from 3.6.0 to 3.6.2. Which didn't seemed to work. Could upgrade to 3.6.1, however I'm still not able to upgrade via the web-interface. Tried multiple times the 3.6.2 version but also 3.6.3-dev2  two times. But I'm still on 3.6.1 (which seems to crash sometimes). I'm not sure why it wouldn't update as I did before. 

### REQUESTED INFORMATION

```lua
{
  "System Info": {
    "version": "3.6.1",
    "platform": "ESP32",
    "uptime": "000+00:00:21.561",
    "uptime (seconds)": 21,
    "free mem": 130,
    "max alloc": 79,
    "free app": 5,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -55,
    "IPv4 address": "192.168.1.43/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.17",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": true,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.1.17",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 48,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 36,
    "bus reads (tx)": 13,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": false,
    "max web log buffer": 50,
    "web log buffer": 17
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 21,
      "handlers received": "0x10 0x11 0x1C 0x19 0x34 0x2A",
      "handlers pending": "0xBF 0xC2 0x15 0x18 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE9"
    },
    {
      "type": "thermostat",
      "name": "RC100/Moduline 1000/1010",
      "device id": "0x18",
      "product id": 165,
      "version": "40.04",
      "entities": 0,
      "handlers pending": "0xA3 0x06 0xA2 0x12 0x13 0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0"
    },
    {
      "type": "controller",
      "name": "BC10",
      "device id": "0x09",
      "product id": 190,
      "version": "01.04",
      "entities": 0
    }
  ]
}

```

martynhr, 2023-10-31:
Same. Update to 3.6.1 ok. 
Apply 3.6.2 and shows 3.6.2 as version (also get BSSID box in wifi settings), after reboot goes back to 3.6.1 and loose BSSID box.

proddy, 2023-10-31:
> Same. Update to 3.6.1 ok. Apply 3.6.2 and shows 3.6.2 as version (also get BSSID box in wifi settings), after reboot goes back to 3.6.1 and loose BSSID box.

you must be uploading an older firmware to a different partition. How are you flashing the chip?

martynhr, 2023-10-31:
OTA. Uploaded the latest dev version and that seems to have stuck!!

proddy, 2023-10-31:
@MichaelDvP could it be the OTA is messing up the boot partition? We've seen examples where after an upload it will run the bootloader version. The current OTA code uses the older ArduinoOTAClass where our File Upload uses the espressif UpdateClass and handles partitions nicer I think. Or we could just remove the OTA feature?

martynhr, 2023-10-31:
OTA is very handy tho.

proddy, 2023-10-31:
yeah, maybe we should keep it. I can see a scenario where you automatically wget the latest binary firmware and automatically flash the EMS-ESP.

kjw1985, 2023-10-31:
Tried today flashing 3.6.1 with ESM-ESP flasher which worked. Imported the settings again and tried flashing 3.6.2. Didn't work. So flashed 3.6.2 and imported settings but got problems when connected to boiler (by jack). Seemed to be rebooting every minute or so.  Connected device with USB to laptop and could OTA update to 3.6.3-dev5a, look stable connected to with USB to laptop. However when moved and connected to boiler by jack it seemed to be rebooting every minute or so. 
Flashed 3.6.1 again as i didn't had time to troubleshoot and seems to be as "stable" as before. Which is rebooting/crashing every 1,5 or 2 weeks.

proddy, 2023-10-31:
> Tried today flashing 3.6.1 with ESM-ESP flasher which worked. Imported the settings again and tried flashing 3.6.2. Didn't work. So flashed 3.6.2 and imported settings but got problems when [...]

how are you flashing? did you ever try OTA? use a CLI like esptool, espota or use the EMS-ESP firmware uploader? Looks like your EMS-ESP is corrupted and to clean the partitions its recommended a full erase & re-flash using a standard method.

martynhr, 2023-10-31:
> Tried today flashing 3.6.1 with ESM-ESP flasher which worked. Imported the settings again and tried flashing 3.6.2. Didn't work. So flashed 3.6.2 and imported settings but got problems when [...]

Try powering by USB, mine was similar when using boiler power. 
If you don't have usb power nearby to test, try a usb power pack.
Mine's been on rock solid on usb for years now

kjw1985, 2023-11-01:
> 
> how are you flashing? did you ever try OTA? use a CLI like esptool, espota or use the EMS-ESP firmware uploader? Looks like your EMS-ESP is corrupted and to clean the partitions its recommended a full erase & re-flash using a standard method.

Until 3.5 I always flashed using OTA. To flash 3.5 I've used EMS-ESP flasher and did update by OTA  until 3.6.1. Doesn't EMS-ESP flasher erase partitions? 

> 
> Try powering by USB, mine was similar when using boiler power. If you don't have usb power nearby to test, try a usb power pack. Mine's been on rock solid on usb for years now

I has been running fine on jack before 3.5 (even up to 36 weeks, and then I had to disconnect it for yearly service on my boiler).  But looking at my grafana graphs since 3.5  4 weeks seem the maximum. (could be that I cut it short because of updating)

MichaelDvP, 2023-11-01:
> @MichaelDvP could it be the OTA is messing up the boot partition? We've seen examples where after an upload it will run the bootloader version. The current OTA code uses the older ArduinoOTAClass [...]

Last time i've used OTA is more than a year ago, but i've also seen the fallback after web-upload and second boot. I think noone is using OTA, people mention OTA, but mean web-upload. 
I think there was a beta writing garbage in the ota-data partition and the esp then always boots from ota_0 (first app-partition). USB-flash should erase and rewrite ota_data.

kjw1985, 2023-11-01:
> Last time i've used OTA is more than a year ago, but i've also seen the fallback after web-upload and second boot. I think noone is using OTA, people mention OTA, but mean web-upload. I think there [...]

I'm sorry, I'm one of those people who is confusing OTA and web-upload. I've never used the OTA, always used web-upload (or EMS-ESP flasher). I'll update my previous post.

proddy, 2023-11-05:
I'll close this and re-open if there is something structural wrong in the code. The Web and OTA uploads should work now.

closed 2023-11-05 by proddy