emsesp/EMS-ESP32 #1364: Avoid API Calls being listed as warnings within System Log

tp1de, 2023-10-29:
API calls (e.g. from ioBroker) are listed within System Log as Warnings. (EMS-ESP Version 3.6.3-dev.4)  
This should be avoided. Maybe as Debug entries, bit I believe that they are not necessary.

I had a firmware version some weeks ago from @MichaelDvP where these warnings were switched off.

proddy, 2023-10-29:
I changed this to WARN because other users were issuing commands and not knowing if there were being received and executed by EMS-ESP. There is no DEBUG anymore in EMS-ESP (unless compiled with DEBUG and only for developers, not for production).

We can use TRACE but need to think how to do that.

Also Michael's versions need to be merged back into dev soon.

tp1de, 2023-10-29:
Sorry @proddy that's very nasty, since all api calls are in warnings. I do not see anything else then api-calls within systemlog.
The same for all ioBroker users. Every Polling cycle (for me every 15 seconds) produces entries. Can't you suppress the info api calls?

2023-10-29 13:28:20.456 W 1330: [command] Calling command 'thermostat/values'
2023-10-29 13:28:20.641 W 1331: [command] Calling command 'mixer/values'
2023-10-29 13:28:35.002 W 1332: [command] Calling command 'thermostat/tempautotemp' (temporäre Solltemperatur) with value -1 and id 1 on device 0x10
2023-10-29 13:28:35.126 W 1333: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:28:35.277 W 1334: [command] Calling command 'boiler/values'
2023-10-29 13:28:35.455 W 1335: [command] Calling command 'thermostat/values'
2023-10-29 13:28:35.648 W 1336: [command] Calling command 'mixer/values'
2023-10-29 13:28:50.127 W 1337: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:28:50.281 W 1338: [command] Calling command 'boiler/values'
2023-10-29 13:28:50.459 W 1339: [command] Calling command 'thermostat/values'
2023-10-29 13:28:50.647 W 1340: [command] Calling command 'mixer/values'
2023-10-29 13:29:05.128 W 1341: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:29:05.280 W 1342: [command] Calling command 'boiler/values'
2023-10-29 13:29:05.455 W 1343: [command] Calling command 'thermostat/values'
2023-10-29 13:29:05.640 W 1344: [command] Calling command 'mixer/values'
2023-10-29 13:29:20.130 W 1345: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:29:20.280 W 1346: [command] Calling command 'boiler/values'
2023-10-29 13:29:20.460 W 1347: [command] Calling command 'thermostat/values'
2023-10-29 13:29:20.643 W 1348: [command] Calling command 'mixer/values'
2023-10-29 13:29:35.131 W 1349: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:29:35.272 W 1350: [command] Calling command 'boiler/values'
2023-10-29 13:29:35.448 W 1351: [command] Calling command 'thermostat/values'
2023-10-29 13:29:35.638 W 1352: [command] Calling command 'mixer/values'
2023-10-29 13:29:50.133 W 1353: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:29:50.283 W 1354: [command] Calling command 'boiler/values'
2023-10-29 13:29:50.460 W 1355: [command] Calling command 'thermostat/values'
2023-10-29 13:29:50.644 W 1356: [command] Calling command 'mixer/values'
2023-10-29 13:30:05.131 W 1357: [command] Calling command 'system/info' (Zeige System-Status)
2023-10-29 13:30:05.277 W 1358: [command] Calling command 'boiler/values'
2023-10-29 13:30:05.450 W 1359: [command] Calling command 'thermostat/values'
2023-10-29 13:30:05.635 W 1360: [command] Calling command 'mixer/values'

proddy, 2023-10-29:
i see, not good. I'll do a quick patch now.

tp1de, 2023-10-29:
@proddy I see that you changed api call entries from warning to info. Is there any chance not to log the pure info api calls? 

system/info and boiler/values, thermostat/values etc.

MichaelDvP, 2023-10-29:
Set the log level to noticeor below to filter out info messages. You can do this by command. 
The system/info, etc.  is a command response, not affected by log level.

tp1de, 2023-10-29:
For me it's fine to filter  ... I just thought that for most users using e.g. the ioBroker adapter, it might be good to see just see change commands and not the ones used for polling actual values.

proddy, 2023-10-29:
yes, please try it out @tp1de and let us know if we need to change it

tp1de, 2023-10-29:
For dev.5 is fine  ... I just thought that avoiding too many entries in log would be better

MichaelDvP, 2023-10-29:
> it might be good to just see change commands and not the ones used for polling actual values.

I agree, in 
https://github.com/emsesp/EMS-ESP32/blob/6a221e84cc6bac48ac2f447b0f63134a27ee16c7/src/command.cpp#L340-L348
Set the line 341 to LOG_DEBUG, these commands always give a json-response back
But leave the commands with values (lines 344, 346) as LOG_INFO.

proddy, 2023-10-30:
I'll make the edit

closed 2023-10-30 by proddy