emsesp/EMS-ESP32 #1366: Inconsistent naming

unstressable, 2023-10-29:
Theres no naming difference between:
[image]

as it should be:
[image]

proddy, 2023-10-30:
what needs to be changed?

unstressable, 2023-10-30:
I think at the screenshot above. The clarification of "selflowtemp" should not be the same as for "setflowtemp".
I think it should be as in screenshot below "gewählte Vorlauftemperatur".

proddy, 2023-10-30:
@MichaelDvP  ?

KarstenBr, 2023-10-30:
I just checked the entities of my boiler: selflowtemp is correctly named “Sollwert Vorlauftemperatur” as it is indeed the target value computed by the boiler dependent of the outside temperature and used as input for the boiler’s controller of the current flow line temperature. For setflowtemp I cannot comment as it is not provided by my boiler.

MichaelDvP, 2023-10-31:
This is the same value. `selflowtemp` is taken from UBAMonitor, the value the boiler reports, 'setflowtemp' from the `UBASettings`, what the thermostat tells the boiler. Depending on version we do not process the command telegram and do not show setflowtemp. In actual main/dev we show it: https://github.com/emsesp/EMS-ESP32/blob/df9f75a5c99a6360d3cf68b7d207f37e3558b17a/src/emsesp.cpp#L944-L945

SebastianSinger, 2023-11-04:
Does not seem to be the same value here. 
[image]
"selflowtemp" is target value of flow temperature ("Sollwert Vorlauftemperatur") and includes heating and warm water heating. 
"setflowtemp" does not include warm water heating and could be a similar value as "targetflowtemp", but shows jumps to 0° every few seconds. 
Data is taken from Buderus GB 192, EMS-ESP Version 3.6.2. 
I really appreciate this EMS-ESP project and the work you are doing.

unstressable, 2023-11-04:
thats what I meant to say. Thanks a lot for these data!
I think these 0° values show us the times, where no heating is needed or the gas is off because of hysteresis for example.

So there must be a difference in clarification between these values to make it clear.

Maybe you shouldname it like this in german:
"selflowtemp" -> "Sollwert Vorlauftemperatur Kessel"
"setflowtemp" -> "Sollwert Vorlauftemperatur Heizkreislauf"

unstressable, 2023-11-04:
another interesting aspect:
these both values are correlating:
[image]

when "setburnpow" is 0%, then "setflowtemp" is also 0° and vice versa (in my case 100% and 33°)

proddy, 2024-01-06:
@unstressable can we close this out?

unstressable, 2024-01-06:
If theres a change in latest version, yes.
If not, then no.

proddy, 2024-01-06:
what needs to be changed exactly? A PR would be very welcome.

unstressable, 2024-01-08:
Please change the german description of the datapoint "selflowtemp" to "Gewählte Vorlauftemperatur".

This should make it more clear. Thanks!

proddy, 2024-01-08:
@MichaelDvP ok?

MichaelDvP, 2024-01-08:
I don't think it matters much, we have also the feature to rename the entities indvidual.
If we change, than we should also change other languages to have different names for these:
```
MAKE_TRANSLATION(selFlowTemp, "selflowtemp", "selected flow temperature", "Sollwert Vorlauftemperatur", "Ingestelde aanvoertemperatuur", "Börvärde Flödestemperatur", "wybrana temperatura zasilania", "valgt turtemperatur", "température de flux selectionnée", "seçili akış sıcaklığı", "flusso temperatura selezionato", "zvolená teplota prívodu")
MAKE_TRANSLATION(setFlowTemp, "setflowtemp", "set flow temperature", "Sollwert Vorlauftemperatur", "Ingestelde aanvoertemperatuur", "Börvärde Flödestemperatur", "zadana temperatura zasilania", "innstilt turtemperatur", "température du flux définie", "akış sıcaklığını ayarla", "impostare temperatura di mandata", "nastavená teplota prívodu")
```
and these
```
MAKE_TRANSLATION(selBurnPow, "selburnpow", "burner selected max power", "Sollwert Brennerleistung", "Ingestelde maximale brandervermogen", "Brännare vald maxeffekt", "wybrana moc źródła ciepła", "settpunkt brennerkapasitet", "puissance max du brûleur selectionnée", "seçili kazan maksimum güç", "Setpoint potenza bruciatore", "horák zvolený maximálny výkon")
MAKE_TRANSLATION(setBurnPow, "setburnpow", "burner set power", "Sollwert Brennerleistung", "Ingesteld brandervermogen", "Värmepanna vald Effekt", "zadana moc palnika", "varmekjele valgt effekt", "puissance du brûleur définie", "kazan gücünü ayarla", "impostare potenza bruciatore", "výkon nastaveného horáka")
```
@unstressable: Please make a PR

unstressable, 2024-03-24:
What is a PR?

proddy, 2024-03-24:
> What is a PR?

It's a Git Pull Request. Basically modifying the code and submitting a change to be merged. Easiest way is via https://github.com/emsesp/EMS-ESP32/blob/dev/src/locale_translations.h to edit the translations from the browser and at the bottom of the page it'll give you a button to submit the changes as a PR

unstressable, 2024-03-24:
Done. Thx.

closed 2024-03-27 by proddy