emsesp/EMS-ESP32 #1375: Support for Relay's

proddy, 2023-10-31:
New feature to handle Relay's attached to GPIOs.

 - We can use the existing Analog Sensors 'Digital Out' with value 0 (off) or 1 (on)
 - The Analog Sensors should be renamed to something like 'Board I/O' as technically they are not all sensors. This would be a breaking change as the MQTT topic 'analogsensor' would be renamed. We can handle this in the auto-upgrade code but it will break existing HA/ioBroker connections.
 - Ensure Digital Out's are executed on EMS-ESP startup. Now it will send 0 and switch off any relays? 

@MichaelDvP I am missing anything?

MichaelDvP, 2023-11-01:
> Ensure Digital Out's are executed on EMS-ESP startup. Now it will send 0 and switch off any relays?

No, it always stores the state in filesystem and sets it on boot. Make sure the pin is output and not used by board profile.

bbqkees, 2023-11-01:
For relay boards you have active low and active high relays.
So that would need to be a setting. For active low ones sending a 0 will turn ON the relay.
And it's nice if you can give the user the option for relays to either go OFF when rebooted/power out or that they maintain their last state.

MichaelDvP, 2023-11-01:
>>  Ensure Digital Out's are executed on EMS-ESP startup. Now it will send 0 and switch off any relays?

> No, it always stores the state in filesystem and sets it on boot. Make sure the pin is output and not used by board profile.

Ups, it's stored when changing via web, it's not stored on changes via api/mqtt. Has to be fixed.

MichaelDvP, 2023-11-01:
> For relay boards you have active low and active high relays.
So that would need to be a setting. For active low ones sending a 0 will turn ON the relay.
And it's nice if you can give the user the option for relays to either go OFF when rebooted/power out or that they maintain their last state.

Something like that?
[image]

proddy, 2023-11-01:
yes, perfect. Just upper case the strings

MichaelDvP, 2023-11-03:
Ok, i have integrated the function in my test(dev2) build. Should i make a PR or do you like to test first? 
I have not changed the names for now, only set all strings with "analogsensor" to `F_(analogsensor)` to change the name quickly in local_common file. In web-i18 there are still a lot of analogsnensor related names to be translated.

hetii, 2023-11-03:
Hey :)

What about input support?

As I plan to design my own hardware I would like to use more GPIOs of esp32 for few inputs/outputs role.

About outputs, it also be nice to see some basic cronjobs syntax support, eg for stuff like enabling hot water circulation pump every day starting from 6am till 10pm for five minutes every two hours, for those ho don't plan to use HA for it.

Regards.

bbqkees, 2023-11-03:
There is already a scheduler in the firmware and several types of inputs
are also available in the Sensors section.

proddy, 2023-11-04:
> Ok, i have integrated the function in my test(dev2) build. Should i make a PR or do you like to test first? I have not changed the names for now, only set all strings with "analogsensor" to [...]

I'm sure it's great, just PR and push into dev2. Do you think we should rename analogsensors at some point?

masterx1981, 2023-11-27:
This would be very nice to have! As by now seem that  "virtually" turning on the inputs isn't working (#1388) i'll use a relay for turning on the smart grid i4 input.
The gpio output are enough for driving directly a small relay like the omron g5v-2 5dc (100ma), or the high sensitivity g5v-2 h1 5vdc (30ma)? Or in any case maybe it's better to use an npn as driver (plus eventually the flyback recirculation diode)?

MichaelDvP, 2023-11-27:
- Turning the inputs by setting invert should work, see https://github.com/emsesp/EMS-ESP32/issues/600#issuecomment-1536745466
- you can't drive a 5V releais from 3.3V SOC. Also the esp32 can sink max. 28 mA and source 20 mA per port (28/40 mA for some ports if used single),
- the 3.3 V regulator on most boards  can not source an additional relais

masterx1981, 2023-11-27:
Whops, forgot the 3.3v thing, and also there aren't a lot of 3.3v relais, in any case would be nice to have this feature, there are out a lot of ready made relays modules with integrated driver...
The problem that i see using the hp1inopt swap bit mode, is that my heat pump hp1~hp4 in opt never change, whatever is the state of the phisical inputs. Or the hp1inopt input is the "logic" of the input (active high/active low) of the signal? This evening i'll check this out. Sorry for the off topic...

bbqkees, 2023-11-27:
@masterx1981 You can already switch physical relays with EMS-ESP.
Via Add Analog Sensor -> type Output.
We just want to make it more logical and and more convenient to use in the web interface.

If you want to make use of a relay currently the best way to do it is to use a ready made board with optocoupler inputs and an external power supply for the relay board. (Preferably with higher quality relays like Omron or Panasonic.)
This way it cannot really interfere on the EMS bus.
In your case you just want to switch a low-voltage input/dry contact of the heat pump, that is relatively safe to do.
But others try to switch heavy loads on mains power. If not done properly this can be dangerous for the boiler/heat pump and the person itself.

For some safety guidelines check here: https://bbqkees-electronics.nl/wiki/gateway/using-gateway-gpio.html

proddy, 2024-02-03:
3.6.5 support this and I've updated the documentaton.

suaveolent, 2024-10-08:
@proddy i habe been playing around wither the feature yesterday. It works great so far. 
However, it only works for Active High for me, which unfortunately has the relay state inverted. 

If I set it to Active-Low the relay does not switch at all.
Am i doing anything wrong here?

proddy, 2024-10-08:
Could be how the offset is handled in the code. I'll see if I can reproduce. Can you tell me
 - which EMS-ESP version you are using
 - which GPIO pin is the relay controlling

I also noticed some fixes needed in the web UI, so I'll do those too!

suaveolent, 2024-10-08:
Thanks, the issue occurred on 3.6.5 as well as latest 3.7.0. 

Im using a Lolin S3 and pins 47/48. (it is a 2 way relay module). 

I could reproduce it, however, on other pins too.

proddy, 2024-10-09:
yeah, I saw that. Its on the fix list for my next PR (ft_dashboard branch)

proddy, 2024-10-09:
and line 759 with the `command_setvalue()` ?

proddy, 2024-10-10:
fixed in ft_dashboard branch, probably 3.7.0-dev.44

closed 2024-10-10 by proddy

suaveolent, 2024-10-14:
@proddy thanks for the fast fix. The relays now also switch when set to "Active low". Unfortunately, they now behave the same as "Active high".

MichaelDvP, 2024-10-14:
Ups, Active low is stored as factor -1 in web (not 0 as i thought), so the switch should be:
`digitalWrite(sensor.gpio(), (sensor.offset() == 0) ^ (sensor.factor() > 0));`

proddy, 2024-10-14:
Fixed in 3.7.0-dev.45

suaveolent, 2024-10-14:
Thanks, its working now! 🚀