emsesp/EMS-ESP32 #1378: Problems with central RC30 and two RC20 as remote

Megapit, 2023-11-01:
### PROBLEM DESCRIPTION

I have one RC30 on the central heater and two RC20 as remotes in each heating circuit (i have two HK1 and HK2 with a mm10)
The entities in rc30 looks good but the RC20 strange. 

For example in RC20 of HK1 have a "HK1 Sollwert Raumtemperatur" of 0°. The same value in the central RC30 is 21°
The second RC20 of HK2 shows values from HK1
I will attach  CSV from each device

The main problem is, in Homeassistant (connected over mqtt) i get the wrong "HK1 Sollwert Raumtemperatur". It looks like RC20/HK1 overrides the values from RC30. I can deal with the other issues.

### REQUESTED INFORMATION

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/13228577/emsesp_info.txt)
[Kessel_GB125_GB135_MC10.csv](https://github.com/emsesp/EMS-ESP32/files/13228602/Kessel_GB125_GB135_MC10.csv)
[Thermostat_RC30.csv](https://github.com/emsesp/EMS-ESP32/files/13228603/Thermostat_RC30.csv)
[Thermostat_ES72_RC20-HK1.csv](https://github.com/emsesp/EMS-ESP32/files/13228605/Thermostat_ES72_RC20-HK1.csv)
[Thermostat_ES72_RC20-HK2.csv](https://github.com/emsesp/EMS-ESP32/files/13228608/Thermostat_ES72_RC20-HK2.csv)
[Mischermodul_MM10.csv](https://github.com/emsesp/EMS-ESP32/files/13228609/Mischermodul_MM10.csv)
[mqtt-node.txt](https://github.com/emsesp/EMS-ESP32/files/13228658/mqtt-node.txt)

MichaelDvP, 2023-11-01:
Could you please make a log-all of more than a minute to see the AE/AD telegrams. I think i know what's going wrong, but need proof in the telegrams.

Megapit, 2023-11-01:
Your wish is my command :)

Feel free to write if you need more infos

[log.txt](https://github.com/emsesp/EMS-ESP32/files/13230900/log.txt)

MichaelDvP, 2023-11-02:
Please test with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
You will get only one entity for the remote thermostats.
If the RC20 is used as remote (device-id 0x18, 0x19, ...) it publshes only a single 0xAF telegram containing the remote room temp. As master (device-id 0x17) it controls only a single heating circuit hc1 with telegrams 0xAE for measurements and 0xAD for settings. I did not expect that remote thermostat reply to 0xAE query, but it replys with a empty telegram (`thermostat(0x18) -W-> Me(0x0B), RC20Monitor(0xAE), data: 00 00 00 00 BA 00 00 00`), causing 0 for set-temperature, etc., with only room temp. I've removed the 0xAE, 0xAD queries for the remote now and read only 0xAF, conaining the remote room temp. The setting values are all stored in the master thermostat RC30, no need to show them also for the remote.

Megapit, 2023-11-02:
Thanks for your fast response. This is working for me. 
I'm a software developer myself but your description with the protokoll is to much for me :) I see it as a black box which should work.

Anyway i added the current state of my heating.

[Kessel_GB125_GB135_MC10.csv](https://github.com/emsesp/EMS-ESP32/files/13236019/Kessel_GB125_GB135_MC10.csv)
[Thermostat_RC30.csv](https://github.com/emsesp/EMS-ESP32/files/13236020/Thermostat_RC30.csv)
[Thermostat_ES72_RC20 HC1.csv](https://github.com/emsesp/EMS-ESP32/files/13236021/Thermostat_ES72_RC20.HC1.csv)
[Thermostat_ES72_RC20 HC2.csv](https://github.com/emsesp/EMS-ESP32/files/13236022/Thermostat_ES72_RC20.HC2.csv)
[Mischermodul_MM10.csv](https://github.com/emsesp/EMS-ESP32/files/13236023/Mischermodul_MM10.csv)
[Thermostat_Data.json](https://github.com/emsesp/EMS-ESP32/files/13236024/Thermostat_Data.json)
[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/13236059/log.1.txt)

closed 2023-11-02 by Megapit

MichaelDvP, 2023-11-02:
Thanks for the infos. I'll make a PR to official version. 

I've seen in your info.txt for the second RC20 `"handlers ignored": "0x35 0xA6 0x01"` (telegrams send by the device, but not handled by emsesp). I've never seen a 0xA6 telegram before, also the first RC20 have not sent it. I'm curious. 
Could you send another info.txt after a few hours uptime and maybe check in terminal `read 19 A6` and `read 18 A6`  what the thermostats respond.

Megapit, 2023-11-02:
OK!
Log is switched on. Will send it in a few hours. Max buffer size of 100 enought?

Console feedback is done:
ems-esp:$ read 19 A6
000+02:01:27.928 N 1: [emsesp] thermostat(0x19) -W-> Me(0x0B), ?(0xA6), data: 00 00 02 00 02 02
ems-esp:$ read 18 A6
000+02:01:49.400 N 2: [emsesp] thermostat(0x18) -W-> Me(0x0B), ?(0xA6), data: 00 00 03 00 01 02

MichaelDvP, 2023-11-02:
Log is not really needed, the two reads are enough (look like some status-bits).  But if you already have it, please attach.
The system->up/download->support info after uptime of a few hours will show me if there are unknown telegrams send rarly.

BTW: Log buffer should be kept small in normal operation to save ram, As long as you stay on the system-log page the log only uses your browser memory and is not limited, Emsesp-log-buffer is for background buffering if you leave log-page, e.g. change a setting in dashboard and then go to log-page you can see the buffered change. But mostly it's easier to open a browser window with log page and do the setting in a second browser window, no need for big buffer.

Megapit, 2023-11-02:
If you need a log over some hours then call. I accenedently closed the browser. So i have to start from the bedinning.

Here is the support info
[emsesp_info.json](https://github.com/emsesp/EMS-ESP32/files/13237797/emsesp_info.json)

MichaelDvP, 2023-11-02:
> If you need a log over some hours then call.

No, thanks, Info shows we have only 0xAF, 0x35 and 0xA6 from the remote, So no possible issue for the remote emulation discussed in #1287.