emsesp/EMS-ESP32 #1379: Buderus FS27/3E identified as Solar Module

philipherbert, 2023-11-02:
### PROBLEM DESCRIPTION
WarmwaterStation/FreshwaterStation is also identified as SolarModule.

When   combined with Boiler and SM200 for Solar,
the Warmwater Values and Solar Values/Settings are mixed in the MQTT Topic solar_data.

This is very confusing.

[emsesp_info (1).json](https://github.com/emsesp/EMS-ESP32/files/13239675/emsesp_info.1.json)

MichaelDvP, 2023-11-03:
>This is very confusing.

Yes, Bosch uses modules for different tasks, dhw could be managed by a solar module (with id 0x2A) or a mixer module (id 0x28-29). We also had complains about module-naming of the mixer-dhw function.

@proddy : What do you think about creating a water module class and move the dhw specific from mixer and solar to this? Because there are a lot of entities, also add a `publish_time_water`. In consequence we also should move the mixer function `pool` to a new pool-device, but use `publish_time_others`. For boiler_water we already have a own mqtt topic, but it's very difficult to split the device, i think keep the entities in boiler.

philipherbert, 2023-11-03:
Is functionality not also defined by the dip setting ?

Regarding entities. I have not yet reached the point, which of the entities are actually working.
Some dhw entities of the boiler seem valid, some of this module.

Buderus identifies a Warmwater (Freshwater) station after initial BUS Scan.

MichaelDvP, 2023-11-03:
> Is functionality not also defined by the dip setting ?

Yes, ems-esp started with ems-modules, they do not have dip-switches. The module is defined by product-id, this does not change with dip setting. For the newer ems+ with dip we have refactor the whole bus management or use a workaround: If the dip changes the device-id we can remap after detection. We already do this for boilers as heatsource (cascaded) and boiler/controller with same produce-id.

proddy, 2023-11-04:
> @proddy : What do you think about creating a water module class and move the dhw specific from mixer and solar to this? Because there are a lot of entities, also add a publish_time_water. In [...]

@MichaelDvP yes I think this is needed. It's not a small change, and possibly a breaking one for those that have integrations (in HA, ioBroker, nodered...). So something for a next major 3.7.0 release? And should we also take this opportunity to move some of the boiler_ww dhw over too? The only concern is how to show this in the WebUI since technically this water module is not an EMS device?

MichaelDvP, 2023-11-04:
@philipherbert Please check build dev6a https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test 
and send the emsesp_info and some screenshots, entity exports to see if it works.

philipherbert, 2023-11-04:
after loading: EMS Tx Errors: syslog shows loads of
Datum,Zeit,Stufe,Hostname,Kategorie,Programm,Nachrichten
2023-11-04,15:55:50,Debug,192.168.121.229,kern,1,2023-11-04T15:55:49.845238+1:00 ems-esp emsesp - - - 000+00:06:00.068 D 68: double json key: collectorshutdown
2023-11-04,15:55:49,Debug,192.168.121.229,kern,1,2023-11-04T15:55:49.845054+1:00 ems-esp emsesp - - - 000+00:06:00.068 D 67: double json key: cylheated

and also at startup: (or before shutdown ?
2023-11-04,15:49:54,Info,192.168.121.229,kern,1,"2023-11-04T15:49:53.508884+1:00 ems-esp emsesp - - - 000+00:00:03.744 I 4: WiFi connected with IP=192.168.121.229, hostname=ems-esp"
2023-11-04,15:49:53,Info,192.168.121.229,kern,1,2023-11-04T15:49:53.507561+1:00 ems-esp emsesp - - - 000+00:00:03.744 I 3: Starting NTP service
2023-11-04,15:49:53,Info,192.168.121.229,kern,1,- ems-esp emsesp - - - 000+00:00:00.000 I 2: Loaded EMS device library (117 records)
2023-11-04,15:49:53,Info,192.168.121.229,daemon,1,- ems-esp analogsensor - - - 000+00:00:00.000 I 1: Starting Analog sensor service
2023-11-04,15:49:53,Info,192.168.121.229,daemon,1,- ems-esp temperaturesensor - - - 000+00:00:00.000 I 0: Starting Temperature sensor service
2023-11-04,15:49:48,Warnung,192.168.121.229,console,1,2023-11-04T15:49:48.496653+1:00 ems-esp telegram - - - 007+22:38:43.399 W 114: Incomplete Rx: AA 00 2A 00 88 00 08 00
2023-11-04,15:49:48,Warnung,192.168.121.229,console,1,2023-11-04T15:49:48.172639+1:00 ems-esp telegram - - - 007+22:38:43.062 W 113: Incomplete Rx: 88 00 08 00
2023-11-04,15:49:48,Warnung,192.168.121.229,console,1,2023-11-04T15:49:48.030178+1:00 ems-esp telegram - - - 007+22:38:42.928 W 112: Incomplete Rx: 97 00 88 00 08 00 88
2023-11-04,15:49:47,Warnung,192.168.121.229,console,1,2023-11-04T15:49:47.894088+1:00 ems-esp telegram - - - 007+22:38:42.794 W 111: Incomplete Rx: 88 00 08 00 94 00
2023-11-04,15:49:47,Warnung,192.168.121.229,console,1,2023-11-04T15:49:47.552504+1:00 ems-esp telegram - - - 007+22:38:42.452 W 110: Incomplete Rx: 88 00 08 00 88 00 08 00
2023-11-04,15:49:47,Warnung,192.168.121.229,console,1,2023-11-04T15:49:47.424994+1:00 ems-esp telegram - - - 007+22:38:42.3
[emsesp_info (2).json](https://github.com/emsesp/EMS-ESP32/files/13256981/emsesp_info.2.json)
16 W 109: Incomplete Rx: 88 00 08 00

philipherbert, 2023-11-04:
watermodule not in mqtt

MichaelDvP, 2023-11-04:
> double json key: collectorshutdown
> double json key: cylheated

Yes, they were double, i'll fix, any other doublettes?
> watermodule not in mqtt

Hmm, Info shows it with 18 entites, is it shown in web? Do you get response to api `ems-esp.local/api/water`, telnet `show values`?

MichaelDvP, 2023-11-04:
Found the typo, mqtt should work now in dev6b.

philipherbert, 2023-11-04:
api/water works

 "wwc3": {
        "wwtemp1": 21.7,
        "wwtemp2": 0,
        "wwtemp3": 21.8,
        "wwtemp4": 22.9,
        "wwtemp5": 45.3,
        "wwtemp7": 21.7,
        "wwpump": false,

telnet:
Water Module: SM100/MS100 (DeviceID:0x2A, ProductID:163, Version:27.09) (18)
  wwc3 temperature 1 (wwtemp1): 21.7
  wwc3 temperature 2 (wwtemp2): 0.0
  wwc3 temperature 3 (wwtemp3): 21.8
  wwc3 temperature 4 (wwtemp4): 22.9
  wwc3 temperature 5 (wwtemp5): 45.3
  wwc3 temperature 7 (wwtemp7): 21.7
  wwc3 pump (wwpump): off
  wwc3 maximum temperature (wwmaxtemp): 60
  wwc3 selected temperature (wwseltemp): 52
  wwc3 reduced temperature (wwredtemp): 45
  wwc3 daily temperature (wwdailytemp): 60
  wwc3 disinfection temperature (wwdisinfectiontem: 65
  wwc3 circulation active (wwcirc): off
  wwc3 circulation pump mode (wwcircmode): own prog
  wwc3 keep warm (wwkeepwarm): off
  wwc3 status 2 (wwstatus2): no heat
  wwc3 pump modulation (wwpumpmod): 0
  wwc3 flow rate (wwflow): 0.0

        "wwmaxtemp": 60,
        "wwseltemp": 52,
        "wwredtemp": 45,
        "wwdailytemp": 60,
        "wwdisinfectiontemp": 65,
        "wwcirc": false,
        "wwcircmode": "own prog",
        "wwkeepwarm": false,
        "wwstatus2": "no heat",
        "wwpumpmod": 0,
        "wwflow": 0
    }
}

philipherbert, 2023-11-04:
Working now also in MQTT. what does wwc3 stand for ?

I believe these errors at startup are new:
<html>
<body>

EMS-Telegramme gelesen (Tx) | 16 | 11 | 60%
-- | -- | -- | --

</body>
</html>

MichaelDvP, 2023-11-04:
There is no change in bus management. The above log shows some incomplete before restart, maybe while uploading the firmware. But these are RX fails and now you mention TX. Can you add a log to see what causes the TX fails? Do the fails increase constantly or do you have only some on startup?

philipherbert, 2023-11-04:
I see those only at startup and not in any logs.
Before running the .dev firmware or the old GW (not S3) or before connecting EM100 or SM 200 the error counter where always 0.

How can I get those into a log after reboot.
They are not in syslog

The RX errors are during firmware update before reboot

MichaelDvP, 2023-11-04:
The only count of tx-fails without log message is a queue overflow (have not found any other). I've added log messages for that. (not counted up version, still dev6b). But we have a 50-queue and your count is 16-success, 11-fail. What is the count of Tx-writes?

philipherbert, 2023-11-04:
TX-writes was 0.
[image]

After firmware upload, the error counter is now only 1 and one message in log
2023-11-04 18:29:56.766 W 18: [telegram] Tx queue overflow, skip one message

philipherbert, 2023-11-04:
regarding the temperature entities
wwc3 temperature 1 (wwtemp1): 21.7
wwc3 temperature 2 (wwtemp2): 0.0
wwc3 temperature 3 (wwtemp3): 21.8
wwc3 temperature 4 (wwtemp4): 22.9
wwc3 temperature 5 (wwtemp5): 45.3
wwc3 temperature 7 (wwtemp7): 21.7

Not all of those temperatures are actually from sensors connected to the water module.

What are the other values, are these reeceived from other modules and echoed from this module ?
(warmwater is currently off, upper boiler sensor is 45.3 C, this sensor is connected to the boiler), 
Would it be possible to give these temperature entities better names ? 

The module has 4 connections for temperature sensors (manual FS27/3E) 
TS17 warmwater output 
TS21 supply temperature from cylinder (sensor in the station)
TS22 (not yet connected) -> cylinder sensor for valve return flow 
TS23 (cold side supply water/but after circulation)

Valve PS5 (controlled via TS22) entity missing ? Or will it show up when connected ?
Output Terminal supports also an error relay (with option to invert)

philipherbert, 2023-11-04:
another question: is the boiler setting circulation pump mode supposed to control the circulation pump on the FS27/3. ? 
"dhw circulation pump mode", or is this setting missing on the water module. 

FS27/3E has similar  settings for Circulation as the boiler has.
- Timecontrolled yes/no
- permanent or own time program
- 1x3min .. 6x3min per hour or continuous
- fixed 3 minutes triggered by short time water flow (2-10 Seconds) Option only on/off

To me, it is not clear, which settings(entities) are relevant from the boiler in this setup

MichaelDvP, 2023-11-05:
> What are the other values, are these reeceived from other modules and echoed from this module ?

I dont know, all values are published on the bus by this module. And we have only the values the users report. Also the names are reported. See #164, #212, #250
If you like to add values and settings, try to catch the telegrams/offsets.

closed 2024-03-24 by proddy