emsesp/EMS-ESP32 #1382: Energy/power entities need to have "diagnostic" as category set in HA to avoid warnings

proddy, 2023-11-04:
In recent updates of HA, any Energy entities must have the category "ent_cat" set to "diagnostic" and not "config". We have two in EMS-ESP in the Boiler device called `nrgheat` (Energy Heating) and `nrgww` (dhw Energy dhw). Both have a command associated to reset the counters making them automatically set to "config".

The change is in `Mqtt::publish_ha_sensor_config()` to ignore the commands if the UOM is KWH.

tp1de, 2023-11-05:
As far as I understood the entity category is not needed. It works for energy states as well without.

proddy, 2023-11-05:
It is required for HA now. The owner of Home Assistant had contacted us asking us to include it.

tp1de, 2023-11-05:
Thanks for info. I will include  in my own ha entities as well.

proddy, 2023-11-05:
Paulus (HA creator) emailed @bbqkees and asked if we could support this. "A configuration entity provides a setting/configuration option for the device, while a diagnostic entity provides information about the device.". So when devices are added by Discovery all the config sections are consistent.

codetrembler, 2023-11-05:
After upgrading the EMS-Gateway to 3.6.3-dev5b and Home Assistant 2023.11.1 with HA Auto-Dicovery there are still some values missing for me:

sensor.thermostat_datetime
sensor.thermostat_hc1_holidays
sensor.thermostat_hc1_switchtime1
sensor.thermostat_hc1_switchtime2
sensor.thermostat_hc1_vacations
sensor.thermostat_wwholidays
sensor.thermostat_wwswitchtime
sensor.thermostat_wwvacations
sensor.thermostat_wwcircswitchtime

sensor.boiler_maintenancedate

sensor.ems_esp_shower_duration
sensor.ems_esp_shower_timestamp

Could you please check if the category for these entities has to be adapted too,
Since I only have one HC I would think that the HC2-values are also missing but cannot check this with my systems.

Thank you very much and keep up the good work!

tp1de, 2023-11-05:
sensor.thermostat_datetime is now text.thermostat_datetime for me. I do not have witchtimes or holidays. Maybe they are text as well?

proddy, 2023-11-05:
yes, as @tp1de mentions these are no longer sensors because they are writeable (with associated commands), so now become 'text' entities in HA. You will need to delete the old ones from the MQTT Broker. If you use MQTT Explorer and search for example 'wwswitchtime' you should find the new ones. Note this is a breaking change in 3.6.3 that need to be documented.

Also 5b is Michael's build, and not merged with the official dev yet, so there may be differences.

closed 2023-11-05 by proddy