emsesp/EMS-ESP32 #1385: Extend meaning of hpactivity in 0x4A2 by a further operational mode for "evaporator defrosting"

KarstenBr, 2023-11-05:
### PROBLEM DESCRIPTION
With SW 3.6.3-dev5b there are 4 operational modes (1,2,3,4) so far for the entity "hpactivity" in message 0x4A2 (offset 7).
By comparing the actual operation mode of my heatpump, I have recognized, that there is also mode 6 which is set while the heatpump performs defrosting of the evaporator.

### EXPECTED BEHAVIOUR
There should be a new entity added for "evaporator defrosting" which is set to true when hpactivity value is 6.

MichaelDvP, 2023-11-05:
I have never understood why we need for the heatpump activity extra states for each activity. We already have the `hpActivity` with enum states including the defrost. 
```
MAKE_ENUM(enum_hpactivity, FL_(none), FL_(heating), FL_(cooling), FL_(hot_water), FL_(pool), FL_(unknown), FL_(defrost))
```
Why add a completly redundant boolstate? I suggest to remove these redundants
```
    has_update(hpHeatingOn_, hpActivity_ == 1 ? 0xFF : 0);
    has_update(hpCoolingOn_, hpActivity_ == 2 ? 0xFF : 0);
    has_update(hpWwOn_, hpActivity_ == 3 ? 0xFF : 0);
    has_update(hpPoolOn_, hpActivity_ == 4 ? 0xFF : 0);
```
and save the memory space for new entites.

proddy, 2023-11-05:
yes agree. this is cleaner. and memory is precious.

KarstenBr, 2023-11-05:
This kind of cleanup is indeed even better

KarstenBr, 2023-11-05:
But in this case I would propose to change the german name of hpactivity: instead of "Kompressoraktivität" it should be considered to rename it to "Kompressor-Betriebsmodus" as this better describes its content

proddy, 2023-11-19:
in 3.6.3

closed 2023-11-19 by proddy