emsesp/EMS-ESP32 #1386: Add new entity for 4-way-valve in message 0x4A2

KarstenBr, 2023-11-05:
### PROBLEM DESCRIPTION
With SW 3.6.3-dev5b the message 0x4A2 (HpInputs) does only contain input states but this message also contains the heatpump's output states. So far I have decoded one of these outputs: the first byte of the message (offset 0) contains the switch state of the 4-way-valve (but do not mix this up with a 3-way-valve used for switching between heating and ww). The 4-way-valve is being used to change the operational mode of the refrigerant circuit, which can be either set for heating & warmwater (value 1) or for cooling & evaporator defrosting (value 0). The valve's state is provided in the 4th bit of the first byte (offset 0) in message 0x4A2.

### REQUESTED INFORMATION
Read command for message 0x4A2 while the 4-way-valve is in position for "heating & ww":
[emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: **9**0 00 00 02 00 00 00 00

Read command for message 0x4A2 while the 4-way-valve is in position for "cooling & defrosting":
[emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: **1**0 00 00 02 00 00 00 00

### EXPECTED BEHAVIOUR
Add a new entity "hp_4-way-valve" with two logical states:
- 0 = "cooling & defrost" (EN) / "Kühlen & Abtauen" (DE)
- 1 = "heating & ww" (EN) / "Heizen & Warmwasserbereitung" (DE)

If 4th bit in the first byte (offset 0) in message 0x4A2 is 0, 4-way-valve shall be "cooling & defrost".
If 4th bit in the first byte (offset 0) in message 0x4A2 is 1, 4-way-valve shall be "heating & ww".

proddy, 2023-11-05:
Thanks, any suggestions for the shortname too? We have for example now `hp3wayValve` and `ww3wayValve` as Booleans.

KarstenBr, 2023-11-05:
I would propose either hpvr4 (Bosch technical manual calls this valve "VR4" of which I assume the "R" is meant for refrigerant) or hprefr4wvalve

MichaelDvP, 2023-11-05:
We have already the hp3way, if we name it hp4way we can use the translations. We can add `(VR4)` in the texts. What do you think of:
```
MAKE_TRANSLATION(hp3wayValve, "hp3way", "3-way valve", "3-Wege-Ventil", "3-weg klep", "", "zawór 3-drogowy pompy ciepła", "3-veisventil", "", "3 yollu vana", "valvola 3-vie") // TODO translate
MAKE_TRANSLATION(hp4wayValve, "hp4way", "4-way valve (VR4)", "4-Wege-Ventil (VR4)", "4-weg klep (VR4)", " (VR4)", "zawór 4-drogowy pompy ciepła (VR4)", "4-veisventil (VR4)", "", "4 yollu vana (VR4)", "valvola 4-vie (VR4)") // TODO translate
```

proddy, 2023-11-05:
that works

KarstenBr, 2023-11-05:
Fine for me as well

KarstenBr, 2023-11-07:
@MichaelDvP I have checked SW 3.6.3-dev6c in which you had integrated hp4way. Unfortunately it seems the logical states displayed for this new entity are mixed up as  cooling/defrosting is shown by the gateway while in reality the valves switch state is heating/ww.

KarstenBr, 2023-11-08:
Thx for fixing the enum for hp4way. I did check again with the lastest dev6c incl. this fix and confirm that it is now working as expected.

proddy, 2023-11-19:
in 3.6.3

closed 2023-11-19 by proddy