emsesp/EMS-ESP32 #1387: Missing Values and naming for FreshwaterStation connected to MS100 dip setting3

philipherbert, 2023-11-05:
I suggest a new issue for naming / missing value issues, to keep possibly relating changes documented in one thread. 
follow up for issues discussed in #1379.

Will start with some values for the freshwaterstation from diagnostic and config page.

(not yet capured:  hot water circulation pump settings, start manual desinfecion now, enable keep hot, temp diff for return valve, error display(relay settings)

I hope watching 0x2A captures the needed data:
If i can caputure more specific, please let me know.

wwc3 temperature7/1 are identical(?) is called TS17 Hot Water temperature, disapper when disconnecting the sensor
wwc3 temperature3 is called TS21  buffer cylinder temperature
wwc3 temperature4 is called cold Water Temperature

diagnostic shows
Set hot water temperature 45 (probably the setpoint if station would be on)
TS17 29.0
TS21 33,2 - 33,5 changed during trace
ColdWaterTenperature 23 
WWM Volume Flow 0
PS11 0% primar pump modulation
PS13 0% DHW circulation pump 0%

ems-esp:$
000+16:05:55.114 N 34: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:06:11.909 N 35: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5), length: 0x19
000+16:06:11.950 N 36: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 03 FF 1E 32
000+16:06:13.859 N 37: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x19
000+16:06:13.946 N 38: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 3C 3C 34 2D 12 41 01 3C 1E 03 07 3C 00 08 00 05
000+16:06:14.452 N 39: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE), length: 0x19
000+16:06:14.481 N 40: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07AE), data: 00 0F 0F
000+16:06:14.754 N 41: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E0), length: 0x19
000+16:06:14.830 N 42: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:06:25.389 N 43: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:06:25.597 N 44: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 03 00 03 00 2D 00 08
000+16:06:26.045 N 45: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 64 00 00 03
000+16:06:26.246 N 46: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+16:06:26.721 N 47: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 2B 00 00 01 4C 00 E6 01 C5 80 00 01 2B 80 00
000+16:06:26.959 N 48: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:06:30.506 N 49: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 00 00 00
ems-esp:$

after chaning system config to add VS5 the return valve
referred to as 2ab configuration in the manual 
the Return temperatur 39.4 is shown (referd to TS22 in the manual, connected to TS3
return valve (VS5 a relay contact: 100%)
ems-esp:$
000+16:11:55.023 N 140: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:12:00.309 N 141: [emsesp] water(0x2A) -B-> All(0x00), ErrorMessage(0xBF),                                          data: 2A A3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+16:12:10.266 N 142: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5),                                          length: 0x19
000+16:12:10.308 N 143: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5),                                          data: 00 03 0A 03 FF 1E 32
000+16:12:10.578 N 144: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6)                                         , length: 0x19
000+16:12:10.665 N 145: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6)                                         , data: FF 05 0F 5F 19 01 3C 3C 3C 34 2D 12 41 01 3C 1E 03 07 3C 00 08 00 05
000+16:12:10.937 N 146: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07                                         AE), length: 0x19
000+16:12:10.967 N 147: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07                                         AE), data: 00 0F 0F
000+16:12:11.237 N 148: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E                                         0), length: 0x19
000+16:12:11.317 N 149: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwStatus2(0x07E                                         0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:12:24.932 N 150: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:12:25.142 N 151: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07A                                         A), data: 00 02 03 00 03 00 2D 00 08
000+16:12:25.586 N 152: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 00                                          64 01 04
000+16:12:25.791 N 153: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01                                          80 00 01 00 00 00 00
000+16:12:26.265 N 154: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(                                         0x07D6), data: 01 2B 00 00 01 40 00 E6 01 C5 80 00 01 2B 01 8A
000+16:12:26.503 N 155: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07                                         E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:12:30.753 N 156: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwComma                                         nd(0x07AB), data: 00 00 00
000+16:12:54.994 N 157: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:13:10.251 N 158: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5), length: 0x19
000+16:13:10.290 N 159: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 03 FF 1E 32
000+16:13:10.562 N 160: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x19
000+16:13:10.650 N 161: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 3C 3C 34 2D 12 41 01 3C 1E 03 07 3C 00 08 00 05
000+16:13:10.902 N 162: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE), length: 0x19
[... 154 lines cut ...]
000+16:32:24.322 N 496: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:32:24.533 N 497: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 03 00 03 00 2D 00 08
000+16:32:24.975 N 498: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 00 64 01 04
000+16:32:25.181 N 499: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+16:32:25.656 N 500: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 21 00 00 01 2A 00 E6 01 C5 80 00 01 21 01 8A
000+16:32:25.896 N 501: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:32:30.520 N 502: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 00 00 00

change to start time: 03:15
00+16:32:54.394 N 503: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:33:16.011 N 504: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5), length: 0x19
000+16:33:16.075 N 505: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 03 FF 1E 32
000+16:33:16.344 N 506: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x19
000+16:33:16.430 N 507: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 3B 3C 34 2D 12 43 01 3C 1E 03 07 3C 00 08 00 05
000+16:33:16.682 N 508: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE), length: 0x19
000+16:33:16.729 N 509: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07AE), data: 00 0F 0F
000+16:33:17.003 N 510: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E0), length: 0x19
000+16:33:17.076 N 511: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:33:24.534 N 512: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+16:33:24.742 N 513: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 03 00 03 00 2D 00 08
000+16:33:25.186 N 514: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 00 64 01 04
000+16:33:25.390 N 515: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+16:33:25.864 N 516: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 21 00 00 01 28 00 E6 01 C5 80 00 01 21 01 8A
000+16:33:26.103 N 517: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+16:33:30.736 N 518: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 00 00 00

philipherbert, 2023-11-05:
to document here: circulation  active is not working in ems-esp in this setup. EMES ESP always shows off state.
Diagnostic page shows PS13 DHW circulation pump 0% or 100% it is connected to the PS1 Terminal of the module. (relay)

The main pump (heating supply) is conneted to mains and PWM Output for Modulation,
modulation is correct in ems esp (identical to boiler)

It seems as if the circulation pump is called wwc pump in ems esp.

flow rate in ems-esp is wrong:
Boiler shows 9.8 /ems esp 2.5 -2.7

000+17:16:36.340 N 1398: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 19 (offset 9)
000+17:16:36.952 N 1399: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: FF F6 (offset 1)
000+17:16:53.086 N 1400: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+17:17:10.860 N 1401: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5), length: 0x19
000+17:17:10.899 N 1402: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 03 00 1E 32
000+17:17:11.172 N 1403: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x19
000+17:17:11.258 N 1404: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 3B 3C 34 2D 12 43 01 3C 1E 03 07 3C 00 08 00 05
000+17:17:11.532 N 1405: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE), length: 0x19
000+17:17:11.564 N 1406: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07AE), data: 00 0F 0F
000+17:17:11.831 N 1407: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E0), length: 0x19
000+17:17:11.911 N 1408: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwStatus2(0x07E0), data: 00 FF F6 00 00 00 1A 19 06 19 00 00 00 00 00 03 03 03 03
000+17:17:23.293 N 1409: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+17:17:23.499 N 1410: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 03 00 03 00 2D 01 08
000+17:17:23.944 N 1411: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 64 00 00 03
000+17:17:24.147 N 1412: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+17:17:24.625 N 1413: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 BE 00 62 02 16 00 95 00 95 80 00 01 C2 01 86
000+17:17:24.864 N 1414: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 FF F6 00 00 00 1A 19 06 19 00 00 00 00 00 03 03 03 03

now boiler 4.4 ems-esp: 1.2
000+17:18:10.725 N 1420: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 03 00 1E 32
000+17:18:10.997 N 1421: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x19
000+17:18:11.084 N 1422: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 3B 3C 34 2D 12 43 01 3C 1E 03 07 3C 00 08 00 05
000+17:18:11.356 N 1423: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE), length: 0x19
000+17:18:11.391 N 1424: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07AE), data: 00 0F 0F
000+17:18:11.657 N 1425: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E0), length: 0x19
000+17:18:11.732 N 1426: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 13 13 06 13 00 00 00 00 00 03 03 03 03
000+17:18:11.889 N 1427: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 00 43 (offset 2)
000+17:18:14.404 N 1428: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 0D 0D (offset 6)
000+17:18:14.590 N 1429: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 0D (offset 9)
000+17:18:15.055 N 1430: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 00 2C (offset 2)
000+17:18:17.431 N 1431: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: FF F6 (offset 1)
000+17:18:23.288 N 1432: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+17:18:23.496 N 1433: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 03 00 03 00 2D 01 08
000+17:18:23.944 N 1434: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 64 00 00 03
000+17:18:24.145 N 1435: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+17:18:24.620 N 1436: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 D9 00 2C 02 16 00 94 00 94 80 00 01 C2 01 86
000+17:18:24.860 N 1437: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 FF F6 00 00 00 0D 0C 06 0C 00 00 00 00 00 03 03 03 03
000+17:18:30.609 N 1438: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 00 00 01
000+17:18:38.798 N 1439: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 A4
000+17:18:44.270 N 1440: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 (offset 1)

philipherbert, 2023-11-05:
DHW Mode on thermostat:

EMS-ESP BC400 Thermostat calls the 'reduced' temperature on freshwater station: 'normal'

Comfort is Comfort
When setting to Auto on Boiler Display: EMS-ESP Shows Own Prog
When setting to Auto in EMS-ESP: EMS-ESP changes to OwnProg

Deinfection **start** time on thermostat: is called desinfection time in ems-esp (confusing) and is not a time setting but rather a duration setting.

DHW circulation pump mode on thermostat:
Auto is called DHW (according to hot waterprogram)

MichaelDvP, 2023-11-08:
I think you always log to late, for changing paramters we expect a telegram like `thermostat -w-> Water, data: xy (offset z)`
changing a single value. But this is not in the logs.
> WWM Volume Flow 0
PS11 0% primar pump modulation
PS13 0% DHW circulation pump 0%

zero values are impossible to find, there are too many zeros in the telegram. We need changing values to be sure.
> return valve (VS5 a relay contact: 100%)

could be second byte in `water(0x2A) -B-> All(0x00), ?(0x07AD), data: 00 64 01 04`
Could you use custom entitiy to proof. device:2A, type: 7AD, offset 1, UINT, uom:percent
> ww3 selected tempeature is called comfort temperature

In other Bosch brands/versions it is called selected temperature, so it's always wrong in some cases, you need to use the customzation to remane it for your likes.
> ww3 extra hot water temperature is missing in emsesp now set to 59 (before 60)

I'll add it.
> display now shows Daily/Weekly: TUE
Start Time: 02:00
changed start time to 03:30

I think clock settings are always stored in the thermostat.We need to log all or watch thermostat while changing the values.
> low rate in ems-esp is wrong:
Boiler shows 9.8 /ems esp 2.5 -2.7

Looks like temperature_2  is the flow rate, please verify.
> When setting to Auto on Boiler Display: EMS-ESP Shows Own Prog
When setting to Auto in EMS-ESP: EMS-ESP changes to OwnProg

Is this a thermostat setting or water module?

philipherbert, 2023-11-09:
for verification, I have captured more data:

vs5 valve seems ok 0x2a, 0x7ad, Add1, uint, Faktor 1, %

thermal desinfect OK
missing temp- not yet captured- extra hot water 59

comments are between the trace lines,

flow first 27.9 change to 26.4

ems-esp:$ watch 0x2a
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x2A
000+00:53:10.207 N 1: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5), l                                                                                                                                                             ength: 0x19
000+00:53:10.239 N 2: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), d                                                                                                                                                             ata: 00 03 0A 03 FF 1E 32
000+00:53:10.512 N 3: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6),                                                                                                                                                              length: 0x19
000+00:53:10.600 N 4: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6),                                                                                                                                                              data: FF 05 0F 5F 19 01 3C 3B 3C 34 2D 12 43 01 3C 1E 03 07 3C 00 08 00 05
000+00:53:10.756 N 5: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+00:53:10.944 N 6: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0                                                                                                                                                             ), data: 53 55 (offset 6)
000+00:53:11.230 N 7: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE                                                                                                                                                             ), length: 0x19
000+00:53:11.278 N 8: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07AE                                                                                                                                                             ), data: 00 0F 0F
000+00:53:11.602 N 9: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0                                                                                                                                                             ), data: 55 (offset 9)
000+00:53:11.907 N 10: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E0                                                                                                                                                             ), length: 0x19
000+00:53:11.982 N 11: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwStatus2(0x07E0                                                                                                                                                             ), data: 00 00 00 00 00 02 53 55 06 55 00 00 00 00 00 03 03 03 03
000+00:53:12.792 N 12: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwTemperature(0x                                                                                                                                                             07D6), length: 0x02 (offset 8)
000+00:53:12.837 N 13: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwTemperature(0x                                                                                                                                                             07D6), data: 00 8B (offset 8)
000+00:53:12.991 N 14: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E                                                                                                                                                             0), data: 00 0A (offset 1)
000+00:53:13.890 N 15: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6),                                                                                                                                                              length: 0x01 (offset 12)
000+00:53:13.895 N 16: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6),                                                                                                                                                              data: 43 (offset 12)
000+00:53:14.186 N 17: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwTemperature(0x                                                                                                                                                             07D6), length: 0x01 (offset 3)
000+00:53:14.210 N 18: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwTemperature(0x                                                                                                                                                             07D6), data: 01 17 (offset 2)
000+00:53:14.484 N 19: [emsesp] Me(0x0B) -R-> water(0x2A), ?(0x07AD), length: 0x                                                                                                                                                             01 (offset 1)
000+00:53:14.489 N 20: [emsesp] water(0x2A) -W-> Me(0x0B), ?(0x07AD), data: 00 (                                                                                                                                                             offset 1)
[... cut ...]
000+00:59:40.563 N 171: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 04 00 03 00 2D 01 08
000+00:59:41.007 N 172: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AD), data: 64 00 00 03
000+00:59:41.210 N 173: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+00:59:41.687 N 174: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 AB 00 00 01 C7 00 BC 00 8E 80 00 01 C2 01 73
000+00:59:41.923 N 175: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03

philipherbert, 2023-11-14:
I have captured a few more settings:
- temp diff return valve K
- setting Circulation Pump Pulse-Control (Yes/No)
- setting TimeControlled Circulation Pump yes/no
- error display (alarm relay contact on the module) Off, Normal, Inverted
- daily heatup setting Off/On (not changed time, was set to 3:15)

temp diff return valve now set to 7K
000+00:04:48.439 N 35: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07AC), data: 07 (offset 1)

toggle setting Circulation Pump Pule-Control
On
ems-esp:$
000+00:07:45.181 N 120: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+00:07:45.390 N 121: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 04 00 03 00 2D 01 08
000+00:07:46.069 N 122: [emsesp] water(0x2A) -B-> All(0x00), SM100ValveStatus(0x07AD), data: 00 64 01 04
000+00:07:46.273 N 123: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+00:07:46.751 N 124: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 CF 00 00 01 D7 01 33 01 C2 80 00 01 C2 01 8E
000+00:07:46.989 N 125: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+00:07:47.541 N 126: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCirc(0x07A5), data: FF
000+00:07:47.892 N 127: [emsesp] water(0x2A) -B-> All(0x00), ?(0xF7), data: FF 06 A5 1F
000+00:07:48.076 N 128: [emsesp] water(0x2A) -B-> All(0x00), SM100wwCirc(0x07A5), data: FF
ems-esp:$
ems-esp:$
ems-esp:$
ems-esp:$
ems-esp:$
Off
000+00:07:51.413 N 129: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCirc(0x07A5), data: 00
000+00:07:51.763 N 130: [emsesp] water(0x2A) -B-> All(0x00), ?(0xF7), data: FF 06 A5 19
000+00:07:52.200 N 131: [emsesp] water(0x2A) -B-> All(0x00), SM100wwCirc(0x07A5), data: 00
ems-esp:$

toggle Time-Controlled-Circulation Pump yes/no

000+00:09:40.794 N 170: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCirc(0x07A5), data: 00 (offset 4)
000+00:09:41.092 N 171: [emsesp] water(0x2A) -B-> All(0x00), ?(0xF7), data: FF 06 A5 11
000+00:09:41.785 N 172: [emsesp] water(0x2A) -B-> All(0x00), SM100wwCirc(0x07A5), data: 00 (offset 4)
000+00:09:43.320 N 173: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 00 00 01
000+00:09:45.891 N 174: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCirc(0x07A5), data: FF (offset 4)
ems-esp:$
000+00:09:46.186 N 175: [emsesp] water(0x2A) -B-> All(0x00), ?(0xF7), data: FF 06 A5 19
ems-esp:$
000+00:09:46.393 N 176: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 00 02 03 00 03 00 2D 01 08
000+00:09:46.886 N 177: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07A9), data: 2F
000+00:09:47.079 N 178: [emsesp] water(0x2A) -B-> All(0x00), SM100ValveStatus(0x07AD), data: 00 64 01 04
000+00:09:47.587 N 179: [emsesp] water(0x2A) -B-> All(0x00), ?(0x07B1), data: 01 80 00 01 00 00 00 00
000+00:09:47.811 N 180: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 01 B5 00 00 01 BC 01 2E 01 C2 80 00 01 C2 01 8E
000+00:09:48.352 N 181: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+00:09:48.538 N 182: [emsesp] water(0x2A) -B-> All(0x00), SM100wwCirc(0x07A5), data: FF (offset 4)
000+00:09:50.330 N 183: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 01 00 01
000+00:09:50.711 N 184: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 05 (offset 2)
000+00:09:51.650 N 185: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus(0x07AA), data: 64

changesd Error display from Off to Normal to Inverted  to Off

ems-esp:$
000+00:11:18.170 N 262: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: FF 9C (offset 1)
000+00:11:18.475 N 263: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x01 (offset 12)
000+00:11:18.504 N 264: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: 43 (offset 12)
000+00:11:18.770 N 265: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwTemperature(0x07D6), length: 0x02 (offset 2)
000+00:11:18.795 N 266: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwTemperature(0x07D6), data: 00 91 (offset 2)
000+00:11:19.067 N 267: [emsesp] Me(0x0B) -R-> water(0x2A), SM100ValveStatus(0x07AD), length: 0x01 (offset 1)
000+00:11:19.090 N 268: [emsesp] water(0x2A) -W-> Me(0x0B), SM100ValveStatus(0x07AD), data: 64 (offset 1)
000+00:11:24.275 N 269: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: FF 92 (offset 1)
ems-esp:$
ems-esp:$
ems-esp:$
ems-esp:$
ems-esp:$
[... 47 lines cut ...]
changed daily heatup from off to enable (3:15 was then visible) and back to off
daily temperature 60 is already in the water module

000+00:13:20.283 N 351: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwCirc(0x07A5), length: 0x19
000+00:13:20.323 N 352: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 03 FF 1E 32
000+00:13:20.596 N 353: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x19
000+00:13:20.683 N 354: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 3B 3C 34 2D 12 43 01 3C 1E 03 07 3C 00 08 00 05
000+00:13:20.955 N 355: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwKeepWarm(0x07AE), length: 0x19
000+00:13:20.983 N 356: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwKeepWarm(0x07AE), data: 00 0F 0F
000+00:13:21.256 N 357: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwStatus2(0x07E0), length: 0x19
[... cut ...]
000+00:13:29.853 N 368: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 01 00 01
000+00:13:31.048 N 369: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 0E (offset 9)
000+00:13:34.332 N 370: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 00 00 (offset 2)
000+00:13:34.525 N 371: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 00 03 (offset 5)
000+00:13:34.962 N 372: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 (offset 9)

proddy, 2023-11-14:
you could test some of these yourself using the Custom Entities to see if it works. For example for Circulation Pump control use:

Device ID 0x2A, Type ID 0x7A5, offset 0, Value Type BOOL

with a few screenshots of these it'll be easier for Michael and myself to code

MichaelDvP, 2023-11-15:
I've added the values to my dev2, but can not find:
> hanged daily heatup from off to enable (3:15 was then visible) and back to off
daily temperature 60 is already in the water module

Maybe this is a thermostat setting and i think you have only watched the water module. Changing a setting is always done on thermostat, so watching the thermostat will always catch it. For sensor values you have to watch the module that broadcast the value. The thermostat contains the clock of the system, so clock related entities are often stored in the thermostat, also if they set by another module.

philipherbert, 2023-11-15:
excellent. For testing time / thermostat I will possibly need more time.
tested. the version downloaded 'test' Tag, I hope this is correct)

Setting for cicrulation pulse mode still missing, but working as custom entity as described by proddy.

[image]

setting time controlled cirucaltion on/off missing on MS100
error display working (also write) and  in sync with boiler display/setting
return valve does not seem to be writable. (there is no setting on boiler UI and ems esp setting(write) is ignored)
what is ciculation active setting?  has no effect.

I now use 'watch 0x10 0x2a'

other comments / test inline:
changed ciculation mod to permanent vi boiler display

000+00:08:02.431 N 1: [emsesp] thermostat(0x10) -R-> water(0x2A), ?(0xF9), length: 0x11
000+00:08:02.500 N 2: [emsesp] water(0x2A) -W-> thermostat(0x10), ?(0xF9), data: FF 06 A5 03 17 00 07 00 01 00 00 00 04 00 00 00 06 00 00 00 03
000+00:08:03.004 N 3: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0291), length: 0x19
000+00:08:03.040 N 4: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0291), data: 00 00 00 00 00 00
000+00:08:03.311 N 5: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0292), length: 0x19
000+00:08:03.347 N 6: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0292), data: 00 00 00 00 00 00
000+00:08:03.617 N 7: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0293), length: 0x19
000+00:08:03.653 N 8: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0293), data: 00 00 00 00 00 00
000+00:08:03.924 N 9: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0294), length: 0x19
000+00:08:03.960 N 10: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0294), data: 00 00 00 00 00 00
000+00:08:04.231 N 11: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode(0x02F5), length: 0x19
000+00:08:04.301 N 12: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 03 00 01 02 00 00 0E 04 00 0D 04 00 00 00 00 00 00 00 00 00 01
000+00:08:04.572 N 13: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWtemp(0x031B), length: 0x19
000+00:08:04.594 N 14: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWtemp(0x031B), data: <empty>
000+00:08:04.866 N 15: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300OutdoorTemp(0x023A), length: 0x19
000+00:08:04.897 N 16: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 00 14
000+00:08:04.959 N 17: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCirc(0x07A5), data: 07 (offset 3)
000+00:08:05.202 N 18: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x19
000+00:08:05.279 N 19: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: FF 00 05 00 00 00 00 00 FF 02 F8 06 FF FF 00 14 00 32 00 00 00 FF 00 01
000+00:08:05.736 N 20: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x19 (offset 24)
000+00:08:05.740 N 21: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: <empty> (offset 24)
000+00:08:07.986 N 22: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Curves(0x029B), length: 0x01 (offset 1)
000+00:08:08.009 N 23: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Curves(0x029B), data: 01 (offset 1)
000+00:08:08.296 N 24: [emsesp] Me(0x0B) -R-> thermostat(0x10), HybridSettings(0xBB), length: 0x1B
000+00:08:08.300 N 25: [emsesp] thermostat(0x10) -W-> Me(0x0B), HybridSettings(0xBB), data: <empty>
000+00:08:09.152 N 26: [emsesp] Me(0x0B) -R-> thermostat(0x10), PVSettings(0x023E), length: 0x19
000+00:08:09.179 N 27: [emsesp] thermostat(0x10) -W-> Me(0x0B), PVSettings(0x023E), data: <empty>

changed via ems esp does not work
000+00:09:44.829 I 28: [command] Calling command 'water/wwcircmode' (circulation pump mode) with value continuous and id 11 on device 0x2A
000+00:09:44.829 E 29: [command] Command 'wwcircmode:continuous' failed with code: 3
000+00:09:44.829 E 30: [emsesp] Write command failed callback function failed (Error)
000+00:09:53.734 N 31: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0xE7), data: 00 02 00
000+00:09:53.787 N 32: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 00 00 00 00 00 00 01 06 5C 07
000+00:09:53.972 N 33: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Floordry(0x0267), data: 00 00
000+00:09:54.150 N 34: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xC4), data: 00
000+00:09:54.760 N 35: [emsesp] thermostat(0x10) -W-> water(0x2A), SM100wwCommand(0x07AB), data: 01 00 01
ems-esp:$

circulation status not shown / not correct in ems esp: (is percent display on boiler status page, but connected to a relay output)

changed from 100 % to 0 % (relay in module clicked), this is a relay output on the MS100
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x2A
000+00:12:19.542 N 122: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwParam(0x07A6), length: 0x01 (offset 12)
000+00:12:19.567 N 123: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwParam(0x07A6), data: 43 (offset 12)
000+00:12:19.838 N 124: [emsesp] Me(0x0B) -R-> water(0x2A), SM100wwTemperature(0x07D6), length: 0x02 (offset 2)
000+00:12:19.869 N 125: [emsesp] water(0x2A) -W-> Me(0x0B), SM100wwTemperature(0x07D6), data: 00 91 (offset 2)
000+00:12:20.148 N 126: [emsesp] Me(0x0B) -R-> water(0x2A), SM100ValveStatus(0x07AD), length: 0x01 (offset 1)
000+00:12:20.228 N 127: [emsesp] water(0x2A) -W-> Me(0x0B), SM100ValveStatus(0x07AD), data: 64 (offset 1)
[... cut ...]
000+00:17:55.403 N 287: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 00 00 (offset 2)
000+00:17:55.839 N 288: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 (offset 6)
000+00:17:56.024 N 289: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 03 (offset 8)
000+00:17:56.211 N 290: [emsesp] water(0x2A) -B-> All(0x00), SM100wwStatus2(0x07E0), data: 00 00 (offset 1)
000+00:17:57.414 N 291: [emsesp] water(0x2A) -B-> All(0x00), SM100wwTemperature(0x07D6), data: 00 00 (offset 2)

adding a screenshot of the water module settings:
[image]

MichaelDvP, 2023-11-15:
> Setting for cicrulation pulse mode still missing

This is `circ active` from earlier solar-dhw settings,  have not changed the name.
> I now use 'watch 0x10 0x2a'

you can only watch on id, second is ignored.
returnValve is fixed now. CircTimecontr. shouldshow up now.

circulation
>  changed from 100 % to 0 % (relay in module clicked), 

could be 7E0, offset 5 changing 0xFF(on) to 0x00(off)`?
Please verify with custom entitiy.

philipherbert, 2023-11-16:
>>could be 7E0, offset 5 changing 0xFF(on) to 0x00(off)`?
no, changes to true when tap is opened, but has nothing to do with the circ pump.

Any Idea what else it could be ?
Can I trace something regarding setting circulation pump mode.
UI shows request failed.
If I change this stting from the boiler UI, EMS-ESP follows the correct setting.
It seems as if only write is causing issues.

closed 2024-03-24 by proddy