emsesp/EMS-ESP32 #1388: Input states from message 0x4A2 not visible in entity list although being correctly received from boiler

KarstenBr, 2023-11-05:
### PROBLEM DESCRIPTION
I have recognized in SW 3.6.3-dev5b, that the states of the heatpump's inputs (named hpInput1...hpInput4 in boiler.cpp) are not being displayed in the list of boilere entities, but only the inputs' configurations (hpin1opt...hpin4opt). Although message 0x4A2 is defined in boiler.cpp with these signals (named hpInput[0].state...hpInput[4].state), the respective code for registering the related entities is disabled as shown in the examplary screenshot:

[image]

### REQUESTED INFORMATION
Reading of message 0x4A2 via console shows correct reception of the message:
ems-esp:$ read 8 0x4a2
000+02:33:26.313 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: 92 00 00 02 00 00 00 00

### EXPECTED BEHAVIOUR
Enable the entities hpInput1, hpInput2, hpInput3, hpInput4 in the code so that they become visible in the list of boiler entities.

KarstenBr, 2023-11-11:
When testing SW 3.6.3-dev6c this week, I have recognized that the beforehand missing 4 hpinput entities have been activated again - thx. Therefore I also checked their proper function but have seen that they do not correctly reflect the physical state of the heatpump’s input ports:

hpinput2 is always reported „on“ in SW 3.6.3-dev6c while I did toggle the respective input 2 (inhibit WW/heating) between off and on; in the raw data of HpInput this change is visible in the second bit of the first byte (offset 0):
input 2 pins electrically closed (i.e. activated):
[emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: 92 00 00 02 00 00 00 00
input 2 pins electrically open (i.e. deactivated):
[emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: 90 00 00 02 00 00 00 00

hpinput4 is always reported „off“ in SW 3.6.3-dev6c while the respective input 4 (PV surplus charging of heating reservoir) is toggled between off and on; in the raw data of HpInput the change of input4 is visible in the lowest bit of the fifth byte (offset 4):
input 4 pins electrically closed (i.e. activated):
[emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: 90 00 00 02 01 00 00 00
input 4 pins electrically open (i.e. deactivated):
[emsesp] boiler(0x08) -W-> Me(0x0B), HpInput(0x04A2), data: 90 00 00 02 00 00 00 00

(hint: I have not been able to also test inputs 1 (inhibit compressor) and 3 (external alarm) - so they might be also incorrect)

MichaelDvP, 2023-11-12:
Ah, there was a reason for not publishing the states, It's mostly unclear, If i check #600 the second bit in offset 0 shows input 1, for you input 2.
I think it may depend on the input configuration and 0x04A2 shows not the input number, but the input function. 
I will remove the inputs again until this is clear.

KarstenBr, 2023-11-12:
ok, then I would propose to instead use logical/function input states as entities:
so, instead of input2 (in my case), 2nd bit of 1st byte is „inhibit warmwater“, which is my configuration is for (my) input2
and instead of input4 (in my case), lowest bit of 5th byte is „charge with photovoltaic surplus power“, which is the configuration of (my) input 4

MichaelDvP, 2023-11-13:
Can you make some tests to verify this guess. E.g. watch 0x04A2 and change the config of input 1 to different functions and see what changes, Also check same function on different input, check if 0x04A2 gives same result.

KarstenBr, 2023-11-13:
ok, I will look into this during the next weekend

KarstenBr, 2023-11-18:
@MichaelDvP I did now some experiments with the input configuration:
- independent what had been configured for input2, the electrical switching of this port did always change the same bit in 0x4A2 (2nd bit of first byte):
In2 closed/active --> 92 ...
In2 open/passive --> 90 ...
- Input1 and Input4 are linked into one status as those inputs can also be used for the so called SmartGrid-Interface of the heatpump; this combined state is located in the lower 2bits of the 4th byte of 0x4A2:
00b =
photovoltaic surplus charging of buffer up to configured max temperature **(in case SG is not configured)**
||
enforced increase of room temperature set value and WW set value and buffer charging up to configured max temperature **(if SG is activated)**
01b = disabling of heating - compressor and add. electrical heater - enforced by grid operator (independent of SG setting)
10b = normal operation (independent of SG setting)
11b = increase of room temperature set value and WW set value by the parameterized delta **(only used if SG is activated)**
1st example while In1 was electrically open and In4 was electrically closed: 03 00 00 **01** 00 00 00 00
2nd example while In1 was electrically open and In4 was electrically open: 82 00 00 **02** 00 00 00 00
3rd example (with SG activated) while In1 was electrically closed and In4 was electrically open: 82 00 00 **03** 00 00 00 00

Bingo2023, 2023-11-22:
kleiner Hinweis:
ich habe bei mir physikalisch nichts an den Pins angeschlossen, ich nutze aber den Trick das erste Bit zu setzen, was die restlichen Bits invertiert. Dadurch kann ich die hinteren Bits per EMS-ESP setzen und die WP erkenn die gesetzten Bits.
Das sehe ich in dem Menü des BC400 Displays.

Allerdings ist mir nicht klar was diese Bits an Reaktion auslösen sollen, ich konnte da keine Änderung feststellen. Habe aber auch nur auf das PV bit geschaut. Da ich keine weiten Buderus Geräte angeschlossen habe meckert die Anlage wenn ich sage PV wäre verbaut....

closed 2024-03-24 by proddy