emsesp/EMS-ESP32 #1389: Add new entity: dhw Mode switch on entities | HC delay modes

MichScha, 2023-11-05:
### PROBLEM DESCRIPTION

RC300 / WLW196i v3.6.2

Just working on finetuning and automate our heatpump.
I could find the switch off temperatures for the three different modes eco, eco+ and komfort but not the on temperatures.
Is this due to the fact that it can be controlled by dhw selected lower temperature?

It is also possible to set delays for the Heating Circuit. I could not find them too but it is already a known telegram within RC300Summer2.

```lua
Wassertemperatur Komfort
Einschalttemperatur: 49
000+00:32:18.140 N 453: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBAParameterWWPlus(0xEA), data: 31 (offset 6)

Wassertemperatur Eco
Einschalttemperatur: 43
000+00:37:23.190 N 562: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBAParameterWWPlus(0xEA), data: 2B (offset 18)

Wassertemperatur Eco+
Einschalttemperatur: 36
000+00:38:35.480 N 602: [emsesp] boiler(0x08) -B-> All(0x00), UBAParameterWWPlus(0xEA), data: 24 (offset 27)
---------------------------------
Heizkreis 1
Grenzwert. Sofortstart Heizen: 2k (Range: 1-10)
000+00:59:24.133 N 1369: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Summer2(0x0471), data: 02 (offset 4)

Ausschaltverzögerung für den Heizbetrieb: 2h (Range: 1-48)
000+01:04:27.021 N 1765: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Summer2(0x0471), data: 02 (offset 3)

Einschaltverzögerung für den Heizbetrieb: 4h (Range: 1-48)
000+01:05:46.495 N 1856: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Summer2(0x0471), data: 04 (offset 2)
```

Thanks for your effort in this great project :-)

MichaelDvP, 2023-11-06:
dhw comfort/eco is already included as wwSelTemp, wwSelTempLow. I'll add the eco+ and the thermostat parameters.

Bosch often uses different names in different brands and changes the names of values with software version. It's impossible to track every name and to store all the possibilities in ems-esp. For that we have introduced the customization.
(For dhw comfort/eco see #250, there it is called "Warmwasser" and " Warmwasser reduziert".)

proddy, 2024-05-08:
I think we can close this @MichScha  ?

MichScha, 2024-06-03:
Yes, thanks!

closed 2024-06-03 by MichScha