emsesp/EMS-ESP32 #1394: hybridSystem MQTT Monitoring in HA

FloCodeRian, 2023-11-06:
### PROBLEM DESCRIPTION

Hi there, I'm new to this project. 
Started with integration of my heatpump and was very happy about the integration. That's why decided to hook up my Gasboiler, too. Unfortunately I experienced some trouble now. Would love, if someone can help me.

As read in the manual, for hybrid systems of Buderus, separate ems-esp modules because of two separate EMS-Busses are need. 

The EMS-Moduls work great but somewho the MQTT integration in Homeassistant mixes those two. 

I kept patience for a different ID, an changed it for more variety, but didn't helped.

heatpump ID: "ems-esp"
Gasboiler ID: was... "ems-esp-gas" changed to "mems-esp-gas" with no advantages.
MQTT discovery used... 

problems: 
#not all entities are shown in Homeassistant
#expected 2 boards for entities with wifi strengh etc. - only one shown
[image]

#gasboiler not showing properly, a lot of entities missing. 

gasboiler payload : 

Received 21:23:34
QoS: 0
Payload:  heatingactive: aus
tapwateractive: aus
selflowtemp: 37
heatingpumpmod: 100
outdoortemp: 9.5
curflowtemp: 40.4
rettemp: 40.3
syspress: 1.9
burngas: aus
flamecurr: 0.1
heatingpump: an
fanwork: aus
ignwork: aus
heatingactivated: an
heatingtemp: 75
burnminperiod: 0
burnminpower: 18
burnmaxpower: 100
boilhyston: -6
boilhystoff: 2
absburnpow: 0
selburnpow: 100
curburnpow: 0
burnstarts: 21734
burnworkmin: 160230
burn2workmin: 160230
heatworkmin: 141575
heatstarts: 20008
ubauptime: 543351
lastcode: A01(808) 23.10.2023 11:42 - 23.10.2023 11:51
servicecode: ' 0Y'
servicecodenumber: 204
maintenancemessage: H20
maintenance: aus
maintenancetime: 6000
emergencyops: aus
emergencytemp: 30
nompower: 30
nrgheat: 4.09

See picture below, for available entities
[image]

- [ ] Provide the output of http://ems-esp.local/api/system :
  System information output here:
 {
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "000+01:12:17.204",
    "uptime (seconds)": 4337,
    "free mem": 126,
    "max alloc": 55,
    "free app": 437,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp-gas",
    "RSSI": -47,
    "IPv4 address": "192.168.178.77/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "253.0.0.0",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 2899,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 2,
    "enabled": true,
    "client id": "mems-esp-gas",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "mems-esp-gas",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 2,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {},
  "API Info": {
    "API calls": 2,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 15154,
    "bus reads (tx)": 1575,
    "bus writes (tx)": 1,
    "bus incomplete telegrams": 3,
    "bus reads failed": 6,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "CUSTOM",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "phy type": 0,
    "rx gpio": 16,
    "tx gpio": 17,
    "dallas gpio": 0,
    "pbutton gpio": 0,
    "led gpio": 0,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Logano GB125/KB195i/Logamatic MC110",
      "device id": "0x08",
      "product id": 133,
      "version": "02.10",
      "entities": 64,
      "handlers received": "0xBF 0xC2 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x02D6 0x08E4 0xF9 0xF7 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 49,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xF9 0x02E0 0x02EA 0x033B 0xBF 0x0313"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x28",
      "product id": 160,
      "version": "24.05",
      "entities": 8,
      "handlers received": "0x0331",
      "handlers fetched": "0x0313",
      "handlers ignored": "0xD4 0x0255 0xBF 0xF9"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "07.02",
      "entities": 0,
      "handlers ignored": "0x125A 0x2040"
    },
    {
      "type": "heatsource",
      "name": "AM200",
      "device id": "0x60",
      "product id": 228,
      "version": "03.03",
      "entities": 3,
      "handlers received": "0x054D 0x054E 0x0550",
      "handlers fetched": "0x054C",
      "handlers pending": "0x054F",
      "handlers ignored": "0xD8 0xBF 0xF6"
    }
  ]
}

proddy, 2023-11-06:
If using multiple EMS-ESPs make sure the Entity Format is set to 'Multiple Instances' (documented in https://emsesp.github.io/docs/Configuring/#setting-up-mqtt).

[image]

FloCodeRian, 2023-11-06:
I’m so sorry, but very thankful. Seems like working now!

bbqkees, 2023-11-07:
Goo hint I'll add it to my wiki as well.

closed 2023-11-11 by proddy