emsesp/EMS-ESP32 #1396: Non ASCII String in JSON / MQTT Message

DoubleYou2, 2023-11-07:
**PROBLEM DESCRIPTION**

RC310 / Buderus GB125 / EMS-ESP V3.6.2:

Non ASCII chars in serviceCode of boiler during warm water heating...
... causes EMS-ESP32 gateway to use non UTF-8 coded string for JSON sent via MQTT to Home Assistant.
Home Assistant responds to this non UTF-8 coded string with "Warning: Can't decode payload..".

This issue  (which is identical to my issue reported here) is thoroughly described, analysed and (probably) fixed
in an problem report 2 years ago for ems-esp: https://github.com/emsesp/EMS-ESP/issues/686

Unfortunately this fix has never been integrated / merged to ems-esp32 v3.5.0 or later.

As I'm not a developer I can't do it on my own. Hope there is somebody who can verify and probably merge existing fix
to v3.6.3 or later.

DoubleYou2, 2023-11-07:
HA shows this warning (only relevant parts shown):
WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{ ...... ,**"servicecode":" \xf0H"**, ......."}'

There seems to be an additional blank inserted?

MichaelDvP, 2023-11-07:
> There seems to be an additional blank inserted?

Telegram 0x18 uses a 2char servicecode, telegram 0xE4 uses 3char,i'll also check/replace the second char of serviceCode in 0xE4 in my dev2 testbuild.

DoubleYou2, 2023-11-07:
... thanks... will verify when available...

MichaelDvP, 2023-11-07:
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

DoubleYou2, 2023-11-07:
... ok, can verify changes in MQTT Explorer and Gateway. HA does not complain any more. ;)

Thanks for your very fast support and fix!!

PS:
Do you recommend to stay with this dev version or do you recommend to revert to official release?

MichaelDvP, 2023-11-07:
You can stay on this version until the fix is also merged in the official dev. My testbuild have breaking changes for mixer/solar water modules, this will go to the next 3.7.0, But i will also make a PR to the 3.6.3 dev with tested fixes.

proddy, 2023-11-19:
Michael's change went into dev

closed 2023-11-19 by proddy