emsesp/EMS-ESP32 #1397: Missing Values on BK13/BK15/Smartline/GB1x2

stipzta, 2023-11-07:
### PROBLEM DESCRIPTION

The value for "wwcylmiddletemp" is not output. However, I can see this value in my RC30. If more information is needed, please let me know. I am still a beginner with the ESP system.

### Devices
   ```
   "type": "boiler",
      "name": "BK13/BK15/Smartline/GB1x2",
      "device id": "0x08",
      "product id": 64,
      "version": "03.02",
      "entities": 76,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC30",
      "device id": "0x10",
      "product id": 67,
      "version": "02.08",
      "entities": 49,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x48 0x47 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
      "handlers pending": "0x13 0x42 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60",
      "handlers ignored": "0x35 0x1A"
    },
    {
      "type": "solar",
      "name": "SM10",
      "device id": "0x30",
      "product id": 73,
      "version": "02.00",
      "entities": 16,
      "handlers received": "0x97",
      "handlers fetched": "0x96",
      "handlers ignored": "0x35"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.02",
      "entities": 0,
      "handlers ignored": "0x29"
```

### SCREENSHOTS
[image]

MichaelDvP, 2023-11-07:
Do you mean the cylindermiddletemp in RC30 solar menu? This is not measured from SM10, it's just taken from boiler wwstoragetemp1/wwcurrtemp. The wwcylmiddletemp from boiler is only published by a MC110 controller in telegram 0x2A, this is not received in your system. (I also have BC10 controller and SM10 solar).

stipzta, 2023-11-07:
So that means the temperature is only passed on and not actually measured? So then I can assume the wwcurrtem, right?

closed 2024-03-24 by proddy