emsesp/EMS-ESP32 #1398: Extend "HK1 Mode"

Bingo2023, 2023-11-07:
SW: 3.6.3-dev.4
as you can see below, we should extend the hc1 mode:

- now: manual, auto
- new: manual, auto, stop

Second pic is Buderus App (via MX300).
As I don't know what is the command in detail, I attached a log where I switched via App from "manual" to "Aus".
@MichaelDvP : should we confirm this via custom entity ?

[image]
[image]
[log.txt](https://github.com/emsesp/EMS-ESP32/files/13289674/log.txt)

Bingo2023, 2023-11-07:
one more thing, I think the above setting is supposed the be reflected in HA on below panel:
[image]

the mode control in this panel is not linked to hc1/mode ?

MichaelDvP, 2023-11-08:
Difficult, log shows:  `gateway(0x48) -W-> thermostat(0x10), RC300Set(0x02B9), data: 00 (offset 21)`
but in all logs before this telegram is only offset 0-20. What thermostat (prod-Id, version) is it? What is setting of offset 21 if another mode is set? manual/auto is set in offset 0. Please switch the modes and do a `read 10 2B9`, hopefully offset 0 and 21 is shown.

Afaik the climate widget is fixed to modes off/heat/auto and can not be changed. HA users may correct me, i'm not using HA.

Bingo2023, 2023-11-09:
Hallo Michael,
ich habe eine neue Log gemacht, wo ich die alle 3 Modes per App mehrmals durchgeschalteten habe.
Vielleicht war die erste Log zu kurz. Die Eingaben habe ich alle mit der Buderus App gemacht. Als Thermostat ist ein Buderus RC 220 verbaut (wobei der nur im Keller an der WP ist damit diese läuft, das Haus hat ansonsten KNX).

Was meinst du mit "read 10 2B9"?

[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/13311938/log.1.txt)

Bingo2023, 2023-11-22:
[image]
I confirm this change, it is working with latest Test release.
Mode change (Aus, Manuell, Auto) is working

t785634, 2023-11-22:
But only with the entity select.thermostat_hc1_mode. The Home Assistant thermostat entity (climate.thermostat_hc1) cannot be used to change the mode from Auto to Off.

Bingo2023, 2023-11-22:
I checked HA situation:
I marked the Bottoms that I pressed.
Functionality wise it is working perfectly - all the modes are send to the HP.

BUT
as you can see on the pictures, the HA visualization is not very good.
- In center is always shows "auto"
- the 3 bottoms: two always grey and left always green

[image]
[image]
[image]

Bingo2023, 2023-11-22:
have to admit, I used latest Testbuilt from Michael
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

t785634, 2023-11-22:
Is there a way to make Home Assistant display the correct mode? When I change the mode from Auto to Off, it goes back to Auto after 1 second.
I also use the latest Testbuilt from Michael.

[image]

proddy, 2023-12-02:
@Bingo2023 @t785634 are you still seeing issues with the HA climate control not showing the correct modes? It works for me, but I'm using a different thermostat

MichaelDvP, 2023-12-02:
We have now two different solutions, in the dev the mode is enhanced to off/manual/auto, but we learned RC300 doesn not have the off-mode. I feat we'll get complains from RC300 users for not working mode. In my test (dev2) i've made a split for RC300 types and BC400 thermostats for thermostat-mode and ww-mode. Don't know if product-id 4 and 253 are the only ones with this mode.  Still null feedback if it works.
Should we merge the testbuild to dev2 here to have more testers?

Bingo2023, 2023-12-02:
@t785634 I can confirm what you describ.
My pic shows the main HA control, from here the control works, but it is not visualized correct. When I click on panel to enlarge it, the mode control is jumping back to auto after 1sec.

So in general the mode control with 3 values works good (off/manual/auto) but the HA implementation needs to be fixed. I'm new to HA so I don't know if EMS-ESP32 or HA...

proddy, 2023-12-04:
> @proddy yes, the problem still exists

can you use MQTT Explorer or something similar and dig out the MQTT Discovery topic and payload for the climate? It'll be something like `homeassistant/climate/ems-esp/thermostat_hc1/config`

I need to see the `mode_stat_tpl` in the JSON as that is probably where the problem lies.

t785634, 2023-12-04:
> can you use MQTT Explorer or something similar and dig out the MQTT Discovery topic and payload for the climate? It'll be something like `homeassistant/climate/ems-esp/thermostat_hc1/config`
>
> I need [...]

{
  "~": "ems-esp",
  "uniq_id": "thermostat_hc1",
  "obj_id": "thermostat_hc1",
  "name": "Hc1",
  "mode_stat_t": "~/thermostat_data",
  "mode_stat_tpl": "{%if value_json.hc1 is undefined or value_json.hc1.mode is undefined%}off{%elif value_json.hc1.mode=='manual'%}heat{%elif value_json.hc1.mode=='day'%}heat{%elif value_json.hc1.mode=='night'%}off{%elif value_json.hc1.mode=='off'%}off{%else%}auto{%endif%}",
  "temp_cmd_t": "~/thermostat/hc1/seltemp",
  "temp_stat_t": "~/thermostat_data",
  "temp_stat_tpl": "{{value_json.hc1.seltemp if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 0}}",
  "curr_temp_t": "~/thermostat_data",
  "curr_temp_tpl": "{{value_json.hc1.currtemp if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 0}}",
  "min_temp": "5",
  "max_temp": "30",
  "temp_step": "0.5",
  "mode_cmd_t": "~/thermostat/hc1/mode",
  "modes": [
    "auto",
    "heat",
    "off"
  ],
  "dev": {
    "ids": [
      "ems-esp-thermostat"
    ],
    "name": "ems-esp-thermostat"
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'offline' if value_json.hc1 is undefined or value_json.hc1.mode is undefined else 'online'}}"
    }
  ],
  "avty_mode": "all"
}

t785634, 2023-12-04:
> can you use MQTT Explorer or something similar and dig out the MQTT Discovery topic and payload for the climate? It'll be something like `homeassistant/climate/ems-esp/thermostat_hc1/config`
>
> I need [...]

`{
  "~": "ems-esp",
  "uniq_id": "thermostat_hc1",
  "obj_id": "thermostat_hc1",
  "name": "Hc1",
  "mode_stat_t": "~/thermostat_data",
  "mode_stat_tpl": "{%if value_json.hc1 is undefined or value_json.hc1.mode is undefined%}off{%elif value_json.hc1.mode=='manual'%}heat{%elif value_json.hc1.mode=='day'%}heat{%elif value_json.hc1.mode=='night'%}off{%elif value_json.hc1.mode=='off'%}off{%else%}auto{%endif%}",
  "temp_cmd_t": "~/thermostat/hc1/seltemp",
  "temp_stat_t": "~/thermostat_data",
  "temp_stat_tpl": "{{value_json.hc1.seltemp if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 0}}",
  "curr_temp_t": "~/thermostat_data",
  "curr_temp_tpl": "{{value_json.hc1.currtemp if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 0}}",
  "min_temp": "5",
  "max_temp": "30",
  "temp_step": "0.5",
  "mode_cmd_t": "~/thermostat/hc1/mode",
  "modes": [
    "auto",
    "heat",
    "off"
  ],
  "dev": {
    "ids": [
      "ems-esp-thermostat"
    ],
    "name": "ems-esp-thermostat"
  },
  "avty": [
    {
      "t": "ems-esp/status",
      "val_tpl": "{{'online' if value == 'online' else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'offline' if value_json.hc1 is undefined or value_json.hc1.mode is undefined else 'online'}}"
    }
  ],
  "avty_mode": "all"
} `

Bingo2023, 2023-12-22:
HA mode control fix see here:
https://github.com/Bingo2023/EMS-ESP32/commit/4c51b90663486f97cb5c7dde20263402cf50b745

MichaelDvP, 2023-12-23:
Thanks, i have never realized that mqtt enums are translated. Should they?
To make it work  with all thermostats in all langauges, please check if this works too:
```
    snprintf(mode_str_tpl,
             sizeof(mode_str_tpl),
             "{%%if %s%%}off{%%elif %s=='%s'%%}heat{%%elif %s=='%s'%%}heat{%%elif %s=='%s'%%}off{%%elif %s=='%s'%%}off{%%else%%}auto{%%endif%%}",
             hc_mode_cond,
             hc_mode_s,
             Helpers::translated_word(FL_(manual)),
             hc_mode_s,
             Helpers::translated_word(FL_(day)),
             hc_mode_s,
             Helpers::translated_word(FL_(night)),
             hc_mode_s,
             Helpers::translated_word(FL_(off)));
```

Bingo2023, 2023-12-23:
I think it is monitoring the topic "ems-esp/thermostat_data".
I you check that message it is translated, see below:

[image]
[image]

In general it would be better not to translate mqtt twice, but at the moment large parts are already translated.
If we are don't translate "ems-esp/thermostat_data" it could solve the problem and we don't have to apply above mentioned fix.
On the other hand I don't know the side effect of removing the translation.

MichaelDvP, 2023-12-23:
I've checked the code for HA discovery, the translation is right and usefull. The config uses the translated fullname and the translated enum-options as list. Sorry for confusiong, but i'm not using HA and it's sometimes difficult to understand.

So the change i've mentioned here https://github.com/emsesp/EMS-ESP32/issues/1398#issuecomment-1868233586 should work. Please try.

Bingo2023, 2023-12-23:
> Thanks, i have never realized that mqtt enums are translated. Should they? To make it work with all thermostats in all langauges, please check if this works too:
>
> ``` [...]

it is working. I did an Commit out of Visual Studio Code, don't know if this helps - it is my first commit :)

MichaelDvP, 2023-12-23:
> I did an Commit out of Visual Studio Code, don't know if this helps - it is my first commit :)

Do you like to try a PR too?

Bingo2023, 2023-12-23:
done.

proddy, 2023-12-29:
done, also merged into dev. Congrats on the PR @Bingo2023 !

closed 2023-12-29 by proddy