emsesp/EMS-ESP32 #1403: Incomplete Rx

FunnyPaul, 2023-11-10:
Hello everyone and thank you for this great project!
I'm new here, have experience with the ESP32 and hardware development and have the system running almost straight away on a BOSCH GC7000F, but I haven't yet set up MQTT etc. I can read all the values etc.

What surprises me, however, is that 
**"... WARNING nnnn: [telegram] Incomplete Rx ..."** 

appears in the system log for every packet. Is this normal or what do I have to do?

Best regards from Germany

proddy, 2023-11-11:
Look at https://emsesp.github.io/docs/Troubleshooting/#many-rx-errors

FunnyPaul, 2023-11-11:
Thank you! It was due to my hardware design. I would like to supply the module via the voltage on the bus. I have now optimized this and the errors are gone.

So the case is not an error or bug. It can be closed :-)

closed 2023-11-11 by FunnyPaul

proddy, 2023-11-11:
great!