emsesp/EMS-ESP32 #1406: Too many arguments for command when using "set timeout 0"

HansRemmerswaal, 2023-11-11:
In the past I could use the set timeout 0 from the prompt, but now I'm getting an error "Too many arguments for command". I recently upgraded so I do not know since which version this is not working anymore.

**Version 3.5.0b12**

┌──────────────────────────────────────┐
│ EMS-ESP version 3.5.0b12 │
│ https://github.com/emsesp/EMS-ESP32 │
│ │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ set timeout 0
Telnet timeout set to 0 minutes
ems-esp:$ watch 8 F7
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xF7

**Version 3.6.30dev.6**

┌────────────────────────────────────────┐
│ EMS-ESP version 3.6.3-dev.6            │
│ https://github.com/emsesp/EMS-ESP32    │
│                                        │
│ type help to show available commands   │
└────────────────────────────────────────┘

00Error writing session log (raw mode) to file: putty.log
0+00:15:12.073 I 0: [shell] User session opened on console pty0
ems-esp:$ set timeout 0
Too many arguments for command
ems-esp:$

proddy, 2023-11-11:
the timeout command was removed. Do you really need it?

HansRemmerswaal, 2023-11-11:
Good question... You gave me these two commands sometime ago.

set timeout 0
watch 8 F7

Will try without.

closed 2023-11-19 by proddy