emsesp/EMS-ESP32 #1410: Missing entity Hot Water Priority @Buderus Logano GB212 + RC300

Fiona3600, 2023-11-11:
EMS-ESP version 3.6.2
Buderus Logano GB212 and RC300

In the control unit RC300,
under the setting Heating/Circuit 1,
there is an option for Hot Water Priority Yes/No.

However, I can't find a corresponding entity in EMS-ESP.

proddy, 2023-11-11:
I think we need to add it. Can you help find the data? https://emsesp.github.io/docs/Troubleshooting/#im-missing-certain-data-from-an-ems-device

Fiona3600, 2023-11-12:
Within the log, I toggled the value of the entity twice. Unfortunately, I can't identify the entity. Does the log help or is a watch needed over the console?
`2023-11-12 13:07:08.416 TRACE 305: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 06 04 00 00 00 00 FF 64 41 00 3C 01 FF 01 (offset 25)
2023-11-12 13:07:08.533 TRACE 306: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Set(0x02B9), length: 0x19
2023-11-12 13:07:08.593 TRACE 307: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: FF 2E 2A 26 24 01 5E FF FF 05 2A 01 00 20 01 0F 05
2023-11-12 13:07:09.220 TRACE 314: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0292), length: 0x19
2023-11-12 13:07:09.379 TRACE 316: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0293), length: 0x19
2023-11-12 13:07:09.426 TRACE 317: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0293), data: 00 00 00 00 FF 00
2023-11-12 13:07:09.916 TRACE 323: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 00 4A
2023-11-12 13:07:10.090 TRACE 325: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 32 00 01 00 00 00 00 00 00 02 F6 06 FF 00
2023-11-12 13:07:13.821 TRACE 327: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 70 00 6F 21 00 3C 20 00 00 (offset 27)
2023-11-12 13:07:19.312 TRACE 329: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00
2023-11-12 13:07:23.794 TRACE 331: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 70 00 6F 21 00 3C 20 00 00 (offset 27)
2023-11-12 13:07:31.713 TRACE 333: [emsesp] thermostat(0x10) -R-> boiler(0x08), UBADevices(0x07), length: 0x0E
2023-11-12 13:07:33.486 TRACE 335: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 20 01 46 50 17 09 01 A7 00 00 FA 7D 00 01 23 00 55 0B 2D 48 00 00 FF 02 10 80 00
2023-11-12 13:07:34.172 TRACE 337: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0B 00 FB 7D 00 00 00 05 03 00 08 C1 AB 00 9C 5C
2023-11-12 13:07:43.460 TRACE 338: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 20 01 46 50 17 09 01 A7 00 00 FB 7D 00 01 23 00 54 0B 2D 48 00 00 FF 02 10 80 00
2023-11-12 13:07:43.740 TRACE 339: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 70 00 71 21 00 3C 20 00 00 (offset 27)
2023-11-12 13:07:44.145 TRACE 340: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0B 00 FB 7D 00 00 00 05 03 00 08 C1 AB 00 9C 5C
2023-11-12 13:07:46.220 TRACE 341: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 80 00 01 2A 20 00 2A 24 02 6F 03 03 01 02 6F 01 AB 00 00 11 01 03 FF FF 00
2023-11-12 13:07:46.441 TRACE 342: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 06 04 00 00 00 00 FF 64 41 00 3C 01 FF 01 (offset 25)
2023-11-12 13:07:47.150 TRACE 345: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Floordry(0x0267), data: 00 00
2023-11-12 13:07:53.533 TRACE 346: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 20 01 46 50 17 09 01 A7 00 00 FB 7D 00 01 23 00 57 0B 2D 48 00 00 FF 02 10 80 00
2023-11-12 13:07:54.219 TRACE 348: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0B 00 FB 7D 00 00 00 05 03 00 08 C1 AB 00 9C 5C
2023-11-12 13:07:55.453 TRACE 349: [emsesp] thermostat(0x10) -B-> All(0x00), RCTime(0x06), data: 17 0B 0D 0C 07 32 06 00 10 FF 00
2023-11-12 13:08:03.505 TRACE 350: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 20 01 46 50 17 09 01 A7 00 00 FB 7D 00 01 23 00 52 0B 2D 48 00 00 FF 02 10 80 00
2023-11-12 13:08:03.785 TRACE 351: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 70 00 70 21 00 3C 20 00 00 (offset 27)
2023-11-12 13:08:04.191 TRACE 352: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0B 00 FB 7D 00 00 00 05 03 00 08 C1 AB 00 9C 5C
2023-11-12 13:08:04.501 TRACE 353: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 00 4A 01 42 7D 00 FF FF 00 64 01 96 38 1C D9 62 1C D9 62 1B C1 1B 01 7C 96 7D 00
2023-11-12 13:08:04.876 TRACE 354: [emsesp] boiler(0x08) -B-> All(0x00), UBATotalUptime(0x14), data: 50 B8 11
2023-11-12 13:08:07.132 TRACE 355: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
2023-11-12 13:08:07.398 TRACE 358: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 2D 50 17 03 FE 05 02 00 64 17 00 08 F8 0F 0F 0F 0F 1E 05 06 00 00 00 28 00 3C
2023-11-12 13:08:07.565 TRACE 360: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: <empty> (offset 27)
2023-11-12 13:08:07.683 TRACE 361: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
2023-11-12 13:08:07.896 TRACE 363: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19
2023-11-12 13:08:08.069 TRACE 365: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 02 6E 01 AC (offset 13)
2023-11-12 13:08:08.354 TRACE 367: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 06 04 00 00 00 00 FF 64 41 00 3C 01 FF 01 (offset 25)
2023-11-12 13:08:08.469 TRACE 368: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Set(0x02B9), length: 0x19
2023-11-12 13:08:08.871 TRACE 373: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Curves(0x029B), data: 02 01 00 00 FF 02 03 30 41
2023-11-12 13:08:09.032 TRACE 375: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0291), data: 01 00 00 00 FF 00
[... cut ...]
2023-11-12 13:08:13.478 TRACE 388: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 20 01 46 50 17 09 01 A7 00 00 FB 7D 00 01 23 00 57 0B 2D 48 00 00 FF 02 10 80 00
2023-11-12 13:08:13.757 TRACE 389: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 70 00 70 21 00 3C 20 00 00 (offset 27)
2023-11-12 13:08:14.164 TRACE 390: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0B 00 FB 7D 00 00 00 05 03 00 08 C1 AB 00 9C 5C
2023-11-12 13:08:19.450 TRACE 392: [emsesp] boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00`

proddy, 2023-11-12:
is there a way you can turn it on and off, and at the same time capture the log?

Fiona3600, 2023-11-12:
Thats what I did. Logging on, change Hot Water Priority and change it again, logging off. Is there a way to encircle the entity?

proddy, 2023-11-13:
timestamps for on and off would be helpful

MichaelDvP, 2023-11-15:
There is no change in the log.  Do you have only selected the menu entry or confirmed the setting? 
Strange that wwprio is missing, we read/set it from 0x2CC offset3 for RC300.
Please attach logs as file, not copy/past in the text field,

Fiona3600, 2023-11-16:
If  wwprio read/set is from 0x2CC offset3 for RC300 I should be able to configure an entity by myself ID 0x10 Typ ID 0x02CC addition 3, type bool, writable. Right? Wont work.

Under cutomizations rc300 there is an entity  hc1 dhw priority showing up as hc1 hc1 dhw priority under dashboard devices RC300. But changing the value has no effect, too.

In the attached log I changed from No to Yes and back to No.
[log.txt](https://github.com/emsesp/EMS-ESP32/files/13384450/log.txt)

Let me know how I can assist more

MichaelDvP, 2023-11-17:
>  If wwprio read/set is from 0x2CC offset3 for RC300 I should be able to configure an entity by myself ID 0x10 Typ ID 0x02CC addition 3, type bool, writable. Right? Wont work.

Yes, i see in the log that thermostat replys with empty message, The entity was found on a RC310, te we have a difference to RC300.
> Let me know how I can assist more

Try (as in first post) to log and change the value on the thermostat. Make sure it is has changed. We have to find the right telegram/offset.

Fiona3600, 2023-11-17:
this is exactly what I did in the log 14 hours ago (log.txt)

MichaelDvP, 2023-11-17:
> this is exactly what I did in the log 14 hours ago (log.txt)

The log contains only monitor values, there not a single setting changed.

Fiona3600, 2023-11-17:
strange. So now I startet a one tim ww loading first to control the result by watching the normal pump. I set the ww prio loading to off (the normal pump went off) and I set the ww prio loading to on again (normal pumt went on again).

Hoping the log-2 catches all.

[log-2.txt](https://github.com/emsesp/EMS-ESP32/files/13392974/log-2.txt)

MichaelDvP, 2023-11-17:
I don't know what you are doing, this log is only 10 sec and contains 18 messages, but log counter goes up 49 messages, where are the rest?. 
Most messages on the ems bus takes much more than 10 sec to show up.

- Go to log page
- select log all
- wait a minute
- go to thermostat
- change the setting, note new value
- wait a minute
- change the setting back, note new value
- wait a minute (
- download log
- post log and the noted settings/values

Fiona3600, 2023-11-21:
Can you work with the data, or is there anything else I can do?

proddy, 2023-11-21:
on my todo

proddy, 2023-12-06:
we need to know roughly when you made the change (we can't tell from the 540 line items) and also what was the original value and what the new value is. Without that its like looking for a needle in a haystack

closed 2024-03-24 by proddy