emsesp/EMS-ESP32 #1414: Unable to set hc1_mode on BC400/GB172i.2 combo

h3llsp4wn88, 2023-11-12:
### PROBLEM DESCRIPTION

When trying to change the mode for hc1 to auto or manual it is not switching/changing.

### REQUESTED INFORMATION
I tried several ways of changing the setting - none of them changes the mode.
Tried on latest dev 7 and on 3.6.2 stable

Cheers,

h3ll

proddy, 2023-11-12:
I will check on my system

proddy, 2023-11-12:
what happens if you try from the console
`call thermostat mode` to show the current mode setting (under value)
`call thermostat mode manual` to change it to manual - watch for errors in the log
`call thermostat mode` again to see if the value has changed to manual

h3llsp4wn88, 2023-11-12:
Ok, found out I have to SU to use call :)

There are no errores in the log - whether I set manual or auto. See watch output below after commands have been send:

000+00:22:12.859 I 344: [command] Calling command 'thermostat/mode' (mode) with value auto
000+00:22:12.919 N 345: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Set(0x02B9), data: FF
000+00:22:15.021 N 346: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19
000+00:22:15.101 N 347: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 E6 01 2E 2D 00 2E 24 02 5D 03 03 01 02 5D 01 27 00 00 11 01 00 08 FA 00
000+00:22:15.329 N 348: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19 (offset 25)
000+00:22:15.400 N 349: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 0A 64 41 00 3C 01 2C 01 02 02 00 00 06 58 0B (offset 25)
000+00:22:17.648 N 350: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 CC 2D 01 9F 64 00 02 00 01 97 00 00 01 A9 00 00 11 09 01 9F 00 00
000+00:22:17.825 N 351: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 8E 00 92 00 61 (offset 35)
000+00:22:18.014 N 352: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 01 00 00 00 00 00 00 00 00 01 9F 00 64 41 00 2D 00 00
000+00:22:18.285 N 353: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 05 02 14 02 14 00 00 00 00 41 3C 00 00 02 00 04 91 00 00 3C 00 00 00 05 02 00
000+00:22:19.226 N 354: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A0 (offset 23)
000+00:22:23.207 N 355: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 02 5C 01 28 (offset 13)
000+00:22:24.416 N 356: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A3 (offset 23)
000+00:22:27.663 N 357: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 CC 2D 01 A4 64 00 02 00 01 9B 00 00 01 A7 00 00 11 08 01 A4 00 00
000+00:22:27.841 N 358: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 8B 00 92 00 61 (offset 35)
000+00:22:28.030 N 359: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 01 00 00 00 00 00 00 00 00 01 A4 00 64 41 00 2D 00 00
000+00:22:28.263 N 360: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 05 02 14 02 14 00 00 00 00 41 3C 00 00 02 00 04 91 00 00 3C 00 00 00 05 02 00

000+00:23:13.032 I 449: [command] Calling command 'thermostat/mode' (mode) with value manual
000+00:23:13.075 N 450: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Set(0x02B9), data: 00
000+00:23:15.204 N 451: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19
000+00:23:15.282 N 452: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 E6 01 2E 2D 00 2E 24 02 5C 03 03 01 02 5C 01 28 00 00 11 01 00 08 FA 00
000+00:23:15.544 N 453: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19 (offset 25)
000+00:23:15.615 N 454: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 0A 64 41 00 3C 01 2C 01 02 00 00 00 06 58 0B (offset 25)
000+00:23:17.681 N 455: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 CC 2D 01 92 64 00 02 00 01 8F 00 00 01 8F 00 00 11 08 01 92 00 00
000+00:23:17.858 N 456: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 86 00 93 00 61 (offset 35)
000+00:23:18.049 N 457: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 01 00 00 00 00 00 00 00 00 01 92 00 64 41 00 2D 00 00
000+00:23:18.253 N 458: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 05 02 14 02 14 00 00 00 00 41 3C 00 00 02 00 04 91 00 00 3C 00 00 00 05 02 00
000+00:23:22.107 N 459: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 91 (offset 23)
000+00:23:23.353 N 460: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 02 5B 01 29 (offset 13)
000+00:23:27.710 N 461: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 CC 2D 01 91 64 00 02 00 01 8E 00 00 01 8F 00 00 11 07 01 91 00 00
000+00:23:27.885 N 462: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 86 00 93 00 61 (offset 35)
000+00:23:28.077 N 463: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 01 00 00 00 00 00 00 00 00 01 91 00 64 41 00 2D 00 00
000+00:23:28.281 N 464: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 05 02 13 02 13 00 00 00 00 41 3C 00 00 02 00 04 91 00 00 3C 00 00 00 05 02 00

Hope this helps and thank you for looking into it!

MichaelDvP, 2023-11-12:
Maybe related to #1398, Try `read 10 2b9`. Or create a custom entitiy for device 0x10, type 0x2B9, offset 21, uint, writeable. Does it show 0 (off), 1 (manual), 2 (auto) ?

h3llsp4wn88, 2023-11-12:
Well - that works perfectly - thanks. I have figured out that with custom I can even set it over MQTT. Only thing that I need so solve is to update hc1_mode when sending the command - so re-reading values (mode change changes a couple of values on the thermostat). Would be nice to have a feature for custom entities to trigger a device update so it get's published accordingly.

Cheers! 

... and thanks a lot!

MichaelDvP, 2023-11-13:
I wonder if we have missed this setting a long time, or if it is new on some thermostats, like BC400, product-id 4 and maybe 253 (users never sends the system info and i'm tired of asking again and again without getting an answer). 

@tp1de Could you check on your older RC310 if you have offset 21 in 0x2B9 and if writing changes the mode? Then we can change for all RC300 compatible, or need to add some logic to find the right setting.

h3llsp4wn88, 2023-11-13:
Well - I guess it is a new one (kind of hidden as it changes a lot of other values or copies them over like e.g. eco or comfort temp to seltemp and some modetypes). It seems at least Buderus changed the internal handling on the new devices. The new series (not so new) with gbXi.2 seem to have it all (most come with the BC400 I think). As I updated mine I had to swap all the RCs as they are no longer compatible with the new burner. The km200 as well did not work - maybe because of this and the change to the "plus" protocol version. But I'm not a "pro" here :)

Maybe most of the people never change it - but I got my own "holiday" program so I like to be able to handle it.

Could be that water heating has similar behaviour - but changing the mode there worked for now.

By the way - the weekday programs were readable with the km200 - I haven't seen these values for the ems-esp so far? (wrong place to ask - I know ...)

tp1de, 2023-11-13:
> Could you check on your older RC310 if you have offset 21 in 0x2B9 and if writing changes the mode? Then we can change for all RC300 compatible, or need to add some logic to find the right setting.

@MichaelDvP no it doesn't work on my RC310

tp1de, 2023-11-13:
@h3llsp4wn88  you are asking for holidaymodes and/or switchtimes. Are you using ioBroker or Home Assistant?

MichaelDvP, 2023-11-13:
> @MichaelDvP no it doesn't work on my RC310

Sad, i've tried a solution for this: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test should work with both settings. Only for old RC3x0 it shows mode off/manual/auto and setting to off results also in manual mode. Is this ok?

Alternative  solution is to guess and roll some dice to flag some thermostats to new mode setting, i would start with product-id 4 and wait for user-complains with attached system info.

MichaelDvP, 2023-11-13:
> other mqtt entities.

**?**

tp1de, 2023-11-13:
I checked again. Missing are :

select.thermostat_hc1/2_summersetmode
number.thermostat_hc1/2_summertemp
select.thermostat_hc1/2_mode

MichaelDvP, 2023-11-13:
Updated, @tp1de please check again, i don't want to break working systems..

tp1de, 2023-11-13:
@MichaelDvP As far as I can see everything works as before.

h3llsp4wn88, 2023-11-15:
@tp1de I'm using FHEM with the MQTT2 module there.

tp1de, 2023-11-15:
I have implemented the coding read/write in javascript for my ioBroker adapter.
The holiday modes are not to difficult. You can test in custom entities:

[image]

closed 2023-11-25 by proddy