emsesp/EMS-ESP32 #1415: Add new signal for condensate reservoir heating EA0 to message 0x48D (hppwr)

KarstenBr, 2023-11-12:
### PROBLEM DESCRIPTION
Message 0x48D (hppwr) provides the switch state of the electrical condensate reservoir heating EA0 which is not processed with SW 3.6.3-dev6c

### REQUESTED INFORMATION
I have monitored the the switch status of EA0 in the heatpump's diagnostic display while the heatpump changed into defrosting mode at low outside temperatures (knowing the activation/deactivation conditions for EA0). I then was able to identify the corresponding signal in the 3rd bit of the 4th byte (offset 3) of message 0x48D (hppwr):
EA0 being not active
[emsesp] boiler(0x08) -W-> Me(0x0B), HpPower(0x048D), data: 00 20 10 10 ...
EA0 being active
[emsesp] boiler(0x08) -W-> Me(0x0B), HpPower(0x048D), data: 00 20 10 **5**0 ...

### EXPECTED BEHAVIOUR
Introduce new entity having two logical states "on" and "off"

naming proposal for entity:
EN "condensate reservoir heating (EA0)"
DE "Heizung Kondensatwanne (EA0)"

KarstenBr, 2023-11-12:
@MichaelDvP Tankd for adding it to 3.6.3-dev7a. As soon as I have a state change of EA0 I will check the correct behavior of the new entity for EA0.

Nevertheless I recognized a wrong German naming of this entity. It is shown as "WW Heizung Kondensatwanne (EA0)", but it should not have "WW" in its name as it is not linked to warmwater but this heating is related to defrosting of the evaporator - regardless if during heating or WW provision.

KarstenBr, 2023-11-12:
The heatpump did just now activate EA0 and this was properly reflected in hpEA0.

proddy, 2024-06-21:
Can we close this? @KarstenBr please verify with the latest dev version