emsesp/EMS-ESP32 #1416: Incorrect value of total uptime (ubauptime)

KarstenBr, 2023-11-12:
### PROBLEM DESCRIPTION
SW 3.6.3-dev6c shows an incorrect value in the total uptime of the boiler. While the heatpump's diagnostic display at the moment shows 1791h total time of operation (=74.625 days) the gateway reports 58d 9h 15min (84075mins).

### REQUESTED INFORMATION
I have checked the raw data of message 0x14 (UBATotalUptime) which since several days shows a constant value which corresponds to the above mentioned 84075 mins:
[emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 01 48 6B

By using the watch command for 0x14 I can confirm that this message is requested by the gateway in regular time intervals of ~1min and receives respective answers from the boiler.
ems-esp:$ watch 0x14
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x14
004+11:30:29.808 N 5: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x08
004+11:30:29.820 N 6: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 01 48 6B
004+11:31:29.381 N 7: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x08
004+11:31:29.393 N 8: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 01 48 6B
004+11:32:28.495 N 9: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x08
004+11:32:28.506 N 10: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 01 48 6B

I don't know if the mapping of ubauptime to message 0x14 is incorrect or if there is a bug in reading message 0x14.

### EXPECTED BEHAVIOUR
Total operation time of the boiler is correctly reflected in entity ubauptime

proddy, 2023-11-12:
HEX 0x1486B is 84,075 so we're reading it correctly, but looks like it's not calculated correctly. I can't read the uptime on my boiler and we would need more examples to see if this is consistent across other boilers and heatpumps.

MichaelDvP, 2023-11-12:
For my GB125 (prod-id 72) the uptime is displayed correctly. Seems for heatpumps there is another counter and 0x14 counts only a special operating state.

KarstenBr, 2023-11-12:
Hm, ok - then I will have to do some reverse engineering on other messages during the next days… At least the counters for active heating and warmwater in 0x495 seem to be correct as far as I can see.

KarstenBr, 2023-11-12:
@proddy That was quicker as i had expected... I found the total uptime value of my heatpump in message 0x494 at offset 0 in [sec] as 32bit.
000+03:58:27.992 N 144: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAEnergySupplied(0x0494), data: **00 62 A6 84** 00 00 04 12 00 00 01 47 00 00 02 CA 00 00 00 00

In the posted example the value 00 62 A6 84 (hex) corresponds to 6465156 (dec) which translates to 1795,656h which is exactly what the heatpump shows on its diagnostic display as total uptime. I have been monitoring this value for some time and it gets increased by the expected amount of seconds which had passed since the message was last received.

Therefore I would propose to add a new entity "hpuptimetotal" at offset 0 in msg 0x494 with 32bit and unit [sec]. Names could eb as follows:
(EN) "heatpump total uptime time"
(DE) "Wärmpepumpe Gesamtbetriebszeit"

proddy, 2023-11-12:
perfect! I think Michael will add that as he's working in that piece of code, or we wait until it's merged into dev2.

MichaelDvP, 2023-11-12:
Strange, 494 holds only energy readings, 495 the uptimes and 495 offset 0 is uptimeControl. Is this different?  Can a heatpump be up without control? But i'll add it to dev2.

KarstenBr, 2023-11-12:
Message 0x495 with offset 0 (uptimecontrol) in my case shows the uptime with heat provision, i.e. uptimecompheating (offset 8) + uptimecompww (offset 4). Likely it would also add uptimecooling (offset 16) and uptimecomppool (offset 12), but those are 0 as my boiler setup does not have them:
[image]

In contrast to this, 0x494 (offset 0) contains the overall operation time of the heatpump - regardless if it had been providing heat or not, i.e. the duration where it had been connected to electrical power.

proddy, 2023-11-12:
great!

proddy, 2023-11-13:
think we can close this one now.

closed 2023-11-13 by proddy