emsesp/EMS-ESP32 #1419: Unable to flash Firmware to ESP32 32S.

Raybom, 2023-11-13:
### PROBLEM DESCRIPTION

Unable to flash Firmware to ESP32 32S.

### REQUESTED INFORMATION

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

Choose .bin file and try to upload.

### EXPECTED BEHAVIOUR

Normally flash to ESP.

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

The error was: 

Unexpected error: Error while retrieving firmware file 'https://github.com/espressif/arduino-esp32/raw/2.0.5/tools/sdk/esp32/bin/bootloader_dio_40m.bin': HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/arduino-esp32/2.0.5/tools/sdk/esp32/bin/bootloader_dio_40m.bin (Caused by ProxyError('Cannot connect to proxy.', OSError('Tunnel connection failed: 407 Proxy Authentication Required')))

**(Please, remember to close the issue when the problem has been addressed)**

Raybom, 2023-11-13:
[image]

proddy, 2023-11-13:
I couldn't reproduce. I could flash both to a 4MB and 16MB ESP32.

Your PC needs to be connected to the internet - if it is, perhaps something is blocking the download?

closed 2023-11-13 by proddy

Raybom, 2023-11-13:
I have flashed ESP8266 with that and had no problem. And forgot to add label, sorry about that.