emsesp/EMS-ESP32 #1420: Missing sensor values after upgrade (and it's not a rename)

HansRemmerswaal, 2023-11-13:
I upgrade from 3.5 to 3.6.3 dev 6 and lost a number or sensor values.
As example the sensors which defines the start and the end of the heat curve;

- thermostat hc1 design temperature
- thermostat hc1 min flow temperature

I know what are the values and I checked directly in EMS-ESP but can see any sensor with the correct value.

[emsesp_info.json](https://github.com/emsesp/EMS-ESP32/files/13331736/emsesp_info.json)

HansRemmerswaal, 2023-11-13:
Ok... I changed the value on the heat pump and now they are visible. So looks like these are sensor values which are not getting published on a regular basis but only when they are changed. So thats why I could not find the sensors are these settings are normally never changed.

HansRemmerswaal, 2023-11-13:
Closing the issue as it's not an issue. Having said that, maybe there is a broadcast message which contains the current set values...

proddy, 2023-11-13:
@MichaelDvP make RC300Summer fetch true to force the read?

MichaelDvP, 2023-11-13:
In your info 
`"handlers fetched": "0x02A5 0x02B9 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",`
i'm missing 0x2AF and 0x471. These are normally fetched every minute and only removed from fetch list if the thermostat reply with a empty message. 
Can you check with terminal `read 10 2AF` and `read 10 471` what is the output?
Also try restart ems-esp and check the system info for fetched telegrams (after some minutes uptime)

proddy, 2023-11-13:
this may be consistent with what I've been seeing the last days - I'm missing values after a firmware upload and the typically come back after a WebUI 'restart' action. Perhaps a timing issue?

HansRemmerswaal, 2023-11-13:
> Can you check with terminal read 10 2AF and read 10 471 what is the output?

ems-esp:$ read 10 2AF
000+00:43:07.850 N 1: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Summer(0x02AF), data: 00 28 00 00 2D 26 00 00 17 0F 00
ems-esp:$ read 10 471
000+00:43:16.859 N 2: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Summer2(0x0471), data: 01 12 01 01 0A 23 01 0F

> Also try restart ems-esp and check the system info for fetched telegrams (after some minutes uptime)

0+00:03:12.672 I 0: [shell] User session opened on console pty0
000+00:03:14.535 E 1: [emsesp] Last Tx write rejected by host
ems-esp:$ read 10 2AF
ems-esp:$ read 10 471
ems-esp:$

nothing...

      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "handlers fetched": "0x02A5 0x02B9 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",

Then I changed the design temperature on the heatpump from 38 -> 37 -> 38

On the dashboard number of sensors increased with +/- 8 and I see the design and min flow temperature

ems-esp:$ read 10 2AF
000+00:09:08.368 N 8: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Summer(0x02AF), data: 00 28 00 00 2D 26 00 00 17 0F 00
ems-esp:$ read 10 471
000+00:09:11.435 N 9: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Summer2(0x0471), data: 01 12 01 01 0A 23 01 0F

      "handlers fetched": "0x02A5 0x02B9 **0x02AF** 0x029B **0x0471** 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",

harvey637, 2023-11-13:
Hi, 
I also have some problems with missing entities, some examples:
the datapoint "dampedoutdoortemp" ist updated only after ems-esp reboot a few times, than stops updating for several days! ..... (it is defintly changing during the nights/days)
the datapoint "hpheatingon" is not found after new firmware/restart (update/restart happened during active heating), but heating is active now (seen on the physical heatpump and with hpactivity, hppower and consumption counters).

I have lots of datapoints (entities) with influxdb and echarts, so the datapoints in iobroker.objects are still there (not deleted), but not updated anymore. This gives strange graphs, as eg. heating goes on (hpactivity and hppower show new values), but hpheatingon stays "Off".
Strange graph also if "outdoortemp" is going up and down and "dampedoutdoortemp" is a horizontal flat line since several days.

It is at minimum confusing seeing wrong (old, not updated) values, other values trigger some action, that is producing wrong results as NOT beeing updated and not triggering action. 

If you need more info or logs please request, thank you
cu
Harvey

MichaelDvP, 2023-11-13:
> @MichaelDvP make RC300Summer fetch true to force the read?

The fetch ist set on generating the heatingcircuit here: 
https://github.com/emsesp/EMS-ESP32/blob/df9f75a5c99a6360d3cf68b7d207f37e3558b17a/src/devices/thermostat.cpp#L395

I think the initial fetch is removed here, but don't know why.
https://github.com/emsesp/EMS-ESP32/blob/df9f75a5c99a6360d3cf68b7d207f37e3558b17a/src/emsdevice.cpp#L1792

Maybe remove line this and see if it works?

I've removed this in my testbuild https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test, @HansRemmerswaal , @harvey637  could you test if entities are back? (**Breaking change:** This testbuild sorts mixer/solar dhw to a new water devcie!)

HansRemmerswaal, 2023-11-13:
@MichaelDvP it solved it for me with that version. I now see the 
thermostat hc1 design temperature
thermostat hc1 min flow temperature

Thanks!

harvey637, 2023-11-13:
some are good some still bad :-(

I see now:
- hk1.summertemp from RC300

Working (since this morning with test.7: 
- dampedoutdoortemp from RC300

I am still missing e.g.
- heatingon (on) from Hybrid7000iAW 
- coolingon (off) from Hybrid7000iAW

I made a diff from my old costomization save to a new one:
```
diff -U 1 emsesp_customizations(24).json emsesp_customizations(25).json
--- emsesp_customizations(24).json	Wed Nov  8 12:11:13 2023
+++ emsesp_customizations(25).json	Mon Nov 13 14:39:56 2023
@@ -13,3 +13,4 @@
           "08hc1/airhumidity",
-          "03hc1/intoffset|>-12<12"
+          "03hc1/intoffset|>-12<12",
+          "03hc1/battery"
         ]
@@ -27,9 +28,2 @@
           "03boiltemp",
-          "03exhausttemp",
-          "03burngas",
-          "03burngas2",
-          "03flamecurr",
-          "03fanwork",
-          "03ignwork",
-          "03oilpreheat",
           "03heatingactivated",
@@ -40,12 +34,6 @@
           "03pumpdelay",
-          "03burnminperiod",
-          "03burnminpower",
-          "03burnmaxpower",
+          "03setflowtemp",
           "03boilhyston",
           "03boilhystoff",
-          "03boil2hyston",
-          "03boil2hystoff",
-          "03setflowtemp",
           "03setburnpow",
-          "03absburnpow",
           "03selburnpow",
@@ -64,2 +52,4 @@
           "03emergencytemp",
+          "03metertotal",
+          "03metercomp",
           "08uptimecontrol",
@@ -92,6 +82,2 @@
           "08hpactivity",
-          "08hpheatingon",
-          "08hpcoolingon",
-          "03hpwwon",
-          "03hppoolon",
           "03hpbrinepumpspd",
@@ -204,3 +190,3 @@
           "03wwdisinfectday",
-          "03wwdisinfecttime|<1431",
+          "03wwdisinfecthour|<1431",
           "03hybridstrategy",
@@ -223,3 +209,3 @@
           "02hc1/comforttemp",
-          "00hc1/summertemp|>10<30",
+          "08hc1/summertemp",
           "02hc1/designtemp",
@@ -243,6 +229,6 @@
           "03hc1/switchonoptimization",
-          "03hc1/reducemode",
           "03hc1/noreducetemp",
           "03hc1/reducetemp",
-          "03hc1/wwprio"
+          "03hc1/wwprio",
+          "03hc1/cooling"
         ]
```
which shows added and removed entities.
Hope this helps!
cu
Harvey

MichaelDvP, 2023-11-13:
There are some entities removed:
- the burner related are removed for heatpumps and HIU, they never have a value, see #1334 If they have a value for your heatpump, i'll add them back. 
- the hp activity shows different states, these were redundant published (hpheating/hpcooling/..) and removed to save memory, see #1385

harvey637, 2023-11-13:
ok.
burner relativ things are not there, as my device is a heatpump, no problem.

The gone hpheatingon, hpcoolingon, ... are MY minor problem now. I used them, as booleans are "easy".
Now I have the make a case statement to select "heating|cooling|defrost".
Before I could have a "blue" area with coolingon=true, a "red" area with heatingon=true. I guess it is not possible in echarts to change color with value, so I can NOT have a blue area with hpactivity=cooling, and red area with hpactivity=heating.

My personal problem is, that I use(ed) the hpheatingon as kind of background, see my screenshot above. Now it is a bit strange, as the number hpactivity=1(heating) is ok to use with scaling 0-1 (heating==1), but cooling and defrost are out of the graph.

I guess that I will create homematic/iobroker variables, that update according to hpactivity. Sad story is only to loose historical values :-(

But this is my personal problem, has nothing to do with with ems-esp. As long as you keep the different counters :-)))
ciao
Harvey

HansRemmerswaal, 2023-11-13:
As a kind of sanity check I'm checking the history in Home Assistant and found a few missing values.

[image]

[image]

harvey637, 2023-11-13:
Hi,
in the end no big deal to fill MY variables hpheatingon and hpcoolingon from hpactivity, already done:
```
    if {[regexp "1|6" $hpactivity]} {
       set hpheatingon true
    } else {
       set hpheatingon false
    }
```
in the homematic-tcl script, that reads all of the ems-esp into homematic variables.
So completly forget about my complaints, I dont need to touch anything else.
Maybe even better, as defrost is part of the heating now :-)
cu
Harvey

HansRemmerswaal, 2023-11-13:
Ok, found the disinfection hour and the thermostat hc1 mode is suddenly back.
That summermode I do not have,

MichaelDvP, 2023-11-13:
summermode is only for boilers and heatpumps without cooling, on most heatpumps it's renamed to hpoperatingmode (same as summermode, but more possible states). For disinfectiontime we had a complain that it's the hour disinfection starts and not a time/duration.

harvey637, 2023-11-13:
Hi @MichaelDvP
just for the info:_
I have a bosch 7400iAW with rc300 (rc310?) WITH cooling and have the "summermode" (summer|winter).

Maybe the wording is a bit un-informative, but it changes to summer, when the "cooling starts above temperature" is nearly reached. It stays in winter during summer :-) with normal temperatures ....
So it looked to me like a "preparation" for cooling, not yet started real cooling.
This temperature ("cooling above this temperature") can be changes in the rc3xx console, but is not (yet) part of the found ems-esp entities, I can wait :-)
cu
Harvey

HansRemmerswaal, 2023-11-13:
> For disinfectiontime we had a complain that it's the hour disinfection starts and not a time/duration.

Thermostat dhw Disinfection time
Thermostat dhw Disinfection hour

On the heat pump menu it's called "Time of Day". Not sure if the "hour" version is better, but ok. 

The value 720 (min). This is 720 / 60 = 12 hours
And my disinfection starts at 12:00

[image]

MichaelDvP, 2023-11-13:
Ups, you're right, other thermostats have only full hour setting, RC300 interval in 15 min and the german translation of disinfecttime was wrong (duration). I'll go back to disinfecttime and correct german translation..

HansRemmerswaal, 2023-11-14:
Great! 

Even better would be to show the value in time format instead in minutes after 00:00, so 12:00 instead of 720 min. But not sure if such translations is something you normally do. It will also cause issues if someone wants to change the time, then you need to convert back to min... But to be honest, if this field is really important for someone then in Home Assistant they can make a sensor which converts it from min to actual time...

proddy, 2023-11-14:
> Great!
>
> Even better would be to show the value in time format instead in minutes after 00:00, so 12:00 instead of 720 min. But not sure if such translations is something you normally do. It will [...]

show as HH:MM in the WebUI or a Home Assistant dashboard?

harvey637, 2023-11-14:
Hi,
just to add some slightly off topic:
also the "silent" hours in the webUI are in german "22 Stunden" and "5 Stunden".,. which is technically correct as it describes the offset from midnight in hours (so I learned). But in normal life I would say "22:00 Uhr" and "5:00 Uhr". ("Ich nehme den Bus um 18 Stunden und 30 Minuten nach Mitternacht" will sound a bit strange :-)

But this is cosmetics for me, as in Infosystems on the wall I dondt show the native ems-esp webUI but something from HA or iobroker.vis, so I can do any translation to that values.

harvey637, 2023-11-14:
Hi Michael, 
yes, that was me :-)

My favourite will be #2 (keep minutes in api/mqtt) - but please remember that english "hours" in this context tranlates to "Uhr" and not to "Stunden" -> "22:00 Uhr" in webUI.

As you said, it is only used a few times, but it is not string (meaningless letters) but a known value (something with minutes and hours) 

If you ever enter the webUI handling of times (for timer settings like holiday timers or daily timers) it might get more important. But for now I have to change the timers in the app, thanks ems-esp the only need for the app and only once used (for weekly calender heat_ heat_off, holidays are in the prog_1 timerprog, so I just switch to prog_2 for holidays, prog_1 for normal days.
With calenders open in future future ems-esp webUI the handling of "hh:mm" becomes more important.

Thank You for your patience and effort!
cu
Harvey

HansRemmerswaal, 2023-11-14:
Maybe we should move the formatting of the time fields to another issue/improvement to keep this more clean. 
I guess this issue 1420 can be closed now?

harvey637, 2023-11-14:
yes for me - please continue in new thread
Thank You
Harvey

closed 2023-11-19 by proddy