emsesp/EMS-ESP32 #1423: custom variables

proddy, 2023-11-13:
From a discussion with Michael:

The scheduler have entity/command to activate/deactivate the schedule. This was for replace the `forceheatingoff`, sending `selflowtemp 0` every minute to stop the boiler, and disable scheduler to activate the boiler again.
Maybe it's a nice addition to also manipulate the schedule command value via command. But difficult to add in the command structure. 

I wonder if we can add custom RAM entities only stored in ems-esp and use them for setting a variable value. E.g. in custom use devcie/telegram type 0x0 for storing 256 bytes. Custom entity: `myseltemp, device 0, type 0, offset 0, UINT`, Scheduler: `every minute, command: boiler/seltemp, value: custom/myseltemp`. Then the flowtemp can be set via mqtt: custom/myseltemp and is used by the scheduler.

_Originally posted by @MichaelDvP in https://github.com/emsesp/EMS-ESP32/pull/1418#discussion_r1390784057_

proddy, 2023-11-13:
I see, That's quite clever. How about instead of using custom entities to store variables we could create a new type called Variables. Maybe limit to 10 max, as strings with a default value that is stored in LittleFS and loaded at boot.

proddy, 2024-01-14:
Something like:
- In the Custom Entity web page you can choose to read/write the value from EMS (with a deviceID, typeID, Offset) or as a Variable with default value. It means extending Custom Entity to include a Custom Entity type, and it's persisted in the FS. The value could be written to NVS or just kept in memory (which means loosing after a reboot)
- Add a new API call to read or write a value to the Custom Entity
- Modify the Scheduler to also accept a Custom Entity as a value, instead of a hardcoded value

MichaelDvP, 2024-01-14:
Comments to:
1. `Custom variable` only with name and default value. The value can be stored as string (no need to set the type), the command-values are always processed as char*. Default value should be stored in FS, changing the value by API should only use RAM (to avoid wear of FS). Setting new default only va web page.
2. API command is already implemented for custom values (`call custom <name> <value>`, `/api/custom/<name> {"value":"<value>"}`)
3. The scheduler already accepts commands (that return a value) as value. e.g. `boiler/flowtemp/value` returns the actual flowtemp can be processed as value parameter.

proddy, 2024-01-14:
excellent, so we only need to implement #1 !

proddy, 2024-01-22:
@MichaelDvP my idea with Variables was to be able to use them in Values

For example if I created a custom entity called 'foo' I could use this in the Scheduler as a value like `{foo}`, so its dynamic/variable and not static/hardcoded.

proddy, 2024-01-22:
I didn't realize that, I need to look at the code again. I thought WebSchedulerService::command() just called the command. But cool!

proddy, 2024-02-04:
tested, works great. I'll add some notes to the doc.

closed 2024-02-04 by proddy