emsesp/EMS-ESP32 #1424: Entry for hc1 own program 1 switchtime

mwolle, 2023-11-14:
Hello, i want to change the following entry: "hc1 own program 1 switchtime". I used this syntay for entrys: 00 mo 06:30 on // 01 mo 17:30 off // 03 tu 06:30 on...-> moep the third entry will not safed, i get an error "Not found". I used a Buderus RC35 controller.

Input: 
[image]

result:
[image]

proddy, 2023-11-14:
I will try and reproduce. Please send system info (https://emsesp.github.io/docs/Support/) so I know which thermostat you are using.

proddy, 2023-11-14:
I started writing some test code for this since I don't have an RC3x and there was nothing in our test.cpp. So glad you tried it out!

What also should work, is from the console (after a su) `call thermostat switchtime1 03 tu 06:30 on` and see what error comes back? We could also uncomment those LOG.WARN lines `Thermostat::set_switchtime()` to trace what is happening.

mwolle, 2023-11-14:
I check the system version, i thought i had updated it on 3.6.2, but i test on 3.5.1. I update ESP to 3.6.2 and now i can enter the entrys without error. But after Update, i can't scroll the Entitylist.
But the major issue is solved with the 3.6.2 Update.
[image]

closed 2023-11-14 by proddy