emsesp/EMS-ESP32 #1427: wwtapactivated - Message if no shower is used ?

zibous, 2023-11-15:
### PROBLEM DESCRIPTION

I don't understand why the message appears even when no shower is being used. 
Could it be that this has something to do with the boiler charge?

```lua
2023-11-14 20:23:01.313 W 846: [command] Calling command 'boiler/values'
2023-11-15 02:24:52.001 W 847: [command] Calling command 'boiler/values'
2023-11-15 05:37:52.076 W 848: [command] Calling command 'boiler/values'
2023-11-15 05:43:23.360 W 849: [command] Calling command 'boiler/values'
2023-11-15 05:43:40.022 W 850: [command] Calling command 'boiler/wwtapactivated' (Durchlauferhitzer aktiv) with value false
2023-11-15 05:43:50.023 W 851: [command] Calling command 'boiler/wwtapactivated' (Durchlauferhitzer aktiv) with value true
2023-11-15 05:49:24.905 W 852: [command] Calling command 'boiler/values'
2023-11-15 05:51:06.560 W 853: [command] Calling command 'boiler/wwtapactivated' (Durchlauferhitzer aktiv) with value false
2023-11-15 05:51:16.561 W 854: [command] Calling command 'boiler/wwtapactivated' (Durchlauferhitzer aktiv) with value true
2023-11-15 05:51:31.566 I 855: [shower] finished with duration 425006
2023-11-15 05:55:57.479 W 856: [command] Calling command 'boiler/values'
2023-11-15 05:56:48.648 I 857: [shower] finished with duration 295938
2023-11-15 06:28:36.642 W 858: [command] Calling command 'boiler/values'
2023-11-15 08:19:28.867 W 859: [command] Calling command 'boiler/values'
```

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "type": "settings",
  "System": {
    "version": "3.6.2"
  },
  "Network": {
    "ssid": "*****",
    "password": "******",
    "hostname": "ems-heizung.local",
    "static_ip_config": false,
    "enableIPv6": false,
    "bandwidth20": false,
    "tx_power": 20,
    "nosleep": false,
    "enableMDNS": true,
    "enableCORS": false,
    "CORSOrigin": "*"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "password": "ems-esp-neo",
    "channel": 1,
    "ssid_hidden": false,
    "max_clients": 4,
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": true,
    "host": "emqx.local",
    "port": 12883,
    "base": "ems_heizung",
    "username": "****",
    "password": "*****",
    "client_id": "ems-esp32",
    "keep_alive": 60,
    "clean_session": true,
    "entity_format": 0,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 60,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "publish_time_heartbeat": 60,
    "mqtt_qos": 0,
    "mqtt_retain": true,
    "ha_enabled": true,
    "nested_format": 1,
    "discovery_prefix": "homeassistant",
    "discovery_type": 0,
    "publish_single": false,
    "publish_single2cmd": false,
    "send_response": false
  },
  "NTP": {
    "enabled": true,
    "server": "europe.pool.ntp.org",
    "tz_label": "Europe/Vaduz",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": true,
    "port": 8266,
    "password": "ems-esp-neo"
  },
  "Security": {
    "jwt_secret": "******",
    "users": [
      {
        "username": "admin",
        "password": "******",
        "admin": true
      },
      {
        "username": "guest",
        "password": "guest",
        "admin": false
      },
      {
        "username": "peter",
        "password": "*******",
        "admin": true
      }
    ]
  },
  "Settings": {
    "version": "3.6.2",
    "locale": "de",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "syslog_enabled": true,
    "syslog_level": 9,
    "trace_raw": false,
    "syslog_mark_interval": 0,
    "syslog_host": "10.1.1.96",
    "syslog_port": 514,
    "shower_timer": true,
    "shower_alert": true,
    "shower_alert_coldshot": 10,
    "shower_alert_trigger": 7,
    "rx_gpio": 23,
    "tx_gpio": 5,
    "dallas_gpio": 18,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "low_clock": false,
    "telnet_enabled": true,
    "notoken_api": false,
    "readonly_mode": false,
    "analog_enabled": true,
    "pbutton_gpio": 0,
    "solar_maxflow": 30,
    "board_profile": "S32",
    "fahrenheit": false,
    "bool_format": 1,
    "bool_dashboard": 1,
    "enum_format": 1,
    "weblog_level": 6,
    "weblog_buffer": 50,
    "weblog_compact": true,
    "phy_type": 0,
    "eth_power": 0,
    "eth_phy_addr": 0,
    "eth_clock_mode": 0,
    "platform": "ESP32"
  }
}
```

### TO REPRODUCE
see log and syslog

[image]

MichaelDvP, 2023-11-15:
Yes, in flow systems it shows the active (running) tapwater, in buffer systems it shows the charging of the buffer. Also the shower timer shows the charging time. It has nothing to do with shower, i wanted to remove it for buffer systems but within a few hours the user complains about missing shower timer.

zibous, 2023-11-15:
@MichaelDvP 

Thanks for the explanation. It just unsettled me because no one uses the shower so early in the morning. 
I'll think of a rule for the home assistant that excludes this.

FYI:
System is very stable since I upgraded to 3.2 and used the emqx mqtt brocker. The eclipse-mosquitto kept crashing.

Thanks

closed 2023-11-15 by zibous