emsesp/EMS-ESP32 #1431: RC35 /GB135  Error switching hc1 mode to day/night/auto

krambox, 2023-11-16:
### PROBLEM DESCRIPTION

I have changed from the RC30 to the RC35 (Buderus GB135 boiler) and can no longer control the heating circuit with  ems-esp. There is an error when switching hc1 mode to day/night/auto. Switching from dhw mode still works. 

### REQUESTED INFORMATION

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "003+05:16:29.050",
    "uptime (seconds)": 278189,
    "free mem": 147,
    "max alloc": 91,
    "free app": 437,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -36,
    "IPv4 address": "192.168.1.4/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "254.128.0.0",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 173507,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 2,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 2,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": true,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": true,
    "publish2command": false,
    "send response": true
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 55522,
    "API fails": 7
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 338727,
    "bus reads (tx)": 57152,
    "bus writes (tx)": 131,
    "bus incomplete telegrams": 10,
    "bus reads failed": 36,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": true,
    "shower alert coldshot": 10,
    "shower alert trigger": 7,
    "hide led": false,
    "notoken api": true,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GB125/GB135/MC10",
      "device id": "0x08",
      "product id": 72,
      "version": "02.04",
      "entities": 75,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35",
      "handlers ignored": "0x01 0x03 0xFD 0xFE 0x1F 0x20"
    },
    {
      "type": "thermostat",
      "name": "RC35",
      "device id": "0x10",
      "product id": 86,
      "version": "01.18",
      "entities": 60,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x13 0x42 0x48 0x47 0x49 0x4C 0x51 0x53 0x56 0x5B 0x5D 0x60 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
      "handlers pending": "0x52 0x5C",
      "handlers ignored": "0x35 0x01 0x1A 0x05 0x33 0xA4"
    },
    {
      "type": "gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x0D",
      "product id": 189,
      "version": "01.10",
      "entities": 0,
      "handlers ignored": "0x37 0x33"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.00",
      "entities": 0,
      "handlers ignored": "0x16 0x05 0x29"
    }
  ]
}
```

### TO REPRODUCE

Switch in the Dashboard h1 mode to other value

### LOG FILE

```
2023-11-16 14:24:29.537 WARNING 56073: [command] Calling command 'boiler/values'
2023-11-16 14:24:30.400 TRACE 56074: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 0B 01 F9 64 00 00 00 00 00 01 DF 83 00 80 00 00 00 FF 30 48 00 00 FF 00 00 00 E5
2023-11-16 14:24:30.681 TRACE 56075: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 0B 00 (offset 27)
2023-11-16 14:24:31.089 TRACE 56076: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 3C 01 DF 83 00 21 00 00 03 00 06 91 0A 00 79 7D
2023-11-16 14:24:33.260 TRACE 56077: [emsesp] thermostat(0x10) -B-> All(0x00), RC35Monitor(0x3E), data: 00 00 22 00 D8 00 00 00 00 00 00 00 00 11 05 00
2023-11-16 14:24:34.760 WARNING 56078: [command] Calling command 'thermostat/values'
2023-11-16 14:24:39.001 TRACE 56079: [emsesp] gateway(0x0D) -R-> thermostat(0x10), RC35Set(0x3D), length: 0x01 (offset 2)
2023-11-16 14:24:39.001 TRACE 56079: [emsesp] gateway(0x0D) -R-> thermostat(0x10), RC35Set(0x3D), length: 0x01 (offset 2)
2023-11-16 14:24:39.028 WARNING 56080: [command] Calling command 'boiler/values'
2023-11-16 14:24:39.028 TRACE 56081: [emsesp] thermostat(0x10) -W-> gateway(0x0D), RC35Set(0x3D), data: 2A (offset 2)
2023-11-16 14:24:39.441 TRACE 56082: [emsesp] gateway(0x0D) -R-> thermostat(0x10), RC35Set(0x3D), length: 0x01 (offset 1)
2023-11-16 14:24:39.478 TRACE 56083: [emsesp] thermostat(0x10) -W-> gateway(0x0D), RC35Set(0x3D), data: 22 (offset 1)
2023-11-16 14:24:40.113 TRACE 56084: [emsesp] gateway(0x0D) -R-> thermostat(0x10), RC35Set(0x3D), length: 0x01 (offset 7)
2023-11-16 14:24:40.227 TRACE 56085: [emsesp] thermostat(0x10) -W-> gateway(0x0D), RC35Set(0x3D), data: 00 (offset 7)
2023-11-16 14:24:40.413 TRACE 56086: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 0B 01 FA 64 00 00 00 00 00 01 DF 83 00 80 00 00 00 FF 30 48 00 00 FF 00 00 00 E5
2023-11-16 14:24:40.693 TRACE 56087: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 0B 00 (offset 27)
2023-11-16 14:24:40.961 TRACE 56088: [emsesp] gateway(0x0D) -R-> thermostat(0x10), WWSettings(0x37), length: 0x01 (offset 2)
2023-11-16 14:24:41.102 TRACE 56089: [emsesp] thermostat(0x10) -W-> gateway(0x0D), WWSettings(0x37), data: 02 (offset 2)
2023-11-16 14:24:41.202 TRACE 56090: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 3C 01 DF 83 00 21 00 00 03 00 06 91 0A 00 79 7D
2023-11-16 14:24:43.517 TRACE 56091: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 0B 01 F9 64 00 00 00 00 00 01 DF 83 00 80 00 00 00 FF 30 48 00 00 FF 00 00 00 E5
2023-11-16 14:24:43.798 TRACE 56092: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 0B 00 (offset 27)
2023-11-16 14:24:44.285 WARNING 56093: [command] Calling command 'thermostat/values'
2023-11-16 14:24:46.477 TRACE 56094: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
2023-11-16 14:24:47.049 WARNING 56095: [command] Calling command 'thermostat/mode' (mode) with value day and id 1 on device 0x10
2023-11-16 14:24:47.049 ERROR 56096: [emsesp] Write command failed callback function failed (Error)
2023-11-16 14:24:49.506 WARNING 56097: [command] Calling command 'boiler/values'
2023-11-16 14:24:49.506 WARNING 56097: [command] Calling command 'boiler/values'
2023-11-16 14:24:50.426 TRACE 56098: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 0B 01 F9 64 00 00 00 00 00 01 DF 83 00 80 00 00 00 FF 30 48 00 00 FF 00 00 00 E5
2023-11-16 14:24:50.706 TRACE 56099: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 0B 00 (offset 27)
2023-11-16 14:24:51.115 TRACE 56100: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 3C 01 DF 83 00 21 00 00 03 00 06 91 0A 00 79 7D
```

proddy, 2023-11-16:
hmm. Any ideas @MichaelDvP ? I could write a test in test.cpp for this

MichaelDvP, 2023-11-17:
Sorry, i have the identical RC35, product-id 86, version 01.18 and can not reproduce. (i have only hc2). Tried with dev2 and v3.6.2.
@krambox How do you send the command? Api/mqtt/telnet? (I see a lot of api calls and no_token enabled in your settings)

krambox, 2023-11-17:
I have tried it via the dashboard interface and MQTT. Water worked and the RC30 also worked. I am currently changing my smarthome setup from KM200 to emsesp. The plan is to switch over via the API or MQTT.

proddy, 2023-11-17:
turn off those API calls which are every 5 seconds, go to Telnet and test from there to see what the log says `su` `call thermostat mode auto`

proddy, 2024-02-22:
please re-open if its still an issue

closed 2024-02-22 by proddy