emsesp/EMS-ESP32 #1433: Cannot Connect to WiFi - always in "fallback" mode

kriger1965, 2023-11-17:
Since a view weeks i have problem to connect Wifi.´

Now i can not connect it to any Wifi in my environment ... because "http://192.168.4.1/network/scan"  no wifi router is listed ..

here the history
- at first I changed the router. some old Router couldnt connect to.
- afterward I found a old fritzbox where it worked, bad get disconnected after 8 or 10 hours. with a smart (WiFi) Relais i disconnected the ems-esp to make him reboot to let him reconnect again
 this worked but was no final solution ...
- now i flashed with the lates beta-version EMS-ESP-3_6_3-dev_9-ESP32.bin  but I have main Problem ! !!!    >>>>" No Wifi Network found "
- I hav manually editied the WiFi SSid and tryed to reboot - but it did not connect to the wifi and fall back into the "access point Activ" - mode 
-
i need help to make it work again as in the beginning or change to the etheret - Version
[image]
[image]

proddy, 2023-11-17:
which version were you on before? Double check in System because sometimes it uses an older version hidden in another partition
please also send the support info if you can

The 'Problem loading' I will need to look into what is happening

What you could do is connect the Gateway board or the ESP32 directly to your PC, via USB and open a serial port. Then after `su` do a `wifi ssid <SSID>` and `wifi password` and then `wifi reconnect` and watch the screen for errors

kriger1965, 2023-11-17:
I found in the Log:

000+00:08:00.174 W 118: [emsesp] WiFi disconnected. Reason: assoc leave (8)

kriger1965, 2023-11-17:
Hi, i am using the EMS-ESP-flasher - utility - which erases also als settings ... so i think it is really the "3.6.3-9 Version"

kriger1965, 2023-11-17:
i did connect per USB - Port to flash the Board with the last version ...

kriger1965, 2023-11-17:
also getting sometime the result of '"no wifi´w found" ..
[image]

...but with the other PC I can see and connect the Wifi´s surrounding me ..

kriger1965, 2023-11-17:
I am now connected with Putty:

EMS-ESP version 3.6.3-dev.9
System:
 Version: 3.6.3-dev.9
 Language: en
 Board profile: S32
 Uptime: 000+00:26:36.806
 SDK version: 4.4.6
 CPU frequency: 240 MHz
 Free heap/Max alloc: 186 KB / 107 KB
 App used/free: 1566 KB / 6562 KB
 FS used/free: 24 KB / 40 KB

Network:
 WiFi Network: Disconnected

Syslog:
 Syslog: disabled

kriger1965, 2023-11-17:
after doing as you declaired:
su
# set wifi ssid XXXX
# set wifi password
XXXXX
# wifi reconnect

....................................

I have got this in the log:

ems-esp:# wifi reconnect
000+00:02:55.183 I 101: [system] WiFi reconnecting...
000+00:02:57.608 W 102: [telegram] Incomplete Rx: 0E 06 0C 11 1E 01 04 19 1A 0B 0D 07 0A
000+00:02:59.373 W 103: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:00.749 W 104: [telegram] Incomplete Rx: 1A 1B 0C 04 05
000+00:03:02.375 W 105: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:03.603 W 106: [telegram] Incomplete Rx: 19 04 07 19 01 07
000+00:03:05.379 W 107: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:06.605 W 108: [telegram] Incomplete Rx: 09 0E 1A 19 14 05
000+00:03:08.380 W 109: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:11.385 W 110: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:12.764 W 111: [telegram] Incomplete Rx: 10 1A 04 1B 0F 13 09 01
000+00:03:14.390 W 112: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:15.466 W 113: [telegram] Incomplete Rx: 08 0A 04 19 17
000+00:03:17.391 W 114: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:18.620 W 115: [telegram] Incomplete Rx: 09 01 01 0F 15 0B 07 07 02
000+00:03:20.396 W 116: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:03:21.776 W 117: [telegram] Incomplete Rx: 01 08 13 07 0F 01 09

MichaelDvP, 2023-11-17:
Are your wifi APs sending on 2.4GHz?

kriger1965, 2023-11-17:
Yes - only 2.4 is in this area available

kriger1965, 2023-11-17:
just tested again ....

ems-esp:# wifi reconnect
000+00:09:02.781 I 395: [system] WiFi reconnecting...
000+00:09:02.781 W 396: [telegram] Incomplete Rx: D7 02 CC 01 78 03 93
000+00:09:02.781 W 397: [emsesp] WiFi disconnected. Reason: 4way handshake timeout (15)
000+00:09:02.781 W 398: [telegram] Incomplete Rx: 5B 0C C7 09 D2 09 AB 1D 9C 08 49 08 4D 08
000+00:09:04.904 W 399: [telegram] Incomplete Rx: 27 03 60 02 3D 08 F7 15 F0 04 FD 30 99
000+00:09:05.907 W 400: [telegram] Incomplete Rx: 45 07 87 04 06 0B 9F 14 EB 15 B2 1A 5E 07 AF 0A DB 06 DF
000+00:09:06.891 W 401: [telegram] Incomplete Rx: F7 08 A7 20 11 07 C5 02 A9 26 18 0E 0F 06 87 0D 0F 12 9B
000+00:09:06.891 W 402: [emsesp] WiFi disconnected. Reason: assoc leave (8)
000+00:09:07.168 W 403: [telegram] Incomplete Rx: D5 03 2B 1B
000+00:09:07.926 W 404: [telegram] Incomplete Rx: 43 31 33 17 E5 35 5A 1D F2 07 38 07 F5
000+00:09:08.398 W 405: [telegram] Incomplete Rx: 9F 1C 3A 03 69 03 FD 01
000+00:09:08.930 W 406: [telegram] Incomplete Rx: 98 05 91 02 C5 01 D5 15 1B 1D
000+00:09:09.909 W 407: [telegram] Incomplete Rx: A5 23 A7 0A D7 0A C9 08 1B 01 7F 18 64 05 B5 14 79
000+00:09:09.909 W 408: [emsesp] WiFi disconnected. Reason: assoc leave (8)
000+00:09:09.949 W 409: [telegram] Incomplete Rx: 05 DB 13 03 01 5C
000+00:09:10.636 W 410: [telegram] Incomplete Rx: E7 09 F3 10 1D 0D 95 03 39 06 01 02 49 05 15 0C FB 18 C4 41 4C 20 AF 8C A4

kriger1965, 2023-11-17:
with the other wifi "Q4u_HA"

MichaelDvP, 2023-11-17:
> 4way handshake timeout (15)

There is problem with your password, double check!

kriger1965, 2023-11-17:
just doing again with ssid Q4u_HA_VT

....
ems-esp:# wifi reconnect
000+00:12:48.800 I 759: [system] WiFi reconnecting...
000+00:12:48.800 W 760: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:12:51.000 W 761: [telegram] Incomplete Rx: 0F 01 55 13 0B 1D DA 01 0E 0B 01
000+00:12:52.772 W 762: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:12:55.777 W 763: [emsesp] WiFi disconnected. Reason: no AP found (201)
000+00:12:56.865 I 764: [emsesp] Starting NTP service
000+00:12:56.865 I 765: [emsesp] WiFi connected with IP=192.168.1.120, hostname=ems-esp
ems-esp:# ems-esp:# wifi reconnect
Command not found
ems-esp:# 000+00:12:48.800 I 759: [system] WiFi reconnecting...
Command not found
ems-esp:# 000+00:12:48.800 W 760: [emsesp] WiFi disconnected. Reason: no AP found (201)
Command not found
ems-esp:# 000+00:12:51.000 W 761: [telegram] Incomplete Rx: 0F 01 55 13 0B 1D DA 01 0E 0B
Command not found
ems-esp:# 000+00:12:52.772 W 762: [emsesp] WiFi disconnected. Reason: no AP found (201)
Command not found
ems-esp:# 000+00:12:55.777 W 763: [emsesp] WiFi disconnected with IP=192.168.1.120, hostna
Command not found
ems-esp:# ems-esp:#
Command not found
000+00:13:12.006 W 766: [telegram] Incomplete Rx: CE 13 7B 05 BD 03 1D 1D 6E 0D 1D 01 14 0B 28 0D 0A 06 EF 1B 10 10 2D 1A 08 04 11 07 59 01 3C
ems-esp:#

kriger1965, 2023-11-17:
just seems to be connected ...
[image]

kriger1965, 2023-11-17:
just reproducing the problem ... if i reconnect with the Web-gui back to the  Wifi of the Fritz-Box named "Q4u_HA" the problems are back: 
1) no Wifi Accesspoints listed in the Web Gui
2)  WiFi disconnected. Reason: assoc leave (8)  ...is reportetd in the log
no reconnect from the web Gui is possible any more ...

kriger1965, 2023-11-17:
not possible to reconnect to the WiFi "Q4u_HA_VT"  again also reproducing the procedure with serial connecton per usb 

always getting

000+00:03:45.719 W 83: [emsesp] WiFi disconnected. Reason: no AP found (201)

kriger1965, 2023-11-17:
I have the same password on the wifi´s ...

kriger1965, 2023-11-17:
I will get crazy ... now it is not possible to cecct to the ems-esp  access point ...

C:\Users\Harald>ping 192.168.4.1

Ping wird ausgeführt für 192.168.4.1 mit 32 Bytes Daten:
Antwort von 192.168.1.15: Zielhost nicht erreichbar.
Antwort von 192.168.1.15: Zielhost nicht erreichbar.
Antwort von 192.168.1.15: Zielhost nicht erreichbar.
Antwort von 192.168.1.15: Zielhost nicht erreichbar.

Ping-Statistik für 192.168.4.1:
    Pakete: Gesendet = 4, Empfangen = 4, Verloren = 0
    (0% Verlust),

kriger1965, 2023-11-17:
... getting CRAZY !!!!

I redo the procedure:
1) flash the ESP with USB - newes Version
2) connect with putty ... and enter the credentioal for WiFi "Q4u_HS_VT"  which seems to be the onlyrouter that works ...
3) getting the wifi connection working

┌───────────────────────────────────────┐
│ EMS-ESP version 3.6.3-dev.9           │
│ https://github.com/emsesp/EMS-ESP32   │
│                                       │
│ type help to show available commands  │
│ use su to access Admin commands       │
└───────────────────────────────────────┘

000+00:00:00.000 I 0: [shell] User session opened on console ttyS0
000+00:00:00.000 I 1: [emsesp] Starting EMS-ESP version 3.6.3-dev.9
000+00:00:00.000 I 2: [emsesp] Last system reset reason Core0: Power on reset, C                   ore1: APP CPU reset by PRO CPU
000+00:00:00.000 I 3: [emsesp] Loading board profile S32
000+00:00:00.000 I 4: [emsesp] Starting Scheduler service
000+00:00:00.000 I 5: [emsesp] Starting Custom entity service
000+00:00:00.000 I 6: [temperaturesensor] Starting Temperature sensor service
000+00:00:00.000 I 7: [analogsensor] Starting Analog sensor service
000+00:00:00.000 I 8: [emsesp] Loaded EMS device library (117 records)
000+00:00:10.000 I 9: [emsesp] Starting Access Point with captive portal on 192.                   168.4.1
ems-esp:$ su
000+00:00:53.200 N 10: [shell] Admin session opened on console ttyS0
ems-esp:# set wifi ssid Q4u_HA_VT
Use `wifi reconnect` to save and apply the new settings
ems-esp:# set wifi password
Enter new password:
Retype new password:
Use `wifi reconnect` to save and apply the new settings
000+00:01:48.464 I 11: [emsesp] WiFi connected with IP=192.168.1.120, hostname=ems-esp
000+00:01:48.493 I 12: [emsesp] mDNS responder service started
ems-esp:# wifi reconnect
000+00:01:50.867 I 13: [system] WiFi reconnecting...
000+00:01:50.867 I 14: [emsesp] WiFi connection dropped, stopping NTP
000+00:01:50.867 W 15: [emsesp] WiFi disconnected. Reason: assoc leave (8)
000+00:01:52.456 I 16: [emsesp] WiFi connected with IP=192.168.1.120, hostname=ems-esp
000+00:01:52.499 I 17: [emsesp] mDNS responder service started
ems-esp:#

kriger1965, 2023-11-17:
Ping wird ausgeführt für 192.168.1.120 mit 32 Bytes Daten:
Antwort von 192.168.1.120: Bytes=32 Zeit=41ms TTL=255
Antwort von 192.168.1.120: Bytes=32 Zeit=58ms TTL=255
Antwort von 192.168.1.120: Bytes=32 Zeit=61ms TTL=255
Antwort von 192.168.1.120: Bytes=32 Zeit=78ms TTL=255

Ping-Statistik für 192.168.1.120:
    Pakete: Gesendet = 4, Empfangen = 4, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 41ms, Maximum = 78ms, Mittelwert = 59ms

....but seems for me pretty "unlogic"

kriger1965, 2023-11-17:
ok, you are right - with the other Wifi "Q4u_HA" i had a password issue - but it is fixed now

kriger1965, 2023-11-18:
Hi, since yesterday after flashing the new version 3.6.3-dev.10 the item seems to stay connected and the surrounding networks can be detected fine. 
Then installed the rest:
The MQTT Services are working  ...but I cannot see the device in homeassistant . What Is wrong ?

[image]

it seems als "not connected" 
[image]

proddy, 2023-11-18:
send us the support info so we can check the settings. Also what do you see in HA under MQTT devices?

kriger1965, 2023-11-18:
hi, thanks for the quick response.  
what do you mean with support infos?

as you can see above - this ist what i can see with the mqtt explorer in homeassistant ...but the itemas are not displayed in the entity´s list ...

kriger1965, 2023-11-18:
[image]
[image]
[image]

kriger1965, 2023-11-18:
do you meant this above with "support info" ?

kriger1965, 2023-11-18:
if I filter in homeassistant the entitys with "ems-esp" nothing listed: 
[image]

proddy, 2023-11-18:
There's a process for asking for help. See https://emsesp.github.io/docs/Support/

An export of the system info helps us. 

For HA you need to manually add this to a dashboard. See https://emsesp.github.io/docs/Home-Assistant/

proddy, 2023-11-18:
I also see from the above screenshots you haven't enabled Mqtt Discovery. 

And you're using an old version. Recommend you upgrade to the latest dev.

kriger1965, 2023-11-18:
ok - i will enable MQTT Discovery   
...bit in ems esp I  have the latest version (3.6.3-dev.10) installed:

kriger1965, 2023-11-18:
now i can see the ems-esp entitys ...

thanks for the support link

what do you mean with ...not latest version - ems-esp or mqtt explorer ?

kriger1965, 2023-11-18:
Hello - sorry to come back to this point of ugly message reprots....

i have tryed to reconfigure and connect to my fritz box router as before and ....
- the same issues appeared again:
-1) I cannot connect to the device fallback wifi ...getting no IP ...
     therefore I configured the WLAN Adapter manually as 192.168.4.100. / 255.255.255.0
     ... this is the only way to get the Web Gui (SOMETIMES)
[image]

- 2) OFTEN Could not conntect to Device Fallback WiFi
- 3) Connect per Putty to set the wifi connection
[image]
... not working
- 4) this was the rison to flasch the device again local per USB
 flasched und could connect to the local fallback wifi where i had bevore so much problems ...   ...the system tryes to connect to http://www.msftconnecttest.com/redirect   ... not working  ... but can connect to 192.168.4.1 it works fine now ...getting also an Ip and a stable  ping  .... much better than bevore

-5) But the time is too short - i get disconnected from the fallback wifi too fast ...
or to be clearer : When I am trying to make the "Network Scanner" to configure the wifi connection the item diceonnects my PC from his WLAN

[image]

... so i need more help from you

kriger1965, 2023-11-18:
ok - tryed again ...
- therefore now the connection seems to be stable (ping 192.168.4.1) is okay
- network scan is working
- connection to local Wifi Router in the room seems to connect and getting his ol ip from the dhcp server 
-
... okay -  I will try to reconfig as in the past

step by step

kriger1965, 2023-11-18:
.... everythin looking good (better than before) ... I will trace the reconnection problem and come back if the connection is dropping / disconnection again ... if not there is all okay

proddy, 2023-11-18:
sorry that you're having so many issues. How can I help?

proddy, 2023-11-19:
I will close for now - please re-open or move to a discussion if you need further help.

closed 2023-11-19 by proddy