emsesp/EMS-ESP32 #1440: hc1 mode off when set to manual

philrich, 2023-11-19:
### PROBLEM DESCRIPTION

After Firmware Update from 3.6.5 to 3.6.6 th climate.hc1 Entity in Homeassistant went to Unavailable state (it was set to manual heating before). In MQTTExplorer there was no `hc1.mode` in `thermostat_data`). After switching to auto mode on the thermostat the climate entity appeared in homeassistant (and `hc1.mode` in `thermostat_data`). But it is not possible to switch to manual mode anymore. If i try to switch to manual mode (heat) it goes to off.
I also tried to switch on manual mode via telnet but same problem.

### REQUESTED INFORMATION

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.6.3",
    "platform": "ESP32",
    "uptime": "000+00:15:16.916",
    "uptime (seconds)": 916,
    "free mem": 151,
    "max alloc": 95,
    "free app": 420,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp2",
    "RSSI": -55,
    "IPv4 address": "10.0.2.80/255.255.255.0",
    "IPv4 gateway": "10.0.2.1",
    "IPv4 nameserver": "253.157.57.218",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": true,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Vienna"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 481,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 0,
    "publish time thermostat": 0,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": true,
    "syslog started": true,
    "syslog level": "info",
    "syslog ip": "10.0.2.13",
    "syslog queue": 0
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 1147,
    "bus reads (tx)": 380,
    "bus writes (tx)": 3,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": false,
    "hide led": true,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 26
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "35.03",
      "entities": 70,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35"
    },
    {
      "type": "thermostat",
      "name": "RC200/CW100",
      "device id": "0x18",
      "product id": 157,
      "version": "41.05",
      "entities": 42,
      "handlers received": "0x06 0xA2 0x031D",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A",
      "handlers pending": "0xA3 0x12 0x13 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x04",
      "handlers ignored": "0x1A 0x35 0x23 0xF7 0xBF"
    },
    {
      "type": "controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "35.03",
      "entities": 0
    }
  ]
}
```

### TO REPRODUCE

Update Firmware from Version 3.6.2 to 3.6.3.

### EXPECTED BEHAVIOUR

Switching thermostat to **manual** should be possible. If thermostat is already in manual mode, climate entity should not be unavailable.

### ADDITIONAL CONTEXT

Downgrading to 3.6.2 fixes the behavior.

proddy, 2023-11-20:
could you use the Telnet Console and 'call thermostat mode manual' with logging enabled so we can see any error messages and what Tx is being sent.

MichaelDvP, 2023-11-20:
Something gone wrong when adding the changes for #1398. Please check if the fix from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest works. 
If not, please send the complete telegram 0x2B9 (`read 10 2b9` in terminal or in web log all and wait for the telegram).

philrich, 2023-11-20:
> Something gone wrong when adding the changes for #1398. Please check if the fix from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest works.

This seems to work. Manual mode is working again. Thanks.

I also noticed that I now have a entity select.thermostat_hc1_control which is set to RC310, but on the devices screen it says "RC200/CW100". I can select RC200 in the entity select box but it stays at RC310. It doesnt seem to have anything to do with this problem but looks incorrect.

MichaelDvP, 2023-11-21:
Thanks for the feedback, i'll make a PR. 

The control entitiy is to set the control to a external roomthermostat as remote control. Setting RC310 means "this thermostat" , the other settings different types of remote. I think the RC200 can not act as master for remotes. You can use the customization to remove the unused setting.

commit aed45968 (2023-11-21): set/read thermostat mode for RC100-RC300, fix #1440
commit cff5b8c9 (2023-11-21): Merge pull request #1444 from MichaelDvP/dev

set/read thermostat mode for RC100-RC300, fix #1440 and fix #1442
commit 5ad03fac (2023-11-21): Merge pull request #1445 from proddy/dev

patch for 3.6.3 with fixes #1440 #1442
commit 6dba452a (2023-11-21): patch for 3.6.3 with fixes #1440 #1442
closed 2023-11-21 by proddy