emsesp/EMS-ESP32 #1442: After Update from 3.6.2 to 3.6.3 some entities are not available in HA any more

DoubleYou2, 2023-11-20:
**PROBLEM DESCRIPTION**

RC310 / Buderus GB125 / EMS-ESP V3.6.3 and HA latest version:

After updating from V3.6.2 (or from 3.6.3-dev6c) some entities are not available in home assistant (HA) any more.
Some entities are incomplete (regarding attributes) and unavailable also.

Here is an example for climate.thermostat_hc1.

This screenshot shows state of entity climate.thermostat_hc1 with V3.6.3-dev6c (similar to V3.6.2):

[image]

This screenshot shows state of entity climate.thermostat_hc1 with V3.6.3 :
[image]

From my (user) perspective this issue is not caused by home assistant. Replacing firmware 3.6.3 (released version) with 3.6.2 or 3.6.3-dev6c immediately solves this issue. Regarding release notes for V3.6.3 I don't think this issue has to do with breaking changes, or does it?

proddy, 2023-11-20:
Do you see the values in the EMS-ESP web dashboard? Are there errors in the log of EMS-ESP or HA? Do they come back after a restart?

MichaelDvP, 2023-11-20:
Same issue as #1440, i've made a fix, but no feedback yet.

DoubleYou2, 2023-11-20:
Yes, I see alt values in the EMS-ESP dashboard. 
Yes, due to missing entities and attributes, HA log is full of error-messages (because my scripts try to access these attributes).
Restart of HA or EMS-ESP do not change situation. 

Is there already a fix for test available?

proddy, 2023-11-20:
I'm still thinking of what could have caused this. Do you see anything interesting in HA->devices->mqtt ? 

And no fix yet as I'm still on a bus on my way home. We're not google !

DoubleYou2, 2023-11-20:
Thanks for your support. 
In HA-> devices -> MQTT there is a red exclamation mark regarding climate.thermostat_hc1_stating "unavailable", but nothing really special.

My question regarding an available fix is because of MichaelDvP comment on a similar issue he is working on...

MichaelDvP, 2023-11-21:
> 'm still thinking of what could have caused this. 

The hc->mode is not set because in this line the `if` is wrong https://github.com/emsesp/EMS-ESP32/blob/ec85a7ec24b41d34bc1328ab6cc48a6d441681c3/src/devices/thermostat.cpp#L1030

A second bug is that i have missed the to adapt the setting here: https://github.com/emsesp/EMS-ESP32/blob/ec85a7ec24b41d34bc1328ab6cc48a6d441681c3/src/devices/thermostat.cpp#L2601

The fix is https://github.com/MichaelDvP/EMS-ESP32/commit/aed45968db1f92c2af756fdd239f7c4c61907e15
and now confirmed working in #1440. 

@DoubleYou2 Please also test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest, i think this wll fix the bug.

proddy, 2023-11-21:
@DoubleYou2 if you can try that build and let us know - then I'll do a patch

proddy, 2023-11-21:
thanks, we'll work on the patch.

Regarding those other entities, the shower ones are still there but will only show in HA after the first shower. `hpoperatingstate` is still there so maybe it's waiting on a value to come in. `dhw_energy_dhw` was renamed because of the duplicate dwh (there was a GH issue for this). Not sure what it's renamed to now, perhaps `total energy warm supplied`. In general it's best to look at the shortnames and not the full translated names which helps us locate the actual entity. And another good practice is when upgrading versions delete any old HA dashboards/cards and re-add them directly from HA->Devices&Service->MQTT

DoubleYou2, 2023-11-21:
... ok. Thanks...

commit cff5b8c9 (2023-11-21): Merge pull request #1444 from MichaelDvP/dev

set/read thermostat mode for RC100-RC300, fix #1440 and fix #1442
closed 2023-11-21 by proddy