emsesp/EMS-ESP32 #1443: missleading label name

Bingo2023, 2023-11-20:
rename "Heizgrenze Heizen" -> "Lstg.Zuheizer begrenzen" see pics below.

[image]
[image]
[image]

I have a BC400 with latest HW ans SW Version installer by Buderus some month ago. They have cleaned up & regrouped menus -> better overview - I like it :)

I addition:
"Außentemperatur Parallelmodus" -> "Bival. pkt. Parallelbetr." is also changed.
Like you can see on top all of these entries are in a menu called "Elektrischer Zuheizer".
maybe we could add something like "Elek. Zuheizer...." in front of all 4 entries? e.g. "Elek. Zuheizer: Lstg.Zuheizer begrenzen" then all of them would appear in one place in out list of 158 entries.

Bingo2023, 2023-11-21:
I'm with you Michael, to much renaming is not worth the effort.

But first point is valid, "Heizgrenze Heizen" = is used to limit the power of the electric heater (Heizstab/Zusatzheizer) - that is very difficult to understand, especialy with the other two limits for HP-heating and HP-WW.

proddy, 2024-01-06:
@Bingo2023 I'd like to close this out. Anything we still need to do on top of the latest dev (3.6.5-dev-8) ?

Bingo2023, 2024-01-08:
sorry could not find what has changed (without installing), can you point it out?

proddy, 2024-03-24:
We can change the DE translation if you tell us which are the new wordings (or do a PR on device_translations)

Bingo2023, 2024-05-09:
proposed:
Heizstab: Limit mit Kompressor
Heizstab: Limit Leistung
Heizstab: Limit für WW
Heizstab: Parallelbetrieb

proddy, 2024-05-10:
@MichaelDvP ?

MichaelDvP, 2024-05-10:
Ok for me. 
@Bingo2023 Please make a PR.

proddy, 2024-05-10:
could you make the PR against the `dev` branch please

closed 2024-06-21 by proddy