emsesp/EMS-ESP32 #1450: Write Access to Thermostat RC310 does not work any more in latest Versions

DoubleYou2, 2023-11-21:
PROBLEM DESCRIPTION

RC310 / Buderus GB125 / EMS-ESP V3.6.3 (released 21.11.23) & V3.6.4-dev_0 & some earlier DEV-Versions /  and HA latest version:

Using attributes "current_temperature" and "temperature" from entity "ems-esp Thermostat Hc1" for loop control.
"current_temperatue" is cyclic read and "temperature" (setpoint) is set if necessary for closed loop control.

Using V3.6.2 and Dev-versions up to V3.6.3-dev_6c setting attribute "temperature"  of entity "climate.theromstat_hc1" of HA
worked fine (although write accessess failed rarely and had to be repeated once or at maximum twice; maybe there
is a race condition in the timing of EMS / MQTT / Loop Control cycle; I don't care about this issue).

Problem: 
Regarding write accesses to attribute "temperature" every version after V3.6.3-dev_6c seems not work at all.
I checked this a few times by up- and downgrading ems firmware and can reproduce.

Summary:
Write accesses initiated by HA python script does not set attribute "temperature" of entity "climate.theromstat_hc1".
There is not additional error massage related to this unsuccessful write access.

EMS System Log logs all these (unsuccessful) write accesses without setting the corresponding attribute
(because my script repeats unsuccessful write accesses every 30 sec the EMS log is flooded with entries):

[image]

proddy, 2023-11-21:
can you post your system info?

What is good to help trace is the error is to go native into the Telnet Console,:
```sh
su
log all
call thermostat seltemp 22.5
```

and see what comes back (only the next 20 telegrams).

theFeverDog, 2023-11-21:
I have the same issue with my BC400 (combined with an RC220) and a GB192i.2, with every version starting from 3.6.3 final (also tested with 3.6.4dev0 and 3.6.4dev1)

When setting the value directly in EMS-ESP, the value updates in the BC400, and the change is reflected in the HC1 Thermostat in HA. However, when changing the value in the HA thermostat, the change does not get picked up.

derSchreifritz, 2023-11-21:
I think I have the same issue.
Not sure how to properly capture the log with telnet, it keeps moving, I managed to save the following file:
[putty.log](https://github.com/emsesp/EMS-ESP32/files/13429868/putty.log)
please let me know if I can do that better.
and this is my system info:
[emsesp_info(2).json](https://github.com/emsesp/EMS-ESP32/files/13429873/emsesp_info.2.json)

DoubleYou2, 2023-11-21:
@proddy : I'm on a business trip next 2 days. Is the feedback from derSchreifritz sufficient for your analysis? 

Otherwise I'll trace down end of the week...

proddy, 2023-11-21:
@theFeverDog from the logs it looks like calling from the Telnet Console, or the EMS-ESP Web UI is working. Only changing the temperature in HA is not working - is this a corrupt assumption? then we know which piece of code to look at (its 23K lines!)

When you change the value in HA (e.g. number.ems_esp_thermostat_hc1_seltemp) do you see any errors in the HA log file? You should also see a new MQTT topic called `ems-esp/thermostat/hc1/seltemp` with the temperature values as the payload.

DoubleYou2, 2023-11-21:
@proddy When looking at my use case I would expect that HA works correct. EMS System logs the correct command and value but it does not update the corresponding attribute...

proddy, 2023-11-21:
ok, I'll wait until @theFeverDog confirms.

theFeverDog, 2023-11-21:
No, I don't see any errors in the HA logs.

But I think I found the root cause for my problem: my heating is currently in "Manual" mode, but in HA it is listed as "Auto". If I change the temperature for climate.thermostat_hc1 (or number.ems_esp_thermostat_hc1_seltemp) nothing happens in this mode. Before 3.6.3 this worked fine. However, if change number.thermostat_hc1_manualtemp, this works fine.

However, if I set my heating to "Auto" mode on the BC400 or RC220, and then change climate.thermostat_hc1 , this gets taken into account correctly.

theFeverDog, 2023-11-21:
Just to add, if I use the service number.set_value in HA on number.thermostat_hc1_seltemp while in manual mode, I get the following error in EMS-ESP:

`[mqtt] MQTT command failed with error: invalid command (Not Found)`

DoubleYou2, 2023-11-21:
In my use case thermostat is in "Manual" mode and this is correctly shown in HA. 

PS:
In "AUTO" mode I would expect that any changes by HA would be overwritten by thermostat...

derSchreifritz, 2023-11-21:
if it helps:
when I change select.thermostat_hc1_mode to auto in HA, I can adjust the HA thermostat as desired. Then I change back to manual, the thermostat changes from auto to heat and cannot be changed anymore

theFeverDog, 2023-11-21:
> PS: In "AUTO" mode I would expect that any changes by HA would be overwritten by thermostat...

"Auto" just means it's running on a time plan. 
So it overwrites when a scheduled event happens (i.e. from day mode to night mode), however between two schedule points (i.e. during the day) you can adjust using HA and this stays until the next scheduled "event".

proddy, 2023-11-21:
> if it helps: when I change select.thermostat_hc1_mode to auto in HA, I can adjust the HA thermostat as desired. Then I change back to manual, the thermostat changes from auto to heat and cannot be changed anymore

could you try that call again from HA, but this time have MQTTExplorer running so you can see which MQTT topic is published.

theFeverDog, 2023-11-21:
The MQTT topic ems-esp/thermostat/hc1/seltemp gets updated when I change the temperature in HA, the change is just not picked up by EMS-ESP while in "Manual Mode"

MichaelDvP, 2023-11-21:
This #1398 enhencement is getting nasty, hopefully the last fix to make all modes work. Try:
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

derSchreifritz, 2023-11-21:
> "Auto" just means it's running on a time plan. So it overwrites when a scheduled event happens (i.e. from day mode to night mode), however between two schedule points (i.e. during the day) you can [...]

thanks, this explains something for me, I can confirm this. On my HPC410 display "auto" and "manual" in select.thermostat_hc1_mode is called "auto" (time program) and "optimise", in Operation mode. Not to be confused with select.thermostat_hc1_hpmode, which has heating, cooling and both. Or select.thermostat_hc1_hpoperatingmode, which now has heating, cooling, auto or off.  and I thought that one used to be reflected in the HA thermostat attribute, but I may be wrong.

derSchreifritz, 2023-11-21:
the last fix to make all modes work does the trick for me as far as I can see. Awesome!

theFeverDog, 2023-11-21:
> This #1398 enhencement is getting nasty, hopefully the last fix to make all modes work. Try: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

For me, this fixed the issue with the climate.thermostat_hc1 temperature (with a BC400/RC220).

Thanks!

philrich, 2023-11-21:
I also had the same problem with @MichaelDvP's version 3.6.4-dev.0 but i can confirm that version 3.6.4-dev.1 fixes it. Thanks!

commit 1e15f65b (2023-11-22): get mode for seltemp, fix #1450
commit 5ab10b7a (2023-11-22): fixes #1450
closed 2023-11-22 by proddy