emsesp/EMS-ESP32 #1452: WW Mode - Rego 3000/UI800/WSW196i/BC400

Bingo2023, 2023-11-22:
We should adapt the WW Mode for the Thermostat.
how the reaction is can be seen in the Picture I prepared.
Just one explanation: setting WW Mode to "auto": after command is send it will jumpt to "normal" and resulting in mode "Eco"

SW used:
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test

[image]

MichaelDvP, 2023-11-22:
The options are enumerated from telegram 0x2F5 offset 2: 0-off, 1-normal, 2-komfort, 3-auto, 4-ownprog
Then there was a request to add eco+ to this mode setting. If the enumeration for the BC400 is different we need to differ between RC300 and BC400. Maybe this is the right way also for the mode setting. 
Please check what number is used for what mode and tell.

Bingo2023, 2023-11-22:
please see attached log.
From: Aus -> Eco+ -> Eco -> Komfort -> Auto -> Aus
[WWmode_log.txt](https://github.com/emsesp/EMS-ESP32/files/13440165/WWmode_log.txt)

MichaelDvP, 2023-11-22:
> Aus -> Eco+ -> Eco -> Komfort -> Auto -> Aus

? -> 5 -> 1 ->  2 -> 4 -> 0
So eco+  is correct, but the missing 3 is nasty, also that here auto is 4, which is own prog for RC300. We need different flags for BC400 and RC300.

Bingo2023, 2024-03-24:
with latest official
EMS-ESP Version 3.6.5

it is broken again. Unfortunately something on the way to release got mixed up.

MichaelDvP, 2024-03-24:
Yes, it was only in the dev2-test-build. It needs a split for thermostats RC300 and BC400. I've updated the test-build to 3.7. We can merge this now to dev, so more users will test.

closed 2024-04-13 by proddy