emsesp/EMS-ESP32 #1453: Last Error and/or actual error Message not updatet @ GB172 with RC300

Peter-Aschinger, 2023-11-22:
### PROBLEM DESCRIPTION

_A clear and concise description of what the problem is._

I use the EMS-ESP32 (with actual 3.6.3 fw)
In the WebIf from EMS-ESP and also MQTT i can see a errormessage from 06.11.23.
When i delete the error history the last errormessage stays in the WebIf and Mqtt.
But during the day occur several same errormessages about lost the connection to the mixer.
This messages did not appear in the webif and also mqtt.

Please see the pictures and screenshot
[image]
[image]
[image]

```lua
  System information output here:

```

MichaelDvP, 2023-11-22:
6A(227) is a flame (ionization) error from boiler, the boiler stores a log in telegrams 0x10, 0x11 and we publish the newest. But these are only boiler errors.
A31 is a communication error, these errors are stored in the thermostat telegrams 0x12, 0x13 for my RC35 and we also show the newest for thermostat last error. Maybe your thermostat use different storage, try to `read 10 12` in terminal.
Actual errors are published in telegram 0xBE and 0xBF from the device that have the error. But these telegrams are not readable, only published on error.

Peter-Aschinger, 2023-11-25:
i watched the mqtt explorer a while an found out that it is puplished under the thermostat data:
[image]

proddy, 2024-06-25:
@Peter-Aschinger what shall we do with this issue?

proddy, 2024-06-25:
ok, I'll close for now and we can always re-open if it appears again, or others report it.

closed 2024-06-25 by proddy