emsesp/EMS-ESP32 #1456: VR4 and TW1 in Home Assistant

HansRemmerswaal, 2023-11-23:
### PROBLEM DESCRIPTION

I noticed in the EMS-ESP the sensor VR4 and TW1 since I implemented EMS-ESP-3_6_4-dev_2-ESP32.

Normally after a firmware upgrade I go to Home Assistant, open the History tab and add the devices **ems-esp Boiler** and **ems-esp Thermostat** to see if there are new sensors added or some sensors not there anymore. I was surprise to see that the VR4 and TW1 where not shown. I then then checked in the History tab if I could add the VR4 and TW1 directly as entity, and that worked.

So somehow the VR4 and TW1 are not considered, in the history tab, to be part of the device **ems-esp Boiler**.

I then went to the Settings -> Devices & Services -> Devices -> ems-esp Boiler. And there I do see the values as part of the group Diagnostic.

In that block Diagnostic I see 5 values which I do not see in History -> Choose device -> ems-esp Boiler...

### EXPECTED BEHAVIOUR

I would expect to see all the values from the ems-esp Boiler in History -> Choose device -> ems-esp Boiler...

### SCREENSHOTS

[image]

proddy, 2023-11-24:
which version did you upgrade from? Those are new entities which I think were added a few weeks ago. Diagnostic sensors in EMS-ESP are required now (see https://github.com/emsesp/EMS-ESP32/issues/1382#issuecomment-1793737317)

HansRemmerswaal, 2023-11-24:
On Nov 13th I upgraded to EMS-ESP-3_6_3-test_7a-ESP32. Not sure if at that time we already had that Diagnostic.
On Nov 23rd I upgraded to EMS-ESP-3_6_4-dev_2-ESP32 and I noticed that Diagnostic with sensors which do not show up in History -> Choose device -> ems-esp Boiler but only in History -> Choose entity.

Is this Diagnostic the same what this [help ](https://www.home-assistant.io/integrations/diagnostics/)document refers to?

_The diagnostics integration
 provides a way to download diagnostic data from a device or integrationIntegrations connect and integrate Home Assistant with your devices, services, and more.
[[Learn more]](https://www.home-assistant.io/integrations/), that can be shared in issue reports. Sharing diagnostics data when reporting an issue allows developers to diagnose and fix your reported problem quicker._

As that more sounds like diagnostic data of the ems-esp gateway itself and not about the sensors it's reading.

HansRemmerswaal, 2023-11-24:
To me it's not consistent that I see 20 EMS-ESP temperatures in "sensors" and 2 in "diagnostic".
And is it then a defect in HA that these temperature sensors in "diagnostic" are not visible in History -> Choose device -> ems-esp Boiler?

proddy, 2023-11-24:
ok.

closed 2023-11-24 by proddy