emsesp/EMS-ESP32 #1463: Add Heatpump Energy Consumption & Produced

proddy, 2023-11-25:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1462

<div type='discussions-op-text'>

<sup>Originally posted by **KLiebrecht1** November 25, 2023</sup>
ems-esp Version 3.6.3

Device HM200
Device-ID 0x53

Handler 0x04AF - Heatpump Energy Consumption - 4 Byte - 1/100 kWh
006+02:35:19.912 N 17: [emsesp] heatpump(0x53) -B-> All(0x00), ?(0x04AF), data: 00 02 85 22 00 00 00 00 00 00 00 00 (offset 24)
00 02 85 22 = 165154 = 1651.54 kWh

Handler 0x04AE - Heatpump Energy Produced - 4 Byte - 1/100 kWh
006+02:30:18.527 N 7: [emsesp] heatpump(0x53) -B-> All(0x00), ?(0x04AE), data: 00 08 8C 93 00 00 00 00 00 00 00 00 (offset 4)
00 08 8C 93 = 560275 = 5602.75 kWh

Can you please integrate them?</div>

proddy, 2023-11-25:
See https://github.com/emsesp/EMS-ESP32/discussions/1462#discussioncomment-7665537 for the Custom Entities

kai-morich, 2024-04-18:
you can remove v3.7.0 milestone, as it is already included in v.3.6.5

closed 2024-04-18 by proddy