emsesp/EMS-ESP32 #1474: MQTT for non standard port always uses TLS

ThetaGamma, 2023-12-03:
### PROBLEM DESCRIPTION

I want to use MQTT but not TLS and port 31883. 
According to https://github.com/emsesp/EMS-ESP32/blob/e00eb8e64f1926a6eda8075c34afd8cca38a37d6/lib/framework/MqttSettingsService.cpp#L380-L382  a empty cert and port > 8800 should make a insecure MQTT connection. tcpdump on the MQTT Server (Mosquito 1.6.3) shows Ems-esp makes a TLS connection .

### REQUESTED INFORMATION

- [ X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [X ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X ] Provide the output of http://ems-esp.local/api/system :

```lua
{
{
  "System Info": {
    "version": "3.6.5-dev.2",
    "platform": "ESP32-S3",
    "uptime": "000+14:22:52.482",
    "uptime (seconds)": 51772,
    "free mem": 196,
    "max alloc": 147,
    "free app": 6079,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems",
    "RSSI": -68,
    "IPv4 address": "192.168.1.15/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1",
    "BSSID": "set",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "MQTT publishes": 0,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 0,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 2588,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 203791,
    "bus reads (tx)": 21644,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 48,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": false,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Logano GB125/KB195i/Logamatic MC110",
      "device id": "0x08",
      "product id": 133,
      "version": "02.11",
      "entities": 71,
      "handlers received": "0xBF 0xC2 0x14 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers ignored": "0x08E4 0x02D6 0x2E 0x3B 0xF9 0x17 0x36"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.04",
      "entities": 49,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E",
      "handlers ignored": "0xE7 0x02E0 0x02EA 0xF9 0xBF"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "07.02",
      "entities": 0,
      "handlers ignored": "0x125A 0x2040 0x0CC3 0x0CBD"
    }
  ]
}
```

### TO REPRODUCE

- set MQTT server port to e.g. 31883
- remove TLS cert
- save settings
- packet capture on MQTT server

### EXPECTED BEHAVIOUR

Non-TLS connection opened

### ADDITIONAL CONTEXT

discussed on EMS-ESP Discord Channel MQTT 

Summary:
Looking into MqttSettingsService.cpp
My gut feeling is that it's related to the fact  setting _state.rootCA to the string "insecure" (and not an real empty string) and testing e.g. in line 60 for _state.rootCA.length() > 0 to enable TLS

Digging deeper into the code of MqttSettingsService.cpp: In case I understand the function MqttSettingsService::configureMqtt correct, a "secure MQTT connection" is made, if  (_state.rootCA.length() > 0)  - which is always true if it's set to the string "insecure"

proddy, 2023-12-03:
#1 is the most sexiest

ThetaGamma, 2023-12-05:
tested 3.6.5-dev.3 and problem is fixed. Great Job - Thanks!

commit 42ba93bd (2023-12-03): add checkbox for MQTT-TLS, fix #1474
closed 2023-12-05 by ThetaGamma