emsesp/EMS-ESP32 #1477: boil2hyston & boil2hystoff lost in version 3.6.4

pudermueller, 2023-12-03:
### PROBLEM DESCRIPTION

In version 3.6.4 (versus my previous old version 3.6.2) the parameters boil2hyston & boil2hystoff are lost in the MQTT and in the menue. I cannot read or change these values anymore. 

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32",
    "uptime": "000+00:41:47.266",
    "uptime (seconds)": 2507,
    "free mem": 143,
    "max alloc": 91,
    "free app": 419,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -68,
    "IPv4 address": "192.168.0.182/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "253.0.0.0",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 2457,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 0,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 0,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 1,
    "temperature sensor reads": 501,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 3697,
    "bus reads (tx)": 546,
    "bus writes (tx)": 1,
    "bus incomplete telegrams": 1,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 2,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 100
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GB125/GB135/MC10",  <== i have a GB135 with a 2 stage burner
      "device id": "0x08",
      "product id": 72,
      "version": "02.07",
      "entities": 74,
      "handlers received": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlers fetched": "0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35"
    },
    {
      "type": "thermostat",
      "name": "RC35",
      "device id": "0x10",
      "product id": 86,
      "version": "01.11",
      "entities": 55,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x42 0x48 0x47 0x52 0x51 0x5C 0x5B 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
      "handlers pending": "0x13 0x49 0x4C 0x53 0x56 0x5D 0x60",
      "handlers ignored": "0x1A 0x35"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.03",
      "entities": 0,
      "handlers ignored": "0x29"
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

Like in version 3.6.2 i had both parameters as read and change parameters. 

### SCREENSHOTS

Output from MQTT Explorer shows only boilhyston/off (without "2)
{"heatingoff":"OFF","heatingactive":"OFF","tapwateractive":"OFF","selflowtemp":57,"heatingpumpmod":100,"outdoortemp":-0.3,"curflowtemp":50.7,"boiltemp":50.5,"exhausttemp":57.3,"burngas":"OFF","burngas2":"OFF","flamecurr":0.0,"fanwork":"OFF","ignwork":"OFF","oilpreheat":"OFF","burnminpower":0,"burnmaxpower":100,"burnminperiod":10,"boilhyston":-8,"boilhystoff":8,"heatingactivated":"ON","heatingtemp":90,"heatingpump":"ON","pumpmodmax":100,"pumpmodmin":100,"pumpmode":"deltaP-2","pumpdelay":10,"setflowtemp":57,"setburnpow":100,"selburnpow":100,"curburnpow":0,"burnstarts":139766,"burnworkmin":1087906,"burn2workmin":563567,"heatworkmin":950853,"heatstarts":122952,"ubauptime":7541099,"servicecode":"0Y","servicecodenumber":0,"maintenancemessage":"H00","maintenance":"date","maintenancetime":6000,"maintenancedate":"12.10.2024","nompower":18,"nrgtotal":3201.82,"nrgheat":2026.91}

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pudermueller, 2023-12-03:
Hi Proddy,
restart didn't solve the issue. Both parameter are still gone. 

Telnet command delivers:
`ems-esp:$ read 8 16
000+00:03:08.541 N 2: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 5A 64 00 08 F8 0A 02 0A 64 64 02 0A F6 0F 0F 0F 0F 1E 05 04 09 04 00
`

Does this help?
Regards P

MichaelDvP, 2023-12-03:
There were some entities registered double, seems i have removed also these two by mistake. I'll add back.

proddy, 2023-12-03:
> There were some entities registered double, seems i have removed also these two by mistake. I'll add back.

that's it! thanks Michael

commit fd5f5d49 (2023-12-03): add back boil2hyst, fix #1477
closed 2023-12-11 by proddy