emsesp/EMS-ESP32 #1484: Keep getting "[mqtt] Publish failed: low memory" in log

mikkel75, 2023-12-05:
### PROBLEM DESCRIPTION

The system log keeps getting filled up with "[mqtt] Publish failed: low memory" warnings.

### REQUESTED INFORMATION
EMS-ESP Version
3.6.4

Device (Platform / SDK)
ESP32 / 4.4.6

System Uptime
000+00:16:12.197

CPU Frequency
160 MHz

Heap (Free / Max Alloc)
141 KB / 47 KB

Flash Chip (Size / Speed)
4,096 KB / 40 MHz

Application (Used / Free)
1,565 KB / 419 KB

File System (Used / Free)
24 KB / 40 KB

MichaelDvP, 2023-12-05:
> Heap (Free / Max Alloc)
141 KB / 47 KB

The warning only appear below 60/40 KB , this should work.
 Try to save some memory, disable unused services like OTA, set log buffer to minimum or log off, and remove unused entities in settings->customization from memory. Or use a esp32(S3) wih PSRAM on board.

proddy, 2024-01-03:
I've seen this error today during testing when I uploaded a wrong binary. Which espressif platform core it was built on and also the chip and memory is important.

closed 2024-01-03 by proddy