emsesp/EMS-ESP32 #1486: Thermostat HC1: can not set mode to OFF

bartplessers, 2023-12-06:
### PROBLEM DESCRIPTION

Recently I updated to EMS-ESP Version 3.6.4
I noticed that my home assistant automation to turn my heating system `OFF `does not work anymore.
Sample automation:

```
alias: Verwarming uit
description: ""
trigger:
  - platform: time
    at: "23:00:00"
condition: []
action:
  - device_id: fc7d3adde5ddb3170b13319aaae53729
    domain: climate
    entity_id: climate.thermostat_hc1
    type: set_hvac_mode
    hvac_mode: "off"
mode: single
```

HARDWARE:
BUDERUS Logamax PlusGB192iT40 - boiler
BUDERUS RC310 - thermostat
BUDERUS KM100 - gateway

Problem is that I can only change mode to `auto` or `heating`, but in both modes, the circulation pump keeps running (+/- 30W continiously) 

To save some power, I want to set my system OFF with ems-esp

### EXPECTED BEHAVIOUR

I want to set my system OFF with ems-esp

bartplessers, 2023-12-06:
MQTT command and response

```
Client (null) received PUBLISH (d0, q0, r0, m0, 'ems-esp/thermostat/hc1/mode', ... (3 bytes))
off
```

```
Client (null) received PUBLISH (d0, q0, r0, m0, 'ems-esp/thermostat_data', ... (1011 bytes))
{"datetime":"06.12.2023 12:54","intoffset":0.0,"floordry":"off","dampedoutdoortemp":5.4,"floordrytemp":0,"building":"medium","minexttemp":-10,"damping":"on","wwsettemp":60,"wwmode":"own prog","wwsettemplow":45,"wwcircmode":"auto","wwchargeduration":15,"wwcharge":"off","wwextra1":0,"wwdisinfecting":"off","wwdisinfectday":"tu","wwdisinfecttime":120,"hc1":{"seltemp":13.0,"currtemp":18.5,"haclimate":"roomTemp","mode":"manual","modetype":"eco","ecotemp":15.0,"manualtemp":13.0,"comforttemp":20.0,"summertemp":17,"designtemp":75,"offsettemp":0,"minflowtemp":25,"maxflowtemp":75,"roominfluence":4,"roominflfactor":4.0,"curroominfl":-4.0,"nofrostmode":"room","nofrosttemp":5,"targetflowtemp":19,"heatingtype":"radiator","summersetmode":"auto","summermode":"winter","controlmode":"weather compensated","program":"prog 1","tempautotemp":19.5,"remoteseltemp":0.0,"fastheatup":0,"switchonoptimization":"off","reducemode":"reduce","noreducetemp":-31,"reducetemp":0,"coolingon":"on","hpmode":"heating","control":"RC310"}}
```

proddy, 2023-12-06:
@bartplessers you said "I noticed that my home assistant automation to turn my heating system OFF does not work anymore."

did it work before? in which version of EMS-ESP?

bartplessers, 2023-12-07:
I have to dig further into this.
Indeed, **before** upgrade, I can see following:
[image]

At 00:15, temperature is set from 21 C to 13 C. (1)
This is done by **ANOTHER** automation.
Result is that 
- boiler stops working (2)
- circulation pump stops working (3)

this is OK.

But normally my automation to turn OFF the boiler should have been running at 23:00, but I can't see any change in behavior at that time. This is an indication that at that specific day (2023-05-06) 
- that automation did not run, OR
- the OFF command did not have any influence

so maybe I'm wrong on this.

I will dig further into my logs to see if the OFF command worked befor.

kind regards,
bartplessers

closed 2024-06-25 by proddy