emsesp/EMS-ESP32 #1487: Internal temperature offset could not be increased

douwefokkema, 2023-12-06:
### PROBLEM DESCRIPTION

With the RC100H (thermostat) HC1 internal temperature offset the value can be decreased but the increase of the value does not function.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

  System information output here:

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/13588500/emsesp_info.txt)

```

### TO REPRODUCE

Dashboard, select RC100H, select internal temperature offset, try to increase

### EXPECTED BEHAVIOUR

internal temperature offset to increase

proddy, 2023-12-06:
look at https://emsesp.github.io/docs/Troubleshooting/#changing-a-value-on-an-ems-device-doesnt-work

and always upgrade to the latest version when you find an issue. we're fixing many!

proddy, 2023-12-06:
ok, so this is known bug previously reported on our Discord [here](https://discord.com/channels/816637840644505620/816958087160266775/1178973237523263528) . User was on firefox browser.

Michael fixed it in the latest dev release. Please try that. https://github.com/emsesp/EMS-ESP32/releases/tag/latest

douwefokkema, 2023-12-06:
I used Firefox. Now tried MS Edge, same bug. Upgraded to 3.6.5-dev3. Now the parameter hc1 internal temperature offset is no longer present. But parameter exists, should be there. So there is an other error.