emsesp/EMS-ESP32 #1488: not setting hc1/seltemp with gui and/or api/http

harvey637, 2023-12-07:
### PROBLEM DESCRIPTION
since 3.6.5-test.3 I cannot set hc1/seltemp with api/script

I do this with a script, that is unchanged since long time, only easy stuff:
```
curl http://ems-esp.fritz.box/api/thermostat/hc1/seltemp -H 'Content-Type: application/json' -H 'Authorization: Bearer xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx' -d '{ "value" : 16.0 }'
/usr/bin/wget --no-check-certificate -q http://ems-esp/api/thermostat/hc1/seltemp/value -O -
```

This behaviour is new with the latest (test) version, so I write this issue here.

The same thing happens in the webgui:

I select a new value in the popup box, "update" ... but new value is not set.

Log contains:
```
2023-12-07 10:50:09.637    INFO 44:     [command]    Calling command 'thermostat/seltemp' (Sollwert Raumtemperatur) with value 18 and id 1 on device 0x10
2023-12-07 10:50:35.034    INFO 45:     [command]    Calling command 'thermostat/seltemp' (Sollwert Raumtemperatur) with value 16.00 and id 1 on device 0x10
``` 
The temperature 18(.0) was selected in GUI (but not updated), the temperature 16.0(0) was entered with the api/http call, also not updated.

```
<html><body>
<!--StartFragment-->
name | "seltemp"
-- | --
fullname | "HK1 Sollwert Raumtemperatur"
circuit | "hc1"
value | 17.5
type | "number"
min | 0
max | 30
uom | "°C"
readable | true
writeable | true
visible | true

<!--EndFragment-->
</body>
</html>
```

Thank you for looking into this.
I have not checked other writes ....
cu
Harvey

harvey637, 2023-12-07:
after reverting back to 3.6.4 both ways (gui and api/http) work as expected ....
so it must be something very new :-)
cu
Harvey

MichaelDvP, 2023-12-07:
Could you please attach the system info http://ems-esp.local/api/system/info, the values http://ems-esp.local/api/system/allvalues and a log (all) when changing changing the value. 
Does it happen in manual or auto-mode?

harvey637, 2023-12-07:
Hi,
yes, but later in the evening ... gone away now :-)
But everything works fine after going back to 3.6.4, and as this is the productive (my only) system for my home I will get in trouble making too much non working heating stuff over the weekend (like going to a not working version) .... its could outside.
cu you later
Harvey

harvey637, 2023-12-08:
Hi, after quick check of your latest version (1h ago), only checked gui ..... WORKING again!
Great work, thank you!
cu
Harvey

harvey637, 2023-12-08:
if nobody complains ... closed

closed 2023-12-08 by harvey637

kriger1965, 2023-12-09:
No - I don´t think that it is fixed now ! I just tested ....in Version 3.6.4 it workes. If I upgrade to Version 3.6.5-dev3 it does not work  - so I needed to rollback to 3.6.4.   !!!

kriger1965, 2023-12-09:
my test is: open in HA the device to edit: 
[image]
In Version 3.6.4 ...if you changes the value 41°C then in the web-gui of ems-esp after refresh it will be changing also to  the same value ...in version 3.6.5-dev3 it definitly does not work !!!

harvey637, 2023-12-09:
Hi, i was  running  3.6.5.TEST3, See the commitlog,  this is the Version i am running and IT IS ok.I did Not check 3.6.5.DEV3 .......CuHarvey

kriger1965, 2023-12-09:
hmm - Where did you download "3.6.5.TEST3" ?  Oh I can see in github - I just made the test:
- upgrade to 3.6.5-test.3  ..."Sie verwenden derzeit 3.6.5-test.3 (ESP32)"
- open the device in HA: "dhw selected temperature"
- try to edit the value in HA
- then visit the web-gui from ems-esp ...and refresh ...and check if the value also will be changing

...so I can report: 
- Version: 3.6.5-test3 ...the issue is NOT solved for me  !!!   I cannot use this Version   ...using GB162, with RC310 and HA Version 2023.12.1
- back to version 3.6.4.    ...the issue is not existing  ...every changes of the value in HA will be send to ems-esp and  changing the value in ems-esp

... so please hook on to fix this bug.  If I can help the development engineer with logging or debugging please let me know!

kriger1965, 2023-12-09:
[info.json](https://github.com/emsesp/EMS-ESP32/files/13623858/info.json)
[allvalues.json](https://github.com/emsesp/EMS-ESP32/files/13623859/allvalues.json)

here the logfile: 
[log.txt](https://github.com/emsesp/EMS-ESP32/files/13623878/log.txt)

kriger1965, 2023-12-09:
Here the logfile with version 3.6.4  ...there are no issues:
[log(1).txt](https://github.com/emsesp/EMS-ESP32/files/13623889/log.1.txt)

kriger1965, 2023-12-09:
please tell me if you will reopen this issue or should I open a new issue?

harvey637, 2023-12-09:
Hi, Open new one, as mine IS solved.Good luckyHarvey--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

proddy, 2023-12-09:
@kriger1965 best indeed if you can create a new issue. Make sure you attach your system info, and try from the Telnet Console after a `su`, `call boiler wwseltemp 41` to see if the command is correctly sent.